MarlinFirmware/Marlin #13445: [BUG] Single byte configuration values not correctly handled when storing settings on STM32F1 CPU flash

pinchies, 2019-03-21:
### Description

Flash on STM32F1 CPU can only save two pages at a time. The STM32F1 HAL file **persistent_store_flash.cpp** ensures this, by rounding up the setting write index after each write

persistent_store_flash.cpp , line 100
`pos += ((size + 1) & ~1);`

Unfortunately, the configuration file **configuration_store.cpp** doesn't seem to be aware of this. If the DEBUG_EEPROM_READWRITE flag is enabled
`  #if ENABLED(DEBUG_EEPROM_READWRITE)`
 Then, this rounding causes the EEPROM write index position to differ from the size calculated from the struct offsetof(), resulting in the EEPROM save attempt failing.

### Steps to Reproduce

Use a STM32F1 example, such as the JGAurora A5S/A1 motherboards.
1. Enable EEPROM support in configuration.h
2. Enable **DEBUG_EEPROM_READWRITE** 
3. Enable both **EEPROM_SETTINGS** and **FLASH_EEPROM_EMULATION**
4. Change MCU in EEPROM.h e.g. #define MCU_STM32F103ZE
5. Compile and flash
6. Run M502, then M500

**Expected behavior:** Default settings should be loaded, and then saved.

**Actual behavior:** Some settings appear to be written to the FLASH/EEPROM, but due to errors encountered, the version and checksum information is not written, so the data cannot be loaded.

thinkyhead, 2019-03-24:
I'll patch this now…

thinkyhead, 2019-03-25:
The (final) above patch should allow word-padded EEPROM to work correctly. Let me know if you see any other quirks!

commit d4201f90 (2019-03-25): Handle word-padded flash-based eeprom (STM32F1)

Fix #13445
commit 380c7719 (2019-03-25): Handle word-padded flash-based eeprom (STM32F1)

Fix #13445
closed 2019-03-25 by thinkyhead