MarlinFirmware/Marlin #13447: [BUG] commits 5b2c37d and 691e5c3 cause regressions with pre-3.0 BLTouch

salfter, 2019-03-21:
### Description

Homing no longer works properly as of commit 5b2c37d if you have a pre-3.0 BLTouch.  

I've also noticed that the BLTouch menu items no longer work properly as of commit 691e5c3.  "Reset" extends the pin, and when you try to push it back in, it drops down again.  (This is normal behavior for "deploy.")  The other options do nothing.

Reverting to commit 49cf92d restores expected behavior.

### Steps to Reproduce

1. merge 691e5c3, compile, and flash printer with a BLTouch 2.0
2. test homing...it still works
3. test BLTouch menu...it doesn't
4. merge 5b2c37d, compile, and flash printer
5. now homing doesn't work either

**Expected behavior:** [What you expect to happen]

See https://home.alfter.us/s/3QXJ3njrx338M4B for video of my printer homing properly.

**Actual behavior:** [What actually happens]

See https://home.alfter.us/s/Px7rSWjjjkya96A for video of failed homing.

(Both videos will work better if you download them before viewing; my home upstream bandwidth is limited and probably insufficient to stream without buffer underruns.)

#### Additional Information

[config.zip](https://github.com/MarlinFirmware/Marlin/files/2993600/config.zip)

InsanityAutomation, 2019-03-21:
I'll go over the current code base again tonight. For the menus, there is an open PR to fix that right now.

A few things changed when it was broken out to a class of it's own that needed to be caught up.

InsanityAutomation, 2019-03-22:
Just ran a test on 3 machines with the code from PR #13441 and the current snapshot without an issue on legacy probes (V1 metal pin and V2 plastic) and 3.0 and didnt have an issue here. I added some output to the bltouch commands in that PR, can you try the code there and report back with the serial log so we can see what its getting?

Also, have you tried to force either 5v or od mode?

salfter, 2019-03-22:
The menu works now, but homing still doesn't.  I applied your PR and got this in OctoPrint's serial log:

```
2019-03-22 08:36:59,179 - Recv:  T:29.00 /0.00 B:31.72 /0.00 @:0 B@:0
2019-03-22 08:37:01,179 - Recv:  T:29.00 /0.00 B:31.72 /0.00 @:0 B@:0
2019-03-22 08:37:03,177 - Recv:  T:29.00 /0.00 B:31.72 /0.00 @:0 B@:0
2019-03-22 08:37:03,490 - Send: G91
2019-03-22 08:37:03,511 - Recv: echo:Unknown command: ""
2019-03-22 08:37:03,514 - Recv: ok P15 B3
2019-03-22 08:37:03,519 - Send: G28 Z0
2019-03-22 08:37:03,533 - Recv: BLTouch Command :160
2019-03-22 08:37:03,676 - Recv: BLTouch Command :140
2019-03-22 08:37:03,827 - Recv: BLTouch Command :90
2019-03-22 08:37:03,979 - Recv: BLTouch Command :140
2019-03-22 08:37:04,130 - Recv: BLTouch Command :10
2019-03-22 08:37:04,282 - Recv: BLTouch Command :60  
2019-03-22 08:37:04,802 - Recv: BLTouch Command :140
2019-03-22 08:37:04,954 - Recv: BLTouch Command :90
2019-03-22 08:37:05,178 - Recv:  T:29.00 /0.00 B:31.72 /0.00 @:0 B@:0
2019-03-22 08:37:05,527 - Recv: echo:busy: processing
2019-03-22 08:37:07,178 - Recv:  T:29.00 /0.00 B:31.72 /0.00 @:0 B@:0
2019-03-22 08:37:07,526 - Recv: echo:busy: processing
2019-03-22 08:37:07,743 - Recv: BLTouch Command :140
2019-03-22 08:37:07,895 - Recv: BLTouch Command :10
2019-03-22 08:37:08,046 - Recv: BLTouch Command :60
2019-03-22 08:37:08,866 - Recv: BLTouch Command :140
2019-03-22 08:37:09,017 - Recv: BLTouch Command :90
2019-03-22 08:37:09,181 - Recv:  T:29.00 /0.00 B:31.72 /0.00 @:0 B@:0
2019-03-22 08:37:09,534 - Recv: echo:busy: processing
2019-03-22 08:37:11,180 - Recv:  T:28.91 /0.00 B:31.72 /0.00 @:0 B@:0
2019-03-22 08:37:11,532 - Recv: echo:busy: processing
2019-03-22 08:37:13,056 - Recv: X:136.00 Y:150.00 Z:15.50 E:0.00 Count X:13615 Y:15068 Z:6204
2019-03-22 08:37:13,065 - Recv: ok P15 B3
2019-03-22 08:37:13,075 - Send: G90
2019-03-22 08:37:13,084 - Recv: ok P15 B3
2019-03-22 08:37:13,178 - Recv:  T:28.87 /0.00 B:31.70 /0.00 @:0 B@:0
```

It looks like it's throwing an undefined command (140) at the probe several times, which is possibly confusing it.

I've put up another video with a closeup on the BLTouch during Z-axis homing...you can see that it never deploys the pin properly.

>Also, have you tried to force either 5v or od mode?

It's a v2, so this should have no effect as it's hardwired to 5V operation and only operates in open-drain mode with a hardware mod, which I haven't done since I use it with an Arduino Mega/RAMPS 1.6 stack.  Nevertheless, I tried with both BLTOUCH_V3 and BLTOUCH_FORCE_5V_MODE defined...didn't make a difference.

I'm patching some debugging info into my working source tree so I can see what sequence of commands previously went to the probe that worked.  I'll post an update when I have results.

salfter, 2019-03-22:
...and the log from my working configuration:

```
2019-03-22 10:14:33,076 - Send: G28 Z0
2019-03-22 10:14:33,088 - Recv: MOVE_SERVO: 160
2019-03-22 10:14:33,239 - Recv: MOVE_SERVO: 90
2019-03-22 10:14:34,652 - Recv:  T:20.50 /0.00 B:23.87 /0.00 @:0 B@:0
2019-03-22 10:14:35,087 - Recv: echo:busy: processing
2019-03-22 10:14:36,651 - Recv:  T:20.50 /0.00 B:23.87 /0.00 @:0 B@:0
2019-03-22 10:14:36,860 - Recv: MOVE_SERVO: 10
2019-03-22 10:14:37,089 - Recv: echo:busy: processing
2019-03-22 10:14:38,651 - Recv:  T:20.50 /0.00 B:23.87 /0.00 @:0 B@:0
2019-03-22 10:14:39,089 - Recv: echo:busy: processing
2019-03-22 10:14:40,653 - Recv:  T:20.50 /0.00 B:23.91 /0.00 @:0 B@:0
2019-03-22 10:14:41,091 - Recv: echo:busy: processing
2019-03-22 10:14:42,652 - Recv:  T:20.50 /0.00 B:23.78 /0.00 @:0 B@:0
2019-03-22 10:14:43,090 - Recv: echo:busy: processing
2019-03-22 10:14:44,650 - Recv:  T:20.50 /0.00 B:23.81 /0.00 @:0 B@:0
2019-03-22 10:14:45,109 - Recv: echo:busy: processing
2019-03-22 10:14:46,655 - Recv:  T:20.50 /0.00 B:23.78 /0.00 @:0 B@:0
2019-03-22 10:14:47,108 - Recv: echo:busy: processing
2019-03-22 10:14:47,129 - Recv: MOVE_SERVO: 90
2019-03-22 10:14:48,656 - Recv:  T:20.50 /0.00 B:23.75 /0.00 @:0 B@:0
2019-03-22 10:14:49,130 - Recv: echo:busy: processing
2019-03-22 10:14:49,914 - Recv: MOVE_SERVO: 10
2019-03-22 10:14:50,655 - Recv:  T:20.50 /0.00 B:23.94 /0.00 @:0 B@:0
2019-03-22 10:14:51,110 - Recv: echo:busy: processing
2019-03-22 10:14:52,654 - Recv:  T:20.50 /0.00 B:23.72 /0.00 @:0 B@:0
2019-03-22 10:14:53,108 - Recv: echo:busy: processing
2019-03-22 10:14:54,656 - Recv:  T:20.50 /0.00 B:23.94 /0.00 @:0 B@:0
2019-03-22 10:14:55,110 - Recv: echo:busy: processing
2019-03-22 10:14:55,217 - Recv: MOVE_SERVO: 90
2019-03-22 10:14:56,654 - Recv:  T:20.50 /0.00 B:23.84 /0.00 @:0 B@:0
2019-03-22 10:14:57,110 - Recv: echo:busy: processing
2019-03-22 10:14:58,658 - Recv:  T:20.50 /0.00 B:23.84 /0.00 @:0 B@:0
2019-03-22 10:14:59,108 - Recv: echo:busy: processing
2019-03-22 10:14:59,256 - Recv: X:136.00 Y:150.00 Z:15.50 E:0.00 Count X:13615 Y:15068 Z:6204
2019-03-22 10:14:59,258 - Recv: ok P15 B3
2019-03-22 10:14:59,260 - Send: G90
2019-03-22 10:14:59,272 - Recv: ok P15 B3
2019-03-22 10:15:00,656 - Recv:  T:20.50 /0.00 B:23.87 /0.00 @:0 B@:0
```

salfter, 2019-03-24:
Looks like everything is fixed as of 407202c.  w00t!

closed 2019-03-24 by salfter