MarlinFirmware/Marlin #13448: BIGTREETRECH SKR V1.3 32bit LPC1768

forkoz, 2019-03-21:
Since the old thread is closed....

My SKR 1.3 arrived today. Look forward to getting it working over the weekend. I know some pins have changed and I'm still not sure if they fixed the LCD cable orientation. Either way, sBase goes in the box and LV8729s can come out and play again. 

Going to try 1/64 stepping. I know most are going to use with TMCs and the wire-less software configuration. Maybe one day in the future.

I know marlin is ready for v1.1 but I haven't looked to see if 1.3 is set up already.

InvalidExcepti0n, 2019-03-21:
Pins for SKR 1.3 is already there and supported, check [pins_BIGTREE_SKR_V1.3.h](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pins_BIGTREE_SKR_V1.3.h) for more details

If you already tried it post issue what is specific related to board/config issue

forkoz, 2019-03-21:
[image]
Ok what are the 5 jumpers there? They all came filled in.

InvalidExcepti0n, 2019-03-22:
Here is example (Explained on page 15 [DRV8834](https://www.pololu.com/file/0J617/drv8834.pdf))
Hope this gonna help you to understand (also it is different for each motor driver) so best advice would be check datasheet

forkoz, 2019-03-22:
Yea but what pin is that? To gnd or vcc? Most boards only give you the MSx pins and in their documentation that jumper doesn't even exist.

InvalidExcepti0n, 2019-03-22:
If you look schematic for SKR-V1.1 on their github **SKR-V1.1SchDoc.pdf** you can see that one side is connected to 3.3V(on this board that would be red pins) and other side to step motor driver (black pins)
(if you have multi-meter you can check connection easily)

Example
If you  wanna TMC-2130 to work in:
- SPI -  (jumper should be set to RED side) 
- STEP/DIR (all 3 opened/jumper removed) will give you 1/256 / Stealthchop1 (more in datasheet)

I can't  confirm yet as still didn't get my 1.3 board and schematic is missing so till than this is my assumption (based on 1.1 schematic)

forkoz, 2019-03-22:
You misunderstand, there is no red side. There is only off and on. Its a 2 pin jumper.
[image]

gloomyandy, 2019-03-22:
If you mean the red two pin socket that is vertical in your picture I think you will find that is provided to allow TMC drivers to feed the diag pin output into the board for sensorless homing etc. It is not a jumper it is an extension to the normal driver socket.

forkoz, 2019-03-22:
YES! Thank you. It looked like a jumper.

landodragon141, 2019-03-22:
I also just got mine yesterday. Anyone had a chance to play with it yet? I'll be running mine with 4 TMC2130s in SPI mode.

gloomyandy, 2019-03-22:
I have a couple of TMC2208s hooked up and working to check that the new software serial works with it. But other than that I've not tried very much.

forkoz, 2019-03-23:
Whats the max step rate for LPC boards? Think I can get away with 1/128?

p3p, 2019-03-23:
80K before multistepping even kicks in, I haven't really tested at really high rates other than varying pulse train generation though.

forkoz, 2019-03-23:
It said on the reprap step page that 1/128 was achievable at 150mm/s and since I never exceed that I guess I'll give it a go. I think with what I have it gives something like 6k steps/mm when added up. So possibly its going into multistepping at that point. If I'm doing this right with it being 1500 @ 1/32 right now, its already hitting multistep after 53mm/s. If it sucks I guess I can just go lower.

forkoz, 2019-03-23:
So of course I install my board and serial port "code 10: cannot start" and LCD doesn't display anything.

Yannik, 2019-03-24:
Anyone else having issue with E1? 

I've opened issues #13469 and #13470.

forkoz, 2019-03-24:
heh, you made my sunday... i really hope its a marlin bug.

p3p, 2019-03-24:
@forkoz 6K steps/mm @ 150 mm/s is 900K steps/s that's a fair bit above what we can support, I wouldn't go above 80K myself, micro-stepping is not that useful apart from making the motors quieter.

forkoz, 2019-03-24:
Yep, I'm leaving it at 1/64... if that extruder can extrude. We could have just gotten 2 boards with defects.

forkoz, 2019-03-25:
@yannik: you just made my week.....

Turns out marlin won't work if you have an E1 but no E0. Redefining the extruder pins did the trick.

forkoz, 2019-03-25:
So all drivers in this board are run at 3.3v, there is a mod to fix it 
[image]

kAdonis, 2019-03-25:
@forkoz 
The above mod shouldn't be necessary. All drivers (in Stepstick- format) I know are fine with 3,3V,
even LV8729 and, according to Pololu, STSPIN82

forkoz, 2019-03-26:
That's what I thought but my LV8729 vref would not adjust past .50 until I did this. Plus multiple people are telling me that LV8729 can't be used in E0 because of some compatibility issue.

choryuidentify, 2019-04-02:
Bigtreetech publish SKR 1.3 Schematic.

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/d3c60d559b3169455445caf457426961f04afefe/hardware/SKR-V1.3-SCH.pdf

stoicastefanv, 2019-04-05:
Hi guys. Hope you do not mind, im asking the same question ... Im an owner of skr1.3 about 12 ours .. Mi issue is with the lcd2004 i just uploaded the firmware and changed 2 perfectly working lcd2004 with ramps board. Had you encountered this issue how did you resolved it. I read something on the web that i need rotate 180 exp1 and exp2 plugins before putting them into the bord. Im asking because i do not whant to toast my board so quickly. 

Kind regards all and thanks

forkoz, 2019-04-05:
Rotating EXP1 and EXP2 at only ONE end  is how I got my reprap discound full graphics display working on sBase.  I've put them the wrong way around before and didn't blow anything up. Supposedly SKR1.3 fixed the rotation issue on the connectors, and that holds true for RRGLCD. My 2004 is collecting dust since it never worked on sBase... so try it both ways and see if it works. On the FB group I saw a lot of people complaining that LCD2004 never worked.

stoicastefanv, 2019-04-05:
No way ...:(( thank you alot ... Test time!

stoicastefanv, 2019-04-05:
Yessss is working, allmost. For all others i had flipped 180 degrees the EXP1 connector..  But when i try to rotate the knob to go thru menue the MCU reset .. But i olso have an error (TMC connection error). Maybe if i clear the error it will stop reset itself.

stoicastefanv, 2019-04-07:
i'm back with questions because i can not go further 
1. what is for the BOOT connector? located above the MCU?
2. i still get TMC connection error :( what i had done : 
for all TMC2130 V1.1 from BIGTREETECH soldered the two small pads and removed the R5 and diagnose pin for the endstops

[image]

[image]

mounted the jumpers according to image below
[image]
but when i send M122 bad luck 

		X	Y	Z	E
Enabled		false	false	false	false
Set current	800	800	800	800
RMS current	1436	1436	1436	1436
MAX current	2025	2025	2025	2025
Run current	25/31	25/31	25/31	25/31
Hold current	12/31	12/31	12/31	12/31
CS actual		0/31	0/31	0/31	0/31
PWM scale	0	0	0	0
vsense		0=.325	0=.325	0=.325	0=.325
stealthChop	false	false	false	false
msteps		256	256	256	256
tstep		0	0	0	0
pwm
threshold		0	0	0	0
[mm/s]		-	-	-	-
OT prewarn	false	false	false	false
off time		0	0	0	0
blank time	16	16	16	16
hysteresis
-end		-3	-3	-3	-3
-start		1	1	1	1
Stallguard thrs	8	0	0	0
DRVSTATUS	X	Y	Z	E
stallguard
sg_result		0	0	0	0
fsactive
stst
olb
ola
s2gb
s2ga
otpw
ot
Driver registers:
		X	0x00:00:00:00	 Bad response!
		Y	0x00:00:00:00	 Bad response!
		Z	0x00:00:00:00	 Bad response!
		E	0x00:00:00:00	 Bad response!
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW 
 
When i made the M122 test, the board was supplied from 24 v (jumper between INT and 5v)

Then started to look at the board to check the connections are according to the schematic:
[https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/hardware/SKR-V1.3-SCH.pdf](url) 
and the MOT_MOSI MOT_SCK MOT_MISO are short to ground ???
[image]
[image]
[image]
[image]
Can someone check this ? and sorry for the mess on my table
Is my interpretation wrong ? I'm doing something wrong?
Kind regards all

jjansen85, 2019-04-07:
Did you enable #define TMC_USE_SW_SPI in configuration_adv.h? (that was my mistake while setting up the TMC2130 SPI on this board..)

stoicastefanv, 2019-04-07:
Yes i double checked and is on enabled..!

gloomyandy, 2019-04-07:
Have you tried just installing a single driver at a time into say the X location to see if any of the drivers work? Check them one at a time, if there is a problem with any of them it may cause all to report an error if you have the bad one installed.

stoicastefanv, 2019-04-08:
Gllomyandy . Yes ... I done that too. All the drivers i put them in different places.  I olso removed sd card from hardware and software.... 
1. Maybe i undersood wrog .. I need jumper wires for SPI? Because no big thing .. I Installed everything according to the bigtreetech youtube video for skr1.3
2.Please while you keep the printer off on skr1.3 remove one driver and please check the continuity between mosi miso sck pins to GND.. As i made above. My big fear is that i have a short between pins and groud even i do no see it. Please someone chech
3. What is for the boot pins above MCU ?

Kind regars

Hawxxer, 2019-04-10:
Same problem "TMC connection error". The "X-DIAG" pins must also be connected to the board right?

stoicastefanv, 2019-04-10:
Hi there 
X-diag pin if you what not to use endstops and you use the TMC2130 as endstop
But i soldered only one pin not both as jumpers in the board because one is diagnostic and one is vref used by the driver. I was afraid not to interfear with MKS board
Can you please check continuity as i requested above ?   Kind regards

Hawxxer, 2019-04-10:
OK I checked mine and again and two of my drivers dont want to communicate M122 shows that the other two works so thats my fault. But I checked you request. MOSI and SCK fluctuating around 100 Ohm MISO is HIGH Ohm. But you should defnitly check with an logic analyser or oszi. Did you check when the Controller is off, than they should all be HIGH Ohm. 
[image]

stoicastefanv, 2019-04-10:
What logic analyzer did you use?? I need one because i do not have. 
Yes the board was off and removed all the jumpers in order to eliminate anny posibility to be a short circuit ..  And that is why i think something is in short ... With the board on and off measurements are Low Ohm.
Your oppinion is not a defective bord but 5 defective TMC2130 drivers? 
Anny other ideas for me to check because im out.

jjansen85, 2019-04-10:
I measured the same pins (X-axis) on my machine (no power). 
All four are about 600 Ohms. 
When the board is On, the values are different (some lower, some higher).

When I got the board there was some metal dirt on the connectors of the NXP CPU. Can there be some dirt on your pcb causing shorts?

Hawxxer, 2019-04-10:
I use a saleae one, cause I got student discount idk if they are better than others, the build quality is good and bus sniffing works fine, software is some kind of strange but works too, maybe check them out. 

So I unpluged every Jumper on board (except of the Power Jumper set to Int +5V) + the board ist not plugged in and I always get HIGH Ohm  on every pin. 

> When I got the board there was some metal dirt on the connectors of the NXP CPU. Can there be some dirt on your pcb causing shorts?

Yeah check this definitly I killed three tmc2130 cause of solder or dust on the pcb maybe also connected solder joints like on the picture. 

No I dont think all of your five TMC2130 are defect, maybe its the board

zombu2, 2019-04-10:
turns out when the board is only connected to usb the trinamics don't talk they need power from the 24v input

doggyfan, 2019-04-15:
BigTreeTech/MKS have no QC and don't test their boards properly (if at all), you often get faulty boards with components badly placed, shorts between pads etc :( .. I know from experience.

zombu2, 2019-04-15:
It sure looks like some components moved on that board or didn t get aassembled right 
the 2 board i have are clean

stoicastefanv, 2019-04-15:
Ive done everything so far... Nothing works...thank you all for help... As last resort i will try using dupond cables (not to bend or cut the tmc2130 pins) to install them on ramps board and then back to skr1.3 but to use different mosi moso sck pins i think the one used for the LCD. I will get back with updates.
Kind regards all

stoicastefanv, 2019-04-16:
Hi all, as previously told i got back with updates:
I used 12V ATX power supply +  Arduino mega + ramps 1.4 + Marlin 1.1.9
Do not lough because it works, :))

[image]
[image]
[image]

I use Dupond wires to connect TMC2130 and ramps because i did not wanted to bend or cut mosi miso sck and cs pins
Attention when define the CS pin not to overlap with MOSI MISO SCK default pins in pins_RAMPS.h

Changes i had done in Marlin:

Configuration.h
> #define SERIAL_PORT 0
> #define POWER_SUPPLY 1
> #define X_DRIVER_TYPE  TMC2130
> // For Inverting Stepper Enable Pins (Active Low) use 0, Non Inverting (Active High) use 1
> // :{ 0:'Low', 1:'High' }
> #define X_ENABLE_ON 1
> //#define SDSUPPORT`

Configuration_adv.h
> // Microstep setting (Only functional when stepper driver microstep pins are connected to MCU.
> #define MICROSTEP_MODES { 16, 16, 16, 16, 16 } // [1,2,4,8,16]
> #define X_MAX_CURRENT     1000 // in mA
> #define X_SENSE_RESISTOR    91 // in mOhms
> #define X_MICROSTEPS        16 // number of microsteps
> //#define TMC_USE_SW_SPI
if i uncomment define #define TMC_USE_SW_SPI and set CS to 44 will not work because il interface with the pins defined in pins_RAMPS.h
> #define STEALTHCHOP
> #define TMC_DEBUG

pins_RAMPS.h
> #define X_STEP_PIN         54
> #define X_DIR_PIN          55
> #define X_ENABLE_PIN       38
> //#define X_CS_PIN         53
> #define X_CS_PIN         44
> //#ifndef X_CS_PIN

if you uncomment above //#define TMC_USE_SW_SPI below pins will be assigned and will not work because will conflict pin 44
> /**
>  * Default pins for TMC software SPI
>  */
> #if ENABLED(TMC_USE_SW_SPI)
>   #ifndef TMC_SW_MOSI
>     #define TMC_SW_MOSI    66
>   #endif
>   #ifndef TMC_SW_MISO
>     #define TMC_SW_MISO    44
>   #endif
>   #ifndef TMC_SW_SCK
>     #define TMC_SW_SCK     64
>   #endif
> #endif

by uncommenting //#define TMC_USE_SW_SPI will use by default 
> pin 50 MISO
> pin 51 MOSI
> pin 52 SCK

result is 

> SENDING:M122
> 		X
> Enabled		true
> Set current	800
> RMS current	795
> MAX current	1121
> Run current	25/31
> Hold current	12/31
> CS actual		12/31
> PWM scale		255
> vsense		1=.18
> stealthChop	true
> msteps		16
> tstep		1048575
> pwm
> threshold		0
> [mm/s]		-
> OT prewarn	false
> OT prewarn has
> been triggered	false
> off time		5
> blank time	24
> hysteresis
> -end		2
> -start		3
> Stallguard thrs	0
> DRVSTATUS	X
> stallguard
> sg_result		0
> fsactive
> stst		X
> olb
> ola
> s2gb
> s2ga
> otpw
> ot
> Driver registers:	X = 0x80:0C:00:00

i think everything is ok, at least TMC2130 is communicating with the Arduino board, i did not make any move. i did not connect the stepper to the driver 
If i had done something wrong please correct me.

glebozaur, 2019-04-16:
Hi guys. I have a delta. Please help me solve the problem with SKR v1.3 + TMC 2130 v1.1 via SPI. Drivers work correctly, but the Sensorless Homing does not work. With mechanical endstops drivers work fine. I tried to weld a pin to diag1 connection on X,Y,Z and it caused no any changes. I put all jumpers correctly as indicated in Biqu instruction and I modify Config files also as indicated in official youtube BIQU video instruction. I dont know what to do. Here is my video of Sensorless Homing failed: [https://youtu.be/jiGckFfKWgE](url)
[image]
[image]
[image]
[image]
[image]
[image]

Black6spdZ, 2019-04-30:
In the same boat.. CANNOT get movement from my TMC2130s.. M122 shows connection OK on all 5 but enabled is "false". Invert stepping pins enable makes no difference. Im at a loss

Black6spdZ, 2019-04-30:
Ahah.. Success.. I had enable invert to 1 and software enable to 1 as well, used those on previous 1.1 board

closed 2019-05-18 by thinkyhead

ritor1sefa, 2019-05-20:
Black6spdZ, 
I did not find the possibility of private messages here, unfortunately.
So what did you do in the end? I get garbage, as I do not switch: https://i.imgur.com/cfzi3qq.png
X	0xFF:FF:FF:FF	 Bad response!
Testing X connection... Error: All HIGH
Share, please, configs: Configuration.h Configuration_adv.h & pins_RAMPS.h

Black6spdZ, 2019-05-20:
Make sure your drivers are configured for SPI mode like this photo.. "Jumpers on cfg4-5 and NO R5
https://cdn.shopify.com/s/files/1/1619/4791/files/wire.png?v=1531713659

ritor1sefa, 2019-05-20:
I bought the kind of already configured in the mode.
https://i.imgur.com/7Qopqy1.jpg

stoicastefanv, 2019-05-22:
Guys i have an issue / i whant to prevent some smoke... 
1.I have some new steppers 42GYGHM810 is 0.9 degrees 2.4A 0.8 ohms
2.I use 24v 10A power supply
3.And TMC2130 ...with 1.2A max RMS and 2.5A Peak

I would expect if i use a bigger voltage 24v i will lower the amps used by the stepper. But if i go to ohms law... I=v/r = 24v/0.8 ohm = 30A is backwords. And something will blow up

And going to marlin to set up the rms current 
2.4A/1.41 = 1.702v = 1702 mv in marlin 

The question is... Im doing something wrong ? 
Somethig will blow up?

Kimd regards all

ritor1sefa, 2019-05-22:
stoicastefanv, 2130 is not the best choice for high loads.  Also, 2130 is very hot.
5160 is more profitable in terms of parameters (but also 2 times more expensive -_-).
https://github.com/teemuatlut/TMCStepper/issues/3

ritor1sefa, 2019-05-22:
Black6spdZ, In a mysterious way, everything worked for me. Mysteriously. With the chip, I did nothing, in the end, I did not solder my legs, I did not change anything.
The problem was somewhere in the firmware.
Thanks for the help.

Black6spdZ, 2019-05-22:
@stoicastefanv , current isn't calculated using simple ohms law with steppers.. the drivers are smart in that they use a closed loop feedback PWM to modulate the current to whatever you set either the pot or SPI software setting at regardless of input voltage

@ritor1sefa glad you got it going, I fought with mine for a couple days until I realized that there was a hardware setting on the BIQU 2130s for either standalone or SPI mode

ritor1sefa, 2019-05-22:
@Black6spdZ, Yes, I had to turn on TMC_USE_SW_SPI, which is strange, I think. Because I physically placed all the jumpers, why also programmatically include? -_-

Black6spdZ, 2019-05-22:
@ritor1sefa  oh yea, definitely need that enabled if using the 2130s in SPI mode and not standalone mode

Mbombace, 2019-06-01:
I have been messing the Bigtreetech SKR v1.3 for just about a week now trying to get Marlin 2.0 loaded up on this board, but it’s definitely not functioning properly and I will probably be returning it. First problem I am having is The board, when plugged in via usb to my computer is not recognized as a USB device. I’ve tried different drivers, removing the SD card and running that via usb, reformatting the SD, nothing works. So there’s that. But the main issue is that for some reason, Marlin 2.0 does not seem to like this processor NXP 1786. Using the Bigtreetech github version of Marlin 2.0 does not help. A freash version of Marlin 2.0 does not work, changing the Motherboard type within the Configuration.h file to the Bigtreetech SKR v1.3 board does not help. There is a compilation error no matter what I do with this board as the selected motherboard in Marlin 2.0. I've obviously referenced boards.h to ensure I am defining the correct board. If I change the motherboard to something different, the firmware.bin file compiles just fine! And before anybody says to downgrade to Marlin 1.9, this board is not supported is any previous versions of Marlin! I’ve tried everything. No dice. By the way- I am using VScode with the PlatformIO extension. Changing the environment within the Platform.ini file to the 1768 environment. No dice. I keep getting a bunch of errors. I can show you my error log if anybody is willing to help?

Grogyan, 2019-06-01:
One thing you can try is to upload the Smoothieware bootloader
You will need a USB to UART converter to do so

Mbombace, 2019-06-01:
Thanks but I had my sights set on Marlin. I just ordered a MKS Gen L v1.0 going to give that board a shot I guess.

stoicastefanv, 2019-06-01:
I can give you my drivers ... And my marlin file but how do i send you the files?  Watch out i have an windows 10 and when i wanted to install the working driver it told me that is not valid... It was needed to desactivate windows driver chech.. Shif + restart... You done something like this?

1 I have a question of my own.... Guys when homing did you have an issue to go in the other way ... Or go in other way and camming back ? 
2 anny one tried to install leds on skr ?

doggyfan, 2019-06-01:
The windows USB driver for the SKR 1.3 board is in ..

Marlin\src\HAL\HAL_LPC1768\win_usb_driver

[5 more replies]