MarlinFirmware/Marlin #13451: [BUG] MKS Gen 1.4 TMC2130 Printer halted when homing first time every day

BoKKeR, 2019-03-22:
### Description

My printer that is using MKS Gen 1.4 TMC2130 gets halted when homing the first time every day. It works always the second time. I tried turning off TMC2130 stop_on_error but it didnt help
### Steps to Reproduce

1. Turn on power
2. Set printer to home
3. Recv: echo:busy: processing 
4. Wait for printer to fail
5. Re-connect printer
6. Home again but successfully this time.

**Expected behavior:**
XY should home or give at least a description of what happened
**Actual behavior:**
1. Turn on power
2. Set printer to home
3. Recv: echo:busy: processing
4. Wait for printer to fail
[GIF](https://i.imgur.com/alD2FJr.mp4)
```Recv:  T:20.43 /0.00 B:21.72 /0.00 @:0 B@:0
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Operational" to "Error: Printer halted. kill() called!"
Changing monitoring state from "Error: Printer halted. kill() called!" to "Offline (Error: Printer halted. kill() called!)"
Connection closed, closing down monitor
```
5. Re-connect printer
6. Home again but successfully this time.
The printer works after I hit connect again for the rest of the day 
#### Additional Information

* I Included a ZIP file containing my `Configuration.h` and `Configuration_adv.h` files.
[marlin-config.zip](https://github.com/MarlinFirmware/Marlin/files/2995351/marlin-config.zip)
* I don't have any display connected
* EEPROM is disabled

BoKKeR, 2019-05-05:
I am still having this issue,

kAdonis, 2019-05-05:
This could be an issue with your powering sequence, the TMC2130 need to have 12V(24V) connected when Marlin boots up, Be sure to switch on your printer before you plug in USB (or cut the 5V line in your USB cable)

BoKKeR, 2019-06-04:
Thank you. It worked once I taped over the 5v pin of the USB cable!

closed 2019-06-04 by BoKKeR