MarlinFirmware/Marlin #13452: [BUG] Compiler Error for #define FAST_PWM_FAN on S

GeorgZiegler, 2019-03-22:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

GeorgZiegler, 2019-03-22:
Description
Compiling Marlin 2.0 with any Sanguino Board, e.g.   #define MOTHERBOARD BOARD_MELZI_TRONXY
or #define MOTHERBOARD BOARD_ANET_10 
results in a Compiler Error: 
sketch\src\module\temperature.cpp: In static member function 'static Temperature::Timer Temperature::get_pwm_timer(pin_t)':
sketch\src\module\temperature.cpp:1628:50: error: 'OCR3C' was not declared in this scope
               /*OCRnQ*/   { &OCR3A,   &OCR3B,   &OCR3C},          
                                                                          --------

I am not expert, but i think the issue comes because an AT Mega1284 doesn't have
Output Compare Register C.
So i came up with following code change in Temperature.cpp.
But also : I did work for me, but i am not 100% sure , what i did program ;-)
Also i know that this is not a genrous solution for all type of processors.
So i think, some if def this_cpu compile directives are missing.
        #ifdef TCCR3A
//          case TIMER3C:   q += 1;
//          case TIMER3B:   q += 1;
//          case TIMER3A: {
//            Temperature::Timer timer = {
//              /*TCCRnQ*/  { &TCCR3A,  &TCCR3B,  &TCCR3C},
//              /*OCRnQ*/   { &OCR3A,   &OCR3B,   &OCR3C},
//              /*ICRn*/      &ICR3,
//              /*n, q*/      3, q
//            };
//            return timer;
//			}
// G.Ziegler No Timer 3C for Sanguino ???
              case TIMER3B:   q += 1;
              case TIMER3A: {
                Temperature::Timer timer = {
                  /*TCCRnQ*/  { &TCCR3A,  &TCCR3B,  NULL},
                  /*OCRnQ*/   { (uint16_t*)&OCR3A, (uint16_t*)&OCR3B, NULL},
                  /*ICRn*/      NULL,
                                2, q
                };
                return timer;

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

sjasonsmith, 2019-09-21:
This is easily reproducible. Here is the error output for reference:
```
Marlin\src\HAL\HAL_AVR\fast_pwm.cpp: In function 'Timer get_pwm_timer(pin_t)':
Marlin\src\HAL\HAL_AVR\fast_pwm.cpp:100:46: error: 'OCR3C' was not declared in this scope
           /*OCRnQ*/   { &OCR3A,   &OCR3B,   &OCR3C},
                                              ^
```
And some configs:
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3638209/Configs.zip)

thinkyhead, 2019-09-23:
This should work…
```diff
-    #ifdef TCCR3A
+    #ifdef TCCR3C
       case TIMER3C:   ++q;
       case TIMER3B:   ++q;
       case TIMER3A: {
         Timer timer = {
-          /*TCCRnQ*/  { &TCCR3A,  &TCCR3B,  &TCCR3C},
-          /*OCRnQ*/   { &OCR3A,   &OCR3B,   &OCR3C},
+          /*TCCRnQ*/  { &TCCR3A,  &TCCR3B,  &TCCR3C },
+          /*OCRnQ*/   { &OCR3A,   &OCR3B,   &OCR3C },
+          /*ICRn*/      &ICR3,
+          /*n, q*/      3, q
+        };
+        return timer;
+      }
+    #elif defined(TCCR3B)
+      case TIMER3B:   ++q;
+      case TIMER3A: {
+        Timer timer = {
+          /*TCCRnQ*/  { &TCCR3A,  &TCCR3B,  nullptr },
+          /*OCRnQ*/   { &OCR3A,   &OCR3B,  nullptr },
           /*ICRn*/      &ICR3,
           /*n, q*/      3, q
         };
```

closed 2019-09-23 by thinkyhead