MarlinFirmware/Marlin #13453: Comment in configuration_adv.h in 2.0.x links to wrong TMC libraries

stahlfabrik, 2019-03-22:
### Description

I learned from @teemuatlut that the TMC library that Marlin 2.0 has changed over the one 1.1.9 used. So the Configuration_adv.h links to the two old Libraries. The problem is that the interface has changed, so many calls in tmc_adv() that used to work in 1.1.9 do not work in marlin 2.0.

So the links should be updated. Perhaps even the example will not compile? I have not tried the example:-)

Old links are
   * https://github.com/teemuatlut/TMC2130Stepper
   * https://github.com/teemuatlut/TMC2208Stepper

In Marlin 2.0 Configuration_adv.h it should be updated to:
https://github.com/teemuatlut/TMCStepper

bossman74, 2019-03-23:
Sorry for the stupid question, but can I do this myself? And if so, how do I update the links ?

InsanityAutomation, 2019-04-08:
@thinkyhead can go ahead and close this. Change is merged.

closed 2019-04-10 by thinkyhead