MarlinFirmware/Marlin #13455: BLTouch response Time On Z Axis

Kaisertron, 2019-03-22:
I can't figure out what is wrong. Ender 3 glass bed, real BLTouch, microswiss hotend. BLTouch is spaced properly and it clicks but doesn't stop the z. It runs it into the bed then after a second brings it back up. https://youtu.be/eAfLy4Y4R9M Link to my configuration files as well
https://drive.google.com/folderview?id=1--6tUwmaqAwArewMTryR3FCwnz01Rmei

BLTouch is spaced appropriately as base of bed 8mm above bed and within tolerance of switch activation.

ManuelMcLure, 2019-03-22:
How recent is this BLTouch? The new V3 BLTouch are known to have issues with Marlin - see #13345. If you received this recently you might have a V3.

InsanityAutomation, 2019-03-25:
As this is all handled in the other thread, please close the duplicate issue here.

Thanks

closed 2019-03-25 by Kaisertron