MarlinFirmware/Marlin #13456: E0 running in Spreadcycle on TMC2130 with Bugfix 2.0.x

bossman74, 2019-03-22:
I've decided to "refresh" my Marlin 2.0.x bugfix installation from last August on my modified CR10-S.

So I've downloaded the latest  2.0.x bugfix (from March 20th), copied my settings in the configurations and pins files and Flashed it.

Since then, my extruder (E0 stepper) is running in Spreadcycle mode, or I believe it is since it's very noisy.  All other steppers are running in Stealthchop and are quiet as can be.

Can you tell me what I've done wrong ?   Here are my `Configuration.h`, `Configuration_adv.h`, `pins_RAMPS.h` and `pins_MKS_GEN_L.h` files:
 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2997823/Marlin.zip)

### Description

Extruder (E0) running only in spreadcycle

### Steps to Reproduce

Every time I move the extruder axis in any direction

**Expected behavior:** [What you expect to happen]
Extruder (E0) to run in Stealthchop

**Actual behavior:** [What actually happens]
Extruder (E0) running only in spreadcycle

#### Additional Information
My printed is a CR-10s with a MKS Gen L,  paired with five TMC 2130 steppers hooked up with these wired connections (apart from the diag pins, still running the mechanical end stops for now)
[MKS_Gen-L_TMC2130_SPI_Sensor-less_Homing_Wiring_Diagram.pdf](https://github.com/MarlinFirmware/Marlin/files/2997780/MKS_Gen-L_TMC2130_SPI_Sensor-less_Homing_Wiring_Diagram.pdf).

I'm using a capacitive fixed probe mounted and I use the E1 driver for my Z2 stepper.

Ludy87, 2019-03-22:
try M502 and M500

bossman74, 2019-03-22:
Sadly that's not it, M502 and M500 are already include in my post flash commands in Octoprint's Firmware Updater.  But for good measure I manually sent the commands as well and it did not fix the issue,

gloomyandy, 2019-03-23:
Can't check your configs at the moment, so sorry for asking questions that could be answered by doing that. Are you using SPI control? If so enable TMC_DEBUG and post the output from M122. Also there have been reports that running the extruder in stealthchop can cause problems with lost steps etc. especially if running linear advance, so perhaps the noise you are getting is the result of missed steps?

teemuatlut, 2019-03-23:
* Post `M569` and/or `M122`
* Try disabling `LIN_ADVANCE`

bossman74, 2019-03-23:
I am running in SPI mode
`LIN_ADVANCE` is activated in Marlin, but the K factor is set at 0 so when moving the extruder manually, shouldn't it be deactivated ?  I didn't know that using stealthChop on the extruder could cause issues, the only time I noticed missed steps was if I tried to do a high acceleration and/or high speed (above 90mm/sec) retraction.

> * Post `M569` and/or `M122`

Send: M122
Recv: 		X	Y	Z	Z2	E
Recv: Enabled		false	false	false	false	false
Recv: Set current	750	750	750	750	750
Recv: RMS current	734	734	734	734	1325
Recv: MAX current	1035	1035	1035	1035	1868
Recv: Run current	23/31	23/31	23/31	23/31	23/31
Recv: Hold current	4/31	4/31	4/31	4/31	4/31
Recv: CS actual		31/31	31/31	31/31	31/31	0/31
Recv: PWM scale	255	255	255	255	0
Recv: vsense		1=.18	1=.18	1=.18	1=.18	0=.325
Recv: stealthChop	true	true	true	true	false
Recv: msteps		0	0	0	0	256
Recv: tstep		4294967295	4294967295	4294967295	4294967295	0
Recv: pwm
Recv: threshold		0	0	0	0	0
Recv: [mm/s]		-	-	-	-	-
Recv: OT prewarn	true	true	true	true	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false	false
Recv: off time		15	15	15	15	0
Recv: blank time	54	54	54	54	16
Recv: hysteresis
Recv: -end		12	12	12	12	-3
Recv: -start		8	8	8	8	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	Z2	E
Recv: stallguard		X	X	X	X
Recv: sg_result		1023	1023	1023	1023	0
Recv: fsactive		X	X	X	X
Recv: stst		X	X	X	X
Recv: olb		X	X	X	X
Recv: ola		X	X	X	X
Recv: s2gb		X	X	X	X
Recv: s2ga		X	X	X	X
Recv: otpw		X	X	X	X
Recv: ot		X	X	X	X
Recv: Driver registers:
Recv: 		X	0xFF:FF:FF:FF	 Bad response!
Recv: 		Y	0xFF:FF:FF:FF	 Bad response!
Recv: 		Z	0xFF:FF:FF:FF	 Bad response!
Recv: 		Z2	0xFF:FF:FF:FF	 Bad response!
Recv: 		E	0x00:00:00:00	 Bad response!
Recv: 
Recv: 
Recv: Testing X connection... Error: All HIGH
Recv: Testing Y connection... Error: All HIGH
Recv: Testing Z connection... Error: All HIGH
Recv: Testing Z2 connection... Error: All HIGH
Recv: Testing E connection... Error: All LOW
Recv: ok

Send: M569
Recv: X driver mode:	stealthChop
Recv: Y driver mode:	stealthChop
Recv: Z driver mode:	stealthChop
Recv: Z2 driver mode:	stealthChop
Recv: E driver mode:	spreadCycle
Recv: ok

gloomyandy, 2019-03-23:
Your SPI connection to the drivers is not working, see the bad responses and the errors at the end of the M122 output.

bossman74, 2019-03-23:
You mean the wires themselves, or is a configuration issue.  They worked with my previous version of marlin 2.0.  Could the connections have always been bad but the steppers worked in Stealthchop without me noticing?

gloomyandy, 2019-03-23:
No idea it could be the configuration or the wiring. But you are getting errors in the above output so something is not right.

bossman74, 2019-03-23:
So I decided to reflash my older version of 2.0.x from last August.  And now all steppers are whisper quiet.
But... when I send the M122 command, here's what I get.

Send: M122
Recv: 		X	Y	Z	Z2	E0
Recv: Enabled		false	false	false	false	false
Recv: Set current	750	750	750	750	750
Recv: RMS current	1325	1325	1325	1325	1325
Recv: MAX current	1868	1868	1868	1868	1868
Recv: Run current	23/31	23/31	23/31	23/31	23/31
Recv: Hold current	4/31	4/31	4/31	4/31	4/31
Recv: CS actual		0/31	0/31	0/31	0/31	0/31
Recv: PWM scale		0	0	0	0	0
Recv: vsense		0=.325	0=.325	0=.325	0=.325	0=.325
Recv: stealthChop	false	false	false	false	false
Recv: msteps		256	256	256	256	256
Recv: tstep		0	0	0	0	0
Recv: pwm
Recv: threshold		0	0	0	0	0
Recv: [mm/s]		-	-	-	-	-
Recv: OT prewarn	false	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false	false
Recv: off time		0	0	0	0	0
Recv: blank time	16	16	16	16	16
Recv: hysteresis
Recv: -end		-3	-3	-3	-3	-3
Recv: -start		1	1	1	1	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	Z2	E0
Recv: stallguard
Recv: sg_result		0	0	0	0	0
Recv: fsactive
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x00:00:00:00
Recv: 	Y = 0x00:00:00:00
Recv: 	Z = 0x00:00:00:00
Recv: 	Z2 = 0x00:00:00:00
Recv: 	E0 = 0x00:00:00:00
Recv: 
Recv: 
Recv: ok

This says that all my drivers are actually not running in Stealthchop, but how come they do no make any noise???

I'm just more confused now...

gloomyandy, 2019-03-23:
You can't trust the results from that m122 because it is still failing (the registers at the end are all zero). Looks like you need to fix what ever is wrong with your SPI setup. Are you sure that your driver boards are actually configured correctly for use with SPI?

bossman74, 2019-03-23:
I'm going to check my wires and use this wiring diagram instead : [https://github.com/teemuatlut/TMC2130Stepper/issues/28](url)

And will of course modify my pin assignment in Marlin.

Once that's done I'll post an update.

gloomyandy, 2019-03-23:
Before you make any changes I would check that your driver boards are correctly setup for use in SPI mode. They will often need a couple of solder bridges making on the board, but it very much depends on exactly which boards you are using, some come with the bridges in place, others do not (even though they may have the pins sticking up from the board).

bossman74, 2019-03-23:
Found the issue, I had one bad connection on the SDO wire on the last driver before connecting to the board pin.

I decided to use the reference wiring diagram anyway, just to be sure.

Now here's my `M122` and `M569` ouput:

Send: M122
Recv: 		X	Y	Z	Z2	E
Recv: Enabled		false	false	false	false	false
Recv: Set current	800	800	800	800	800
Recv: RMS current	795	795	795	795	795
Recv: MAX current	1121	1121	1121	1121	1121
Recv: Run current	25/31	25/31	25/31	25/31	25/31
Recv: Hold current	5/31	5/31	5/31	5/31	5/31
Recv: CS actual		5/31	5/31	5/31	5/31	5/31
Recv: PWM scale	0	0	0	0	0
Recv: vsense		1=.18	1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true	true
Recv: msteps		16	16	16	16	16
Recv: tstep		max	max	max	max	max
Recv: pwm
Recv: threshold		0	0	0	0	0
Recv: [mm/s]		-	-	-	-	-
Recv: OT prewarn	false	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false	false
Recv: off time		3	3	3	3	3
Recv: blank time	24	24	24	24	24
Recv: hysteresis
Recv: -end		-1	-1	-1	-1	-1
Recv: -start		1	1	1	1	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	Z2	E
Recv: stallguard
Recv: sg_result		0	0	0	0	0
Recv: fsactive
Recv: stst		X	X	X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:
Recv: 		X	0x80:05:00:00
Recv: 		Y	0x80:05:00:00
Recv: 		Z	0x80:05:00:00
Recv: 		Z2	0x80:05:00:00
Recv: 		E	0x80:05:00:00
Recv: 
Recv: 
Recv: Testing X connection... OK
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing Z2 connection... OK
Recv: Testing E connection... OK
Recv: ok
[...]

Send: M569
Recv: X driver mode:	stealthChop
Recv: Y driver mode:	stealthChop
Recv: Z driver mode:	stealthChop
Recv: Z2 driver mode:	stealthChop
Recv: E driver mode:	stealthChop
Recv: ok

BTW, the drivers I use are FYSETC TMC2130 V1.1 with the modifications already done to work in SPI.

I think I should be good to go, thank you all for your help in troubleshooting "my" mistake!
 ;-)

Next step will be to finally get around to using sensorless homing on the X and Y axis.

teemuatlut, 2019-03-23:
Please remember to close issues if the matter is solved.

closed 2019-03-23 by bossman74