MarlinFirmware/Marlin #13458: [BUG] [bugfix-2.0.x] Will not compile for GTM32_PRO_VB using Geeetech Rostock 301 example config

Phr3d13, 2019-03-23:
### Description

Compile failure - Arduino  IDE 1.8.8 - Marlin bugfix-2.0.x from 3.22.2019

### Steps to Reproduce

1. Download marlin bugfix-2.0.x branch
2. Copy files from Rostock 301 example directory
3. Open Marlin.ino
4. Set board as image below
5. Click verify.

**Expected behavior:** Compile success

**Actual behavior:** Compile failure
C:\Users\Phr3d\Documents\Arduino\libraries\IWatchdog\src\IWatchdog.cpp:2:31: fatal error: stm32yyxx_ll_iwdg.h: No such file or directory

#### Additional Information

[image]

Vertabreak, 2019-03-24:
missing library for iwatchdog
https://github.com/stm32duino/Arduino_Core_STM32/tree/master/libraries/IWatchdog

Vertabreak, 2019-03-27:
commenting out watchdog allowed compiler to get a little further but still results in more and more errors down the line i kept disabling problem features.

Vertabreak, 2019-03-27:
after disabling a bunch of features throwing errors im left with the following.

In file included from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL.h:42:0,
                 from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL.cpp:31:
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/fastio_STM32.h:34:8: error: 'GPIO_TypeDef' does not name a type
 extern GPIO_TypeDef * FastIOPortMap[];
        ^
In file included from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL.h:102:0,
                 from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL.cpp:31:
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.h:138:26: error: variable or field 'Step_Handler' declared void
 extern void Step_Handler(stimer_t *htim);
                          ^
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.h:138:26: error: 'stimer_t' was not declared in this scope
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.h:138:36: error: 'htim' was not declared in this scope
 extern void Step_Handler(stimer_t *htim);
                                    ^
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.h:139:26: error: variable or field 'Temp_Handler' declared void
 extern void Temp_Handler(stimer_t *htim);
                          ^
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.h:139:26: error: 'stimer_t' was not declared in this scope
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.h:139:36: error: 'htim' was not declared in this scope
 extern void Temp_Handler(stimer_t *htim);
                                    ^
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.h:147:9: error: 'stimer_t' does not name a type
 typedef stimer_t stm32f4_timer_t;
         ^
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.h:153:8: error: 'stm32f4_timer_t' does not name a type
 extern stm32f4_timer_t TimerHandle[];
        ^
In file included from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/../../inc/../HAL/HAL_STM32F1/HAL.h:53:0,
                 from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:30,
                 from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.h:29,
                 from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL.h:102,
                 from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL.cpp:31:
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/../../inc/../HAL/HAL_STM32F1/HAL_timers_STM32F1.h:73:24: error: 'timer_get_count' declared as an 'inline' variable
 #define TIMER_DEV(num) get_timer_dev(num)
                        ^
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/../../inc/../HAL/HAL_STM32F1/HAL_timers_STM32F1.h:84:56: note: in expansion of macro 'TIMER_DEV'
 #define HAL_timer_get_count(timer_num) timer_get_count(TIMER_DEV(timer_num))
                                                        ^
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.h:164:30: note: in expansion of macro 'HAL_timer_get_count'
 FORCE_INLINE static uint32_t HAL_timer_get_count(const uint8_t timer_num) {
                              ^
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/../../inc/../HAL/HAL_STM32F1/HAL_timers_STM32F1.h:73:24: error: 'uint32_t timer_get_count' redeclared as different kind of symbol
 #define TIMER_DEV(num) get_timer_dev(num)
                        ^
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/../../inc/../HAL/HAL_STM32F1/HAL_timers_STM32F1.h:84:56: note: in expansion of macro 'TIMER_DEV'
 #define HAL_timer_get_count(timer_num) timer_get_count(TIMER_DEV(timer_num))
                                                        ^
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.h:164:30: note: in expansion of macro 'HAL_timer_get_count'
 FORCE_INLINE static uint32_t HAL_timer_get_count(const uint8_t timer_num) {
                              ^
In file included from /home/vertabreaker/Arduino/hardware/Arduino_STM32/STM32F1/cores/maple/wirish_types.h:38:0,
                 from /home/vertabreaker/Arduino/hardware/Arduino_STM32/STM32F1/cores/maple/boards.h:38,
                 from /home/vertabreaker/Arduino/hardware/Arduino_STM32/STM32F1/cores/maple/wirish.h:54,
                 from /home/vertabreaker/Arduino/hardware/Arduino_STM32/STM32F1/cores/maple/Arduino.h:30,
                 from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL.h:33,
                 from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL.cpp:31:
/home/vertabreaker/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/include/libmaple/timer.h:688:22: note: previous declaration 'uint16 timer_get_count(timer_dev*)'
 static inline uint16 timer_get_count(timer_dev *dev) {
                      ^
In file included from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/../../inc/../HAL/HAL_STM32F1/HAL.h:53:0,
                 from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/../../inc/MarlinConfig.h:30,
                 from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.h:29,
                 from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL.h:102,
                 from /tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL.cpp:31:
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.h:164:50: error: expected primary-expression before 'const'
[... cut ...]
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.cpp:114:67: error: 'getTimerIrq' was not declared in this scope
   const uint32_t IRQ_Id = getTimerIrq(TimerHandle[timer_num].timer);
                                                                   ^
/tmp/arduino_build_238785/sketch/src/HAL/HAL_STM32/HAL_timers_STM32.cpp:115:10: error: 'NVIC' was not declared in this scope
   return NVIC->ISER[IRQ_Id >> 5] & _BV32(IRQ_Id & 0x1F);
          ^
Error compiling for board Generic STM32F103V series.

AluOne, 2019-03-27:
If I delete the directory HAL_STM32 I can compile for GTM32/Rostock 301.

Vertabreak, 2019-03-27:
deleted folder as suggested battled a few more errors and over size issues and finally got a compile.

Using library STM32ADC at version 1.0 in folder: /home/vertabreaker/Arduino/hardware/Arduino_STM32/STM32F1/libraries/STM32ADC 
Using library SPI at version 1.0 in folder: /home/vertabreaker/Arduino/hardware/Arduino_STM32/STM32F1/libraries/SPI 
/home/vertabreaker/.arduino15/packages/intorobot/tools/arm-none-eabi-gcc/4.9-2015q3/bin/arm-none-eabi-size -A /tmp/arduino_build_668438/Marlin.ino.elf
Sketch uses 129188 bytes (98%) of program storage space. Maximum is 131072 bytes.
Global variables use 13496 bytes (65%) of dynamic memory, leaving 6984 bytes for local variables. Maximum is 20480 bytes.

AluOne, 2019-03-27:
When you start the version, can you save/read settings in the EEPROM?
I always get a EEPROM version mismatch (EEPROM= Marlin=V63).

Vertabreak, 2019-03-27:
Have not flashed yet to my gtm32 board

Vertabreak, 2019-03-28:
My biggest issue right now is trying to get the screen working it compiled with the basic reprap display but there isn't any any to connect that display to the board it has some kind of touch screen but was not able to compile with anything but the reprap screen yet still messing with it

Phr3d13, 2019-05-17:
Tested this issue again today. Needed a little platformio.ini tweaking, and a tiny bit of configuration.h editing, but I have an almost useable build. All Axis steppers (x, y, z) and endstops work for my board/machine (Rostock 301 running a gtm32pro VB running the stm32f103vet6 chip). Temps appear to work, but are skewed/wrong (might need to find the right thermistor to define). So I can agree that with this build/board/machine, all that isn't working at all is SD read/write, at time of writing.
[Rostock 301 changed files.zip](https://github.com/MarlinFirmware/Marlin/files/3190444/Rostock.301.changed.files.zip)

InnerBushman, 2019-08-14:
> all that isn't working at all is SD read/write, at time of writing.

What about feedrate override? Are you able to change the feedrate with the encoder knob?

boelle, 2019-09-24:
@Phr3d13 is this still an issue?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle