MarlinFirmware/Marlin #13459: [BUG] [bugfix-2.0.x] Compile failure after the last revision of macros.h

mr-miky, 2019-03-23:
After the last revision of macros.h the compilation fails.

If I disable the checking section if the SS pin exists  in HAL_spi_STM32F4.cpp  the build ends successfully.

//  #if !PIN_EXISTS(SS)
//    #error SS_PIN not defined!
//  #endif

`````
../src/HAL/HAL_STM32F4/HAL_spi_STM32F4.cpp:86:1: error: operator "defined" requires an identifier
   #if !PIN_EXISTS(SS)
^ ~~~~~~
In file included from ../src/HAL/HAL_STM32F4/../shared/math_32bit.h:24:0,
                 from ../src/HAL/HAL_STM32F4/HAL.h:44,
                 from ../src/HAL/HAL_STM32F4/HAL_spi_STM32F4.cpp:39:
../src/HAL/HAL_STM32F4/fastio_STM32F4.h:72:33: error: missing binary operator before token "("
 #define _STM32_PIN(_PORT,_PIN) ((PORT##_PORT * 16) + _PIN)
                                 ^
../src/HAL/HAL_STM32F4/../shared/../../core/macros.h:174:38: note: in definition of macro '_PINEX_1'
 #define _PINEX_1(PN)        (defined(PN##_PIN) && PN##_PIN >= 0)
                                      ^~
../src/HAL/HAL_STM32F4/../shared/../../core/macros.h:149:28: note: in expansion of macro '_DO_1'
 #define __DO_N(W,C,N,V...) _DO_##N(W,C,V)
                            ^~~~
../src/HAL/HAL_STM32F4/../shared/../../core/macros.h:150:28: note: in expansion of macro '__DO_N'
 #define _DO_N(W,C,N,V...)  __DO_N(W,C,N,V)
                            ^~~~~~
../src/HAL/HAL_STM32F4/../shared/../../core/macros.h:151:28: note: in expansion of macro '_DO_N'
 #define DO(W,C,V...)       _DO_N(W,C,NUM_ARGS(V),V)
                            ^~~~~
../src/HAL/HAL_STM32F4/../shared/../../core/macros.h:175:29: note: in expansion of macro 'DO'
 #define PIN_EXISTS(V...)    DO(PINEX,&&,V)
                             ^~
../src/HAL/HAL_STM32F4/HAL_spi_STM32F4.cpp:86:8: note: in expansion of macro 'PIN_EXISTS'
   #if !PIN_EXISTS(SS)
        ^~~~~~~~~~
../src/HAL/HAL_STM32F4/fastio_STM32F4.h:89:14: note: in expansion of macro '_STM32_PIN'
 #define PA15 _STM32_PIN(A, 15)
              ^~~~~~~~~~
/home/michele/Arduino/hardware/STM32GENERIC/STM32/variants/BLACK_F407ZG/variant.h:14:14: note: in expansion of macro 'PA15'
 #define SS   PA15
              ^~~~
../src/HAL/HAL_STM32F4/HAL_spi_STM32F4.cpp:86:19: note: in expansion of macro 'SS'
   #if !PIN_EXISTS(SS)
                   ^~
````

InsanityAutomation, 2019-03-25:
FYI, same issue with SPINDLE_ENABLE pin in the sanity checks.

InsanityAutomation, 2019-04-08:
See #13614 - issue at least is known and fixed for one occurrence. Need to look at the definition of SS a bit more to see a good resolve there.

closed 2019-05-18 by thinkyhead