MarlinFirmware/Marlin #13460: Hotend PID Autotuning failure

rohitnair180, 2019-03-23:
Hi All, I am new to 3d printing. I have recently started on working on a reprap in which my primary task is to try tune my hotend temperature around 280-300deg C. As i am trying to autotune, i am getting a error "PID Autotune failed! timeout" 

Can anyone please help me with this. Do let me know if i need to provide additional details.
I am using a RTD sensor MAX31865.

TIA

kAdonis, 2019-03-23:
Post a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files as requested in the issue template
We need more details - does the hotend heat at all? Do you get the error at once or after a minute?
A video or a picture of your temperature graph (host program) would help

rohitnair180, 2019-03-23:
I tried to Autotune to a value of 280deg C, the temperature would rise uptill 250deg C and then all of a sudden "PID Autotune failed! timeout" pops up and the temperatures starts dipping. 
[Files.zip](https://github.com/MarlinFirmware/Marlin/files/2999343/Files.zip)

kAdonis, 2019-03-23:
Your hotend heater heats probably simply to slow, so that PID autotune reaches timeout
Preheat hotend to 240-250 deg C and then start PID Autotune and it should work,
Make sure your heater and your powersource are powerful enough to reach these high temperatures
One particular problem here is the 5A polyswitch fuse on the RAMPS as 5A might not be enough for a powerful heater

rohitnair180, 2019-03-23:
I tried preheating and then PID autotuning but, the temp would not go beyond 220 deg C. What can be done in this case?

Is there any alternative for 5A polyswitch fuse for RAMPS??

TIA

kAdonis, 2019-03-23:
> Is there any alternative for 5A polyswitch fuse for RAMPS??

I replaced the polyswitches on the RAMPS with wires and use automotive fuses in the cables from the PSU to the RAMPS power connector
something like this:
[image]
In your above temperature graph you reached more than 240 deg C
Try preheating to 220 deg and autotune to 240 deg C, use your new PID values and try again to reach higher temperatures
But you need probably a more powerful heater for 280 deg C, google for "heater cartridge 50W"

rohitnair180, 2019-03-25:
Thank you, I have reached 280 deg C by gradually increasing the temperature and PID Autotuning simultaneously. But, I have done this keeping my Thermal runaway function off for the hotend. Will this cause some issue?

kAdonis, 2019-03-25:
Thermal protection is a safety feature, it's okay to disable it for testing, but I would feel bad, if I had to print without it
try to increase THERMAL_PROTECTION_PERIOD

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle