MarlinFirmware/Marlin #13461: [BUG] [bugfix-2.0.x] PlatformIO and REPRAP_DISCOUNT_SMART_CONTROLLER

Bob-the-Kuhn, 2019-03-23:
This may be a duplication of issue #12937.

As mentioned in issue #13418, the display on a REPRAP_DISCOUNT_SMART_CONTROLLER works as expected if today's bugfix-2.0.x is compiled with Arduino but get only two rows of solid characters if compiled with PlatformIO.

STEPS TO REPRODUCE:
1. Grab bugfix-2.0.x
2. Set the below changes in **configuration.h**
3. Compile & download with PlatformIo within Atom.

Changes to **configuration.h**
- #define MOTHERBOARD BOARD_RAMPS_FD_V1
- #define REPRAP_DISCOUNT_SMART_CONTROLLER

It appears that the problem is Arduino is using **[Liquid Crystal 1.0.7](https://www.arduino.cc/en/Reference/LiquidCrystal)** while PlatformIo is using **[New Liquid Crystal 1.3.4](https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/Home)** (by F.Malpartida).  If I do the following then PlatformIO results in a working display:
- In **platformio.ini**, comment out **LiquidCrystal@1.3.4**
- Delete the directory **LiquidCrystal_ID136** under  **.piolibdeps**
- Within PlatformIO, install library **Liquid Crystal 1.0.7**

Restoring **New Liquid Crystal 1.3.4** brings the problem back.

I don't see any differences in the initialization sections of the WIKIs or the examples.

Miran81, 2019-03-23:
Yess !! Thanks a lot .. I was searching a mounth. I will try it this at tomorrow

Miran81, 2019-03-24:
Marlin 2.0.x   
where is LiquidCrystal_ID136
[platformio]
src_dir     = Marlin
build_dir   = .pioenvs
lib_dir     = .piolib
libdeps_dir = .piolibdeps
boards_dir  = buildroot/share/PlatformIO/boards
env_default = megaatmega2560

[common]
default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
build_flags = -fmax-errors=5
  -g
  -ggdb
lib_deps =
  https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip
  LiquidCrystal@1.3.4
  TMCStepper@<1.0.0
  Adafruit NeoPixel@1.1.3

Miran81, 2019-03-24:
@Bob-the-Kuhn  when I just delete LiquidCrystal_ID136 , platformio downloaded 1.3.4 version, because id:156. But 1.0.7 id :887 . I searched and I did not find this code. I deleted LiquidCrystal_ID136 and .platformio/lib /LiquidCrystal_ID887 copy and paste from .piolibdebs. My lcd is ok

closed 2019-05-18 by thinkyhead