MarlinFirmware/Marlin #13462: Wrong behavior of STRINGIFY macro in Bltouch commands menu

AndreaMZ, 2019-03-23:
Hi, i have compiled Marlin 2 , commit hash 144018736aa1e0d466e0c0693cf7992190e4981f dated 18/03/2019 - branch bugfix-2.0.x
Compiled with GCC 8.2.0 emitting AVR binary code.

in module menu_configuration.cpp there are BlTouch commands like this
**MENU_ITEM(gcode, MSG_BLTOUCH_RESET, PSTR("M280 P" STRINGIFY(Z_PROBE_SERVO_NR) " S" STRINGIFY(BLTOUCH_RESET)));**
The gcode string obtained is wrong , the second STRINGIFY expands to "BLTOUCH_RESET" instead of "160" so the string sent to command interpreter is "M280 P0 SBLTOUCH_RESET" instead of the correct "M280 P0 S160"

I have temporary fixed the line as:
**MENU_ITEM(gcode, MSG_BLTOUCH_RESET, PSTR("M280 P" STRINGIFY(Z_PROBE_SERVO_NR) " S160" ));**

best regards
amz

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

InsanityAutomation, 2019-03-24:
A pull request is already open with code to resolve this.

InsanityAutomation, 2019-03-25:
FYI this is fixed upstream, please verify and close this issue.

AndreaMZ, 2019-03-25:
ok

closed 2019-03-25 by AndreaMZ

AndreaMZ, 2019-03-25:
Further tests done : ok. bug closed
thank you for feedback