MarlinFirmware/Marlin #13465: E1 Doesn't move. Confirm my board defective.

forkoz, 2019-03-24:
I know I'm grasping at some hope here. Could there be anything I've missed?
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3000355/Marlin.zip)

E0 is bad so I put a driver in E1. No retraction or advance from the extruder motor from s3d or marlin interface.

Ludy87, 2019-03-24:
is your "hotend" heated?

forkoz, 2019-03-24:
Yep. I tried retracting when cold too and nothing.

forkoz, 2019-03-25:
Also tried with M302 P1

forkoz, 2019-03-25:
wooo, I fixed it.

I had to define E1 as E0... and now it moves correctly.
#define E0_DRIVER_TYPE LV8729
#define E0_STEP_PIN        P0_01
#define E0_DIR_PIN         P0_00
#define E0_ENABLE_PIN      P0_10
#ifndef E0_CS_PIN
  #define E0_CS_PIN        P1_01
#endif

Marlin won't work on E1 if there is no E0

closed 2019-03-25 by forkoz