MarlinFirmware/Marlin #13467: Arduino Due and secondary serial port

Roman-RV78, 2019-03-24:
Hello.
I'm trying to connect NodeMcu to Serial 3 arduino due, but I can't connect. 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3000442/Configuration.zip)
With Repetier-Firmware it is working.
In pins_RAMPS.h replaced
#define Y_MIN_PIN 14
#define Y_MAX_PIN 15
on
#define Y_MIN_PIN 7
#define Y_MAX_PIN 22
Made the switch Y_MIN Endstop and Y_MAX Endstop in Ramps 1.4, to pins 7 and 22 arduino Due
In Configuration.h
#define SERIAL_PORT 0
#define SERIAL_PORT_2 3
#define BAUDRATE 115200

PINS_DEBUGGING indicates that pin14 and pin15 <unused / unknown> Input = 1

Is it possible to solve the problem. thank
Translated using google

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

Roman-RV78, 2019-07-24:
[HAL.zip](https://github.com/MarlinFirmware/Marlin/files/3428529/HAL.zip)
Fixed by editing src \ HAL \ HAL_DUE \ HAL.h, just not sure whether it is true, but the port has earned

boelle, 2019-08-28:
ould you submit a PR to bugfix 2.0.x instead?

boelle, 2019-08-28:
it was not  a question, but a suggestion/request that you submit a PR instead of attaching file

boelle, 2019-08-28:
i tried to do it for you, but the hal file have changed since you made the zip file so could not do it

Roman-RV78, 2019-08-28:
I unfortunately do not quite understand the question )))

Roman-RV78, 2019-08-28:
I’m just not very oriented in this site and I think that I can’t do it

sjasonsmith, 2019-09-21:
Was this an error during compilation, or after building and flashing the firmware?
If it was during build then I suspect this was fixed back in #11988. I was able to build the posted configs after I removed/renamed the parts which are no longer valid.

I think for this to be actionable the report would need to be updated with new information based on the latest state of Marlin 2.0.

boelle, 2019-10-12:
@sjasonsmith i agree, people should always test with latest 2.0.x 

@Roman-RV78 tried the latest 2.0.x ?

Roman-RV78, 2019-10-12:
Doesn't work, you have to change HAL .h
[HAL.zip](https://github.com/MarlinFirmware/Marlin/files/3720826/HAL.zip)

boelle, 2019-10-12:
is the zip file the same you attached 24 jul?

boelle, 2019-10-12:
should be fixed with https://github.com/MarlinFirmware/Marlin/pull/15536

assuming i did the PR correct

Roman-RV78, 2019-10-12:
No, the HAL .h file in the zip archive, from today's build

boelle, 2019-10-12:
Great :-D

then the PR i made should work fine

thinkyhead, 2019-10-13:
NOTE: Github now allows anyone on Github to make changes in the "View File" screen and those edits are turned into a PR.

thinkyhead, 2019-10-13:
Patch applied!

closed 2019-10-13 by thinkyhead

Evg33, 2019-10-16:
DUE_USB broken
https://github.com/MarlinFirmware/Marlin/pull/15536#issuecomment-542654084

Roman-RV78, 2019-10-16:
Yes there is some mistake, we need to look at how it is implemented in HAL.h from LPC1768 and do it in a similar way

Evg33, 2019-10-16:
Marlin\src\HAL\HAL_LPC1768\HAL.h 
```#if SERIAL_PORT == -1
  #define MYSERIAL0 UsbSerial
...
  #if SERIAL_PORT_2 == -1
    #define MYSERIAL1 UsbSerial
```