MarlinFirmware/Marlin #13468: Error compiling "BOARD_TRIGORILLA_14" card

angelxulo, 2019-03-24:
After adding the card "`BOARD_TRIGORILLA_14`" in the line "`134`" (`#define MOTHERBOARD`) in "`Configuration.h`", I get the following compilation error:

"" * ""
```
Arduino: 1.8.9 (Windows 10), Card: "Arduino / Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch \ Marlin_main.cpp: In function 'extrapolate_one_point':

sketch \ Marlin_main.cpp: 2759: 3: internal compiler error: Segmentation fault

   }

   ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C: \ Program Files (x86) \ Arduino \ hardware \ tools \ avr / bin / avr-gcc returned 1 exit status

compilation terminated.

c: / program files (x86) / arduino / hardware / tools / avr / bin /../ lib / gcc / avr / 5.4.0 /../../../../ avr / bin / ld. exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for the Arduino / Genuino Mega or Mega 2560 card.

This report could contain more information with
"Show detailed output during compilation"
option enabled in File -> Preferences.
```
"" * ""

I do not understand why this may be happening ...
What I have noticed is that if I add the number equivalent to the "`BOARD_TRIGORILLA_14`" card that would be "`443`".
Being that way:

"" * ""
```
#ifndef MOTHERBOARD
  #define MOTHERBOARD 443
#endif
```
"" * ""

It does not give me any type of error, I would like there to be no error in the compilation of it.

Thank you!
And sorry for my English, I'm Spanish and I use 80% translator ...

Ludy87, 2019-03-24:
try again Bugfix-2.0.x

angelxulo, 2019-03-25:
I am using Marlin 1.1.9... 
I have not yet updated to 2.0.x. 
I have updated the 1.1.9 with the latest Bugfix 1.1.x

Stady234, 2019-03-26:
I just ran a successful compile... maybe try Arduino 1.8.8... i have not had any issues with that version...

boelle, 2019-07-17:
@angelxulo you tried to compile recently?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle