MarlinFirmware/Marlin #13469: [BUG] [bugfix-2.0.x] Dual Z Setup - E1 not reported by M122 (SKR 1.3 + TMC2208)

Yannik, 2019-03-24:
On a Dual-Z setup with a SKR1.3 and TMC2208's running the latest bugfix-2.0.x, the second Z stepper (E1) is not reported by the M122 command. X/Y/Z/E are reported by M122 correctly.

I have swapped the stepper driver to ensure that it is not the cause of this issue.
Also, the UART connection is of course installed correctly.

teemuatlut, 2019-03-25:
```cpp
#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2208
#define E1_DRIVER_TYPE TMC2208
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
```
You have Z2 disabled.

Yannik, 2019-03-25:
Fixed by https://github.com/MarlinFirmware/Marlin/commit/0d3ea59bcf515787310ee2b7cbe6aeeb494bab92.

closed 2019-03-25 by Yannik