MarlinFirmware/Marlin #13470: [BUG] [bugfix-2.0.x] Dual Z Setup - Z steppers run at different steps/mm (SKR 1.3 + TMC2208)

Yannik, 2019-03-24:
### Description

On a Dual-Z setup with a SKR1.3 and TMC2208's running the latest bugfix-2.0.x, the Z stepper runs with the correct steps/mm, while the second Z stepper (E1) runs at an incorrect (lower) steps/mm.

This may be related to #13469.
[Uploading config.zip…]()

kAdonis, 2019-03-24:
It **is** related to #13469 
you have defined Z_DUAL_STEPPER_DRIVERS
so you need to define
`#define Z2_DRIVER_TYPE TMC2208`
and comment
`//#define E1_DRIVER_TYPE TMC2208`

Yannik, 2019-03-24:
@kAdonis 
Okay, that's a bit confusing. I still have to use `INVERT_E1_DIR` to set the right direction, right?

The build is now running into an `TMC2208 on Z2 requires Z2_HARDWARE_SERIAL or Z2_SERIAL_(RX|TX)_PIN.` error.

I could manually set these constants in the board configuration, but shouldn't this be handled automatically (and therefore be fixed upstream?).

kAdonis, 2019-03-24:
>  Okay, that's a bit confusing. I still have to use INVERT_E1_DIR to set the right direction, right?

yes, it is 

I dont have a SKR1.3 and no TMC2208
I see no way to handle all possible configurations automatically
AFAIK the SKR1.3 is kept basic to be cheap, with only the pins broken out, the average user needs

Yannik, 2019-03-24:
@kAdonis The UART pins are defined for E1:

```
#define E1_SERIAL_TX_PIN P1_04
#define E1_SERIAL_RX_PIN P1_01
```

It works fine if I simply add

```
#define Z2_SERIAL_TX_PIN P1_04
#define Z2_SERIAL_RX_PIN P1_01
```

to the pins config.

I thought that this could be handled automatically, same as the motor pins are automatically set e1->z2 if dual z steppers are used.

gloomyandy, 2019-03-24:
The pins file for the SKR V1.3 is very new (as is the board) and came I think from bigtreetech, these sorts of things take time to iron out. It should probably have both sets of defines in the pins files, but to be honest it is not that unusual for you to have to make modifications to the pins files when using slightly non-standard configurations.

I suppose you could argue that the mapping should happen automatically when Marlin maps Z2 to E1, but it all adds to the (already complex) Marlin config. I don't think the "right" way to handle this has become established yet. Very few pins files currently have any TMC2208 configurations especially for software serial.

For now a PR that defines the Z2 values might help others avoid the problems.

forkoz, 2019-03-25:
They left out serial0 but its the same pins as mks sbase. Jumpers also had to be removed, not just moved to the side to set the correct stepping on LV8729.

thinkyhead, 2019-03-25:
> I thought that this could be handled automatically, same as the motor pins are automatically set e1->z2 if dual z steppers are used.

It probably could be. These are some new things and haven't been integrated at every level.

closed 2019-03-25 by thinkyhead

YMita2017, 2019-06-26:
Thanks, but seems reverse.
And can't change by #define E1_invert_dir.

YMita2017, 2019-06-26:
Find problem and locally fix it.
But problem of Z axis support triple stepper support left. (not fixed)

Configuration_adv.h
449 +   #define INVERT_Z2_VS_Z_DIR true   // Set 'true' if Z motors should rotate in opposite directions

L6470_Marlin.cpp
60 -   INVERT_Z_DIR                        ,  // 5 Z3
60 +   #if ENABLED(Z_DUAL_STEPPER_DRIVERS)
61 +    INVERT_Z_DIR ^ INVERT_Z2_VS_Z_DIR ,  // 5 Z2
62 +   #else
63 +    INVERT_Z_DIR                      ,  // 5 Z2
64 +   #endif

stepper.cpp
322 -   #define Z_APPLY_DIR(v,Q) do{ Z_DIR_WRITE(v); Z2_DIR_WRITE(v); }while(0)
322 +   #define Z_APPLY_DIR(v,Q) do{ Z_DIR_WRITE(v); Z2_DIR_WRITE((v) != INVERT_Z2_VS_Z_DIR); }while(0)

PR #13474 (2019-03-25): Add pins for dual z uart on skr 1.3 (fixes #13470)
### Requirements

Add pins for dual z uart on skr 1.3. This fixes #13470.
commit 0d3ea59b (2019-03-25): Add pins for Dual Z UART on SKR 1.3 (#13474)

Fix #13470