MarlinFirmware/Marlin #13480: BLTouch

jacobwtyler, 2019-03-25:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

reloxx13, 2019-03-25:
Start with posting your configiguration and use/try origin Marlin.

Bob-the-Kuhn, 2019-03-25:
You may have a V3 BLTouch.  We're having problems with them.  Take a look at #13454 and #13345.

Ragaroah, 2019-03-25:
Exact same issue I've been having @jacobwtyler. Marlin v2.0 (bugfix version) solved my issues, together with uncommenting:
#define BLTOUCH_V3

jacobwtyler, 2019-03-26:
Is the v2.0 bugfix the stable release at this point? I’m going to try it when I get home in an hour but I want to be sure and try the correct version. Pull up resistor worked but the mesh leveling isn’t nearly as accurate as my other ender 3 with a v2.1 BLTouch. Hopefully it’s just the pull-up resistor causing issues and the bug fix will be the answer. Thanks for your help guys.

jacobwtyler, 2019-03-26:
Yeah okay I’m going to expose my newbness but can anyone point me in the direction of the correct v2.0.x bug fix? I see @InsanityAutomation has a branch. Is that the one I should be trying?

jacobwtyler, 2019-03-26:
Okay I’ve got the v2.0.c bug fix zip downloaded and extracted but there are no example configurations included. Obviously this would be way too hard to keep up with and fix bugs but I’m not sure how much I have to edit now that would’ve been included in the creality cr10 example configuration.h file. Is it pretty much just bed size or is there way more to it?

InsanityAutomation, 2019-03-26:
@jacobwtyler can look at the CR10 branches here - https://github.com/InsanityAutomation/Marlin/tree/Creality_2.0_Devel

jacobwtyler, 2019-03-26:
Cool. I think I’ve got it set up and ready to flash. Just wondering why there are way less tabs in v2.0. Is there anything I’m missing or have they been combined somewhere else?

InsanityAutomation, 2019-03-26:
Arduino ide can't see subfolders.

jacobwtyler, 2019-03-26:
Everything flashed fine and I was able to remove the pull-up resistor from the BLTouch on my CR10 and it worked! Great fix!

psavva, 2019-04-01:
Suggest you close this issue :)

closed 2019-05-18 by thinkyhead