MarlinFirmware/Marlin #13483: Aurduino 1.8.7 don't compile Marlin Bugfix-1.1.x

patelmontu, 2019-03-25:
### Compile Error

In Marlin_main.cpp I get the following error:

```
sketch\Marlin_main.cpp: In function 'void set_axis_is_at_home(AxisEnum)':

Marlin_main.cpp:1565:1: error: expected primary-expression before '}' token

 }

 ^

exit status 1
expected primary-expression before '}' token

```
[Marlin-Bugfix-1.1.x.zip](https://github.com/MarlinFirmware/Marlin/files/3003265/Marlin-Bugfix-1.1.x.zip)

boelle, 2019-07-17:
does it compile with arudino 1.9 ?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

patelmontu, 2019-07-25:
Hello,

I try to compile in Aurduino 1.8.9 but I got same error,

And I just change in Configuration.h below,
#define MORGAN_SCARA   To   //#define MORGAN_SCARA
And it works.

Mayank,

boelle, 2019-07-25:
tried arduino ide 1.9 ?

patelmontu, 2019-07-25:
No.

boelle, 2019-07-25:
try it

shitcreek, 2019-09-17:
Cleaning house. LMK if this issue needs to be reopened.

closed 2019-09-17 by shitcreek