MarlinFirmware/Marlin #13484: [BUG] BABYSTEP_ZPROBE_OFFSET babysteps change only with positive values.

Knives79, 2019-03-25:
### Description

With BABYSTEP_ZPROBE_OFFSET, BABYSTEPPING and BABYSTEP_ZPROBE_GFX_OVERLAY enabled, if I double click the encoder I can adjust the Z Probe Offset only in one direction: when I go on positive values, the Z axis rise up, when I go on negative values, the LCD value change but the Z axis doesn't move and the stepper driver make like a false movement (a small movement in one direction and immediately in the other). 
### Steps to Reproduce
`#define BABYSTEPPING
#if ENABLED(BABYSTEPPING)
  #define BABYSTEP_WITHOUT_HOMING
  //#define BABYSTEP_XY                     // Also enable X/Y Babystepping. Not supported on DELTA!
  #define BABYSTEP_INVERT_Z false           // Change if Z babysteps should go the other way
  #define BABYSTEP_MULTIPLICATOR  20         // Babysteps are very small. Increase for faster motion.
  #define DOUBLECLICK_FOR_Z_BABYSTEPPING  // Double-click on the Status Screen for Z Babystepping.
  #if ENABLED(DOUBLECLICK_FOR_Z_BABYSTEPPING)
    #define DOUBLECLICK_MAX_INTERVAL 1250   // Maximum interval between clicks, in milliseconds.
                                            // Note: Extra time may be added to mitigate controller latency.
    #define BABYSTEP_ALWAYS_AVAILABLE     // Allow babystepping at all times (not just during movement).
    //#define MOVE_Z_WHEN_IDLE              // Jump to the move Z menu on doubleclick when printer is idle.
    #if ENABLED(MOVE_Z_WHEN_IDLE)
      #define MOVE_Z_IDLE_MULTIPLICATOR 1   // Multiply 1mm by this factor for the move step size.
    #endif
  #endif

  #define BABYSTEP_ZPROBE_OFFSET          // Combine M851 Z and Babystepping
  #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
    //#define BABYSTEP_HOTEND_Z_OFFSET      // For multiple hotends, babystep relative Z offsets
    #define BABYSTEP_ZPROBE_GFX_OVERLAY   // Enable graphical overlay on Z-offset editor
  #endif
#endif`
**Expected behavior:** The Z Probe Offset move the axis up AND down like the LCD values.

**Actual behavior:** The Z Probe Offset move the axis up BUT NOT down.

JupaCreations, 2019-03-28:
Did you make sure you are allowed in configuration.h to move Z below 0 (negative direction)?

Knives79, 2019-03-28:
Yes, I disable from menu before trying to move down. I attach my config and conf._adv.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3016772/Marlin.zip)

JupaCreations, 2019-04-11:
// Min software endstops constrain movement within minimum coordinate bounds
#define MIN_SOFTWARE_ENDSTOPS
#if ENABLED(MIN_SOFTWARE_ENDSTOPS)
  #define MIN_SOFTWARE_ENDSTOP_X
  #define MIN_SOFTWARE_ENDSTOP_Y
  //#define MIN_SOFTWARE_ENDSTOP_Z //comment this line to go past ZERO?
#endif

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@Knives79 do you still have issues?

closed 2019-10-12 by boelle

drphil3d, 2019-12-28:
I'm having the same exact issue with marlin 2.0.1 using an inductive probe. I noticed It will move in the negative direction but it only moves when the z is following the mesh leveling grid. I can move negative by -5 and only have moved down by .25mm.
I've tried numerous combinations of changes to no affect. 

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4007529/Marlin.zip)