MarlinFirmware/Marlin #13488: Printer stops responding after auto bed probing and during mesh fine-tuning

funbobby2001, 2019-03-25:
### Description
1.1.9
Printer becomes unresponsive shortly after UBL probing and has to be reset. Also happens after fine-tuning one or two mesh points. 

### Steps to Reproduce

1. M502
2. M500
3. M501
4. G28
5. G29 P1
6. G29 P3 T
7. G29 T
8. G29 S1
9. G29 F 10.0
10. G29 A
11. M500
12. After 30 seconds or so, or after a few jog moves the printer stops responding to all commands
13. Reset printer and reconnect
14. G26 C P T3.0  to print the test pattern (which looks like it didn't even try to level the bed btw)
15. G29 P4 T to start fine-tuning
16. After one or two points the printer stops responding again and has to be reset 

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3004559/Config.zip)

funbobby2001, 2019-03-25:
Switched to manual mesh leveling and it still happens after printing the pattern and while fine-tuning

funbobby2001, 2019-04-10:
Can anyone help with this?

ManuelMcLure, 2019-04-10:
A couple of possibilities:
- Running low on RAM. How much free memory shows after compiling? Should be the last few lines of the console when you finish compiling in Arduino.
- The host you're running the commands from is timing out and disconnecting. What host are you using?

thinkyhead, 2019-04-10:
Please test the `bugfix-2.0.x` branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

funbobby2001, 2019-04-11:
> A couple of possibilities:
> 
> * Running low on RAM. How much free memory shows after compiling? Should be the last few lines of the console when you finish compiling in Arduino.
> * The host you're running the commands from is timing out and disconnecting. What host are you using?

Hi Manuel. 
Sketch is using 149200 (58%) of 253952 bytes
I am using Simplify 3D.

funbobby2001, 2019-04-11:
Hi ThinkyHead, I will try bugfix 2.0... I assume I need to redo the config and config_adv from scratch when upgrading from 1.1.9?

boelle, 2019-07-17:
@funbobby2001 no just copy over the settings from old config to new

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

InsanityAutomation, 2019-07-23:
As an update to this, it is the S3D terminal that has issues with the response from G29T. This is not a Marlin issue.