MarlinFirmware/Marlin #13489: [BUG] Unable to compile for BOARD_MKS_ROBIN

Exquisitism, 2019-03-25:
### Description

Atom with Platformio is unable to compile for BOARD_MKS_ROBIN

### Steps to Reproduce

1. Try to compile with BOARD_MKS_ROBIN set as motherboard.

**Expected behavior:** To compile successfully

**Actual behavior:** Doesn't compile.

#### Compile log

`Processing STM32F1 (platform: ststm32@<4.4.0; board: genericSTM32F103RE; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option

CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html

PLATFORM: ST STM32 > STM32F103RE (64k RAM. 512k Flash)

HARDWARE: STM32F103RET6 72MHz 64KB RAM (512KB Flash)

DEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)

Converting Marlin.ino

Library Dependency Finder -> http://bit.ly/configure-pio-ldf

LDF MODES: FINDER(chain) COMPATIBILITY(soft)

Collected 35 compatible libraries

Scanning dependencies...

Dependency Graph

|-- <LiquidCrystal> 1.3.4

|   |-- <Wire> 1.0

|-- <TMCStepper> 0.3.1

|   |-- <SPI> 1.0

|-- <30aa480>

|   |-- <Wire> 1.0

|-- <Arduino-L6470> 0.7.0

|-- <SailfishLCD> #c8ac22f

|-- <SailfishRGB_LED> #2426fa2

|   |-- <Wire> 1.0

|-- <SlowSoftI2CMaster> #3a18be5

|-- <Wire> 1.0

|-- <SPI> 1.0

|-- <STM32ADC> 1.0

|-- <EEPROM>

Compiling .pioenvs\STM32F1\src\src\Marlin.cpp.o

[... 300 lines cut ...]
Environment STM32F1             	[ERROR]

Environment STM32F4             	[SKIP]

Environment ARMED               	[SKIP]

Environment mks_robin           	[SKIP]

Environment teensy35            	[SKIP]

Environment malyanm200          	[SKIP]

Environment esp32               	[SKIP]

Environment fysetc_f6_13        	[SKIP]

Environment linux_native        	[SKIP]

 [ERROR] Took 7.56 seconds 

Board name: BOARD_MKS_ROBIN
Build type: build
Environment used: STM32F1
2019-03-25 11:46:43.365000
`

dekra54, 2019-03-25:
Would be interesting to see a solution for this problem.
I have similar issues with the stm32h743 i'm working on.

Eno423, 2019-03-25:
I think all STMs are broken atm w/ the newest in platformio/.

some have said they were able to get it to run using ver 4.x of the STM, but i didnt have any luck

I am trying to compile for the Malyan/MP Select mini board & get a atomic.h or a boards.h error...

zillarob, 2019-03-25:
Watching like a hawk also.

The hacked up version of Marlin that MKS has locked down for Robin is just a crippled mess. And it looks like its an older version also.

[image]

pinchies, 2019-03-27:
To compile at the moment do these things:

Change the platform in "platformio.ini":
"**platform      = ststm32@5.1.0**"

Also in that file, insert a new line _after_ "build_flags   = !...":
" **-DSTM32_XL_DENSITY "**

Change the file "/Marlin/src/HAL/HAL_STM32F1/fastio_STM32F1.h", change the line:
"**#define PWM_PIN(P)              digitalPinHasPWM(P)**"
to
"**#define PWM_PIN(P)              1 //digitalPinHasPWM(P)**"

That should get you a working build 👍  Let me know how you go!

pinchies, 2019-03-27:
If you disable the U8glib library references, then following the above steps will also work on STSTM32 5.2.0, 5.2.0 and the latest STSTM32 build, which you can access with:
platform = https://github.com/platformio/platform-ststm32.git

jmz52, 2019-03-27:
@zillarob 
Firmware you were looking at is for new MKS Robin2 board based on STM32F407ZET6.
It will not run on MKS Robin v2.x boards based on STM32F103ZET6.
Original firmware for MKS Robin v2.x is based on Repetier firmware, not on Marlin.

zillarob, 2019-03-27:
@jmz52 
Robin Nano VET6 Iirc.

Been following your work and hoping that someday that board can run a legit fw.
I see the board your are working on has the same chip so hoping something in there is compatible and makes this a lil easier. New to the 32bit stuff and much of the behind the scenes stuff is just over my head =D

pinchies, 2019-03-28:
@zillarob VET6 should be well supported in the future. Both Geeetech's GTM32 and Alfawise/Longer3D Motherboards use VET6, both of those are currently actively work in progress.

zillarob, 2019-03-28:
Great to hear Sam! Thank you for the heads up =)

AlexanderAmelkin, 2019-03-29:
Why do we need the U8glib library at all? It appears that it's used whenever `REPRAP_DISCOUNT_SMART_CONTROLLER` is used in `Configuration.h`.
I thought U8glib was for matrix displays (TFTs and OLEDs) while RepRap Discount Smart Controller has a usual chargen HD44780-based display.

Anyway, that library is actively using C++ code in files ending with `.c` and `.h` which confuses VScode (and Atom as well), so they fail to compile complaining like `unknown type 'bool'` or `unknown type 'class'`. I'd say that's a bug in U8glib. So I'd prefer to completely get rid of that lib.

jmz52, 2019-03-29:
@AlexanderAmelkin 
U8glib is needed to display 12864 UI (upscaled  to 256x128) on MKS Robin TFT display.
If you are using 2004 display you can add u8glib to lib_ignore list.

Marlin uses patched version of u8glib that can be compiled for STM32.
https://github.com/MarlinFirmware/U8glib-HAL/tree/dev
Note: you might have to wait until PR https://github.com/MarlinFirmware/U8glib-HAL/pull/6 is merged.

closed 2019-03-29 by thinkyhead

jmz52, 2019-03-29:
@Exquisitism
One more thing - according to your logs you were using 'STM32F1' environment and 'genericSTM32F103RE' board.
MKS Robin has F103ZE MCU not the F103RE. F103RE does not have some of the pins used in pins_MKS_ROBIN.h file.
There is 'mks_robin' build environment pre-configured for MKS Robin board.

AlexanderAmelkin, 2019-04-11:
@jmz52, that's because I'm not building for MKS Robin :)
I'm trying (without much success so far) to port Marlin to Geeetech Rostock 301 (GTM32_PRO_VB).