MarlinFirmware/Marlin #13493: [BUG] [bugfix-2.0.x] BL-TOUCH servo pin doesn't work

TheMechanos, 2019-03-26:
Hello. I design my own motherboard (atmega2560). When I test bl-touch section i have a problem. Probe sensor pin is 67 (PK5), its work perfect. Problem is in serwo pin, it's 68 (PK6). I don't have any signal on this pin. I connect oscilloscope to this pin, there is only noise on GND level like this pin was not set to output.

Bob-the-Kuhn, 2019-03-26:
Marlin defaults the BLTouch to the first servo.  Unless you've explicitly configured it differently, your pins file should say **#define SERVO0_PIN 68**

Next use the **M42 I P68 S255** and **M42 I P68 S0** commands to toggle the pin.

Next use the M43 commands to see what logical pin is tied to that connector.  I like using these
commands:
- **M43 W I** - I use a 1K resistor to pull the pin to + 5V and then to ground.  75% of the time The logical pin number will pop up on the host display.
- **M43 T I** - If the above doesn't show anything then I tie an LED and resistor between the pin and ground and use this command to start toggling the outputs.  I start with the defaults, make note of the approximate pin number on the host screen when it flashes and then use the start and end pin numbers to narrow it down to the exact one.
- **M43 S I** - I use this one to verify that the BLTouch is working properly.

---

Curiosity - why design a custom controller?

TheMechanos, 2019-03-26:
Okej. I check and test it. 
Results:
-My pin configuration is ok, i have #define SERVO0_PIN 68
-I can operate pin68. I send "M42 I P68 S<x>"  
   a)x==0 I have continous GND on servo0_pin
   b)0<x<127 I have short 5v pick on servo0_pin
   c)127<x<255  I have continous 5v on servo0_pin
-When I set servo0_pin to 37(PC0) bl touch work correct.

Why, I don't know 😄 Maybe for satisfaction, I like design and make own PCB's.

boelle, 2019-07-21:
so is this a bug?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle