MarlinFirmware/Marlin #13494: [BUG] I think there's a problem with the Marlin firmware algorithm.

haoshuiwuxiang, 2019-03-26:
### Description

I noticed that the firmware called the kinematic inverse solution of the SCARA robot before calling the plan_buffer_line function, and then assigns the calculated three joint angle values to the plan_buffer_line function.After reading the plan_buffer_line function in detail, I feel that this is wrong. 
This is because all the computational algorithmic processes in the function are directed to Cartesian Cartesian coordinate space, for example:

block->millimeters = sqrt(square(delta_mm[X_HEAD]) + square(delta_mm[Y_HEAD]) + square(delta_mm[Z_AXIS]));

This is to calculate the spatial moving distance through the three target coordinates of Cartesian space. If it is replaced by three joint angle values, is this wrong?
The latter algorithm has similar problems, especially the Breasham algorithm in ISR, which is completely an algorithm of Cartesian space. If the value passed to the algorithm is three joint angle values, I think it should be incorrect.
I looked at the code for the marlin 1.1 version and there are similar issues.
I hope that my doubts can be answered, thank you!

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

haoshuiwuxiang, 2019-04-01:
no one replay?

AnHardt, 2019-04-01:
Please reread the code
```
block->millimeters = sqrt(square(delta_mm[X_HEAD]) + square(delta_mm[Y_HEAD]) + square(delta_mm[Z_AXIS]));

```
is only calculated for CORE_IS_XY. 
For the kinematic systems `_populate_block`, `_buffer_steps`, `buffer_segment` and 'buffer_line' are called with already set (different from 0) `millimeters` - from `prepare_kinematic_move_to`.
But i also see a suspiciously high amount of direct calls of `buffer_line()`, widely spread in the code - what could indicate a possibly wrong use of that.

Bresenham just handles a number of steps and their relation to each other. For short (sub-)segments, we hope, its linearity does not matter too much.

haoshuiwuxiang, 2019-04-02:
> Please reread the code
>
> ```
> block->millimeters = sqrt(square(delta_mm[X_HEAD]) + square(delta_mm[Y_HEAD]) + square(delta_mm[Z_AXIS]));
> ```
> is only calculated for CORE_IS_XY.
> For the kinematic [...]

Thank you for your reply.
The example is that I copied and pasted from the source code incorrectly.
But the contents of the plan_buffer_line I have studied carefully, all for the Cartesian Cartesian coordinate system. The non-Cartesian machine directly passes the three angle values obtained by inverse kinematics to the plan_buffer_line, which should be problematic.You mean that the bresenham algorithm is used here, each small segment is not a straight line, but because the line segment is small, it does not affect the result? But from a rigorous point of view, this will result in a non-Cartesian type printer that is less accurate than a Cartesian type machine.
I am also thinking about an algorithm that can replace bresenham in non-Cartesian machines.And I don't think I should do the inverse kinematics before the plan_buffer_line function. I am re-reading the code and finding a suitable place to do the inverse。  
I will be very grateful if I can get your support.Thank you! @AnHardt

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@haoshuiwuxiang is this still a problem?

closed 2019-10-12 by boelle