MarlinFirmware/Marlin #13496: [BUG] Incompatible options for babystepping

Stady234, 2019-03-26:
### Description

#error "BABYSTEP_ALWAYS_AVAILABLE and MOVE_Z_WHEN_IDLE are incompatible."

### Steps to Reproduce

Enabled both
    #define BABYSTEP_ALWAYS_AVAILABLE     // Allow babystepping at all times (not just during movement).
    #define MOVE_Z_WHEN_IDLE              // Jump to the move Z menu on doubleclick when printer is idle.

**Expected behavior:** Not error.
**Actual behavior:** compile error as stated in description.

#### Additional Information

I may have just missed the doc describing that they should not both be enabled.

forkoz, 2019-03-26:
They can't work together. Always available means it would do baby stepping when idle. You would need to enable move z when idle and plain babystepping.

kAdonis, 2019-03-26:
It's not a bug, it's a feature ;-)
It is probably not (yet) documented, but sanity check gives an error, to let you know the options are incompatible
It would be a bug, if there was no error

closed 2019-04-08 by Stady234