MarlinFirmware/Marlin #13497: [question] TMC2130 with SDSUPPORT

Alissonverd, 2019-03-26:
Dear friends.

Today I'am trying to install the TMC2130 (replacing A4988). But I found a difficulty that I can not solve.
If I enable SDSUPPORT, the driver dont work, and work without SDSUPPORT. What I need do?

Hardware:
MKS GEN L
TMC2130 SPI only in X and Y

Software:
  #define X_CS_PIN            59
  #define Y_CS_PIN            63

  #define TMC_USE_SW_SPI
  #define TMC_SW_MOSI       51
  #define TMC_SW_MISO       50
  #define TMC_SW_SCK        52

I need change the SPI pins?

Best regards,
Alisson

Ludy87, 2019-03-27:
You have to change the tmc's to SPI software, so there is the option.
The pins are freely selectable, then enter them at `TMC_SW_MOSI`, `TMC_SW_MISO`, `TMC_SW_SCK`.

7eggert, 2019-03-27:
That's what @Alissonverd did, but he's using the hardware SPI pins for software SPI. I suspect that by not using software but hardware SPI, it might work (Commenting out TMC_SW_* ?).

Bob-the-Kuhn, 2019-03-27:
Sharing the same SPI pins only works if both SPIs are soft or both hardware SPIs play nice together.

The Marlin SD code uses a hard SPI.  In most cases this means that the soft SPI can't control MOSI nor SCK.  

Yes, commenting out **TMC_USE_SW_SPI** may result in both being functional.  If one of them doesn't work then you'll have to enable **TMC_USE_SW_SPI** and then assign different pins.

igorepi, 2019-03-29:
Don't use software SPI with MKS Gen L. Gen L works flawlessly with hardware SPI, TMC2130 & SDSUPPORT (my Gen L/TMC2130 SPI is printing from SD card right now). So comment out TMC_USE_SW_SPI.

RuiCarneiro, 2019-03-30:
I had that problem, intermittently, I uncommented the option to use SPI half speed in Configuration.h and so far, no problems.

In my understanding, the cables for the LCD are too long and unshielded to use the same SPI speed as the TMC drivers, but my board has built in TMC2130's (Einsy RAMBo).

Alissonverd, 2019-03-31:
Thank you! 
With the answers the problem has solved.

Thank you all.

closed 2019-03-31 by Alissonverd

RuiCarneiro, 2019-04-13:
> Thank you!
> With the answers the problem has solved.
> 
> Thank you all.

It would be helpfull if you told how you solved.

Alissonverd, 2019-04-18:
@RuiCarneiro 

> Don't use software SPI with MKS Gen L. Gen L works flawlessly with hardware SPI, TMC2130 & SDSUPPORT (my Gen L/TMC2130 SPI is printing from SD card right now). So comment out TMC_USE_SW_SPI.