MarlinFirmware/Marlin #13503: digitalPinHasPWM function unavailalble on STM32F1

pinchies, 2019-03-27:
digitalPinHasPWM doesn't seem to work ... stops the builds working on STM32F1.

    platform = https://github.com/platformio/platform-ststm32.git
    framework = arduino
    board = genericSTM32F103ZE

Marlin\src\module\thermistor/../../inc/../HAL/HAL_STM32F1/fastio_STM32F1.h:52:33: error: 'digitalPinHasPWM' was not declared in this scope

pinchies, 2019-03-27:
Once this is fixed, it should resolve issue https://github.com/MarlinFirmware/Marlin/issues/13489

pinchies, 2019-03-27:
Example solution for this one, by programmers better than me:
https://github.com/jmz52/Marlin/commit/cc39d62ab608b4550fde4e3bd7b60ddba1f69bc3

ShohninDmitriy, 2019-03-27:
the entire stm32f1 line does not compile with this error

pinchies, 2019-03-27:
Ones that use software pwm will not fail, is that the only exception?

PR #13520 (2019-03-29): Replace digitalPinHasPWM with HAS_TIMER
Please see commit logs.

Fixes #13503
closed 2019-03-29 by thinkyhead