MarlinFirmware/Marlin #13505: [BUG] Extruder only moving in one way

pauhull, 2019-03-27:
I installed Marlin-1.1.x on my Ramps 1.4 board with A4988 stepper drivers set to 1/16 steps.
Extruding works fine, but if I try to retract the filament the stepper motor goes in the wrong direction and extrudes. I switched the stepper driver and it didn't make a difference, however if I connect my extruder motor to the x-Axis header the directions work fine. I already tried different Marlin versions and they had the same problem.

These are my configurations: https://www.mediafire.com/file/4hobn79bio9cozx/Configuration.zip/file

7eggert, 2019-03-27:
This sounds like a problem with the arduino or the connection to the DIR pin. Does it work if you use the E1 extruder socket?

Change in pins_RAMPS.h :
```
#define E0_STEP_PIN        36
#define E0_DIR_PIN         34
#define E0_ENABLE_PIN      30
#ifndef E0_CS_PIN
  #define E0_CS_PIN        44
#endif
```

Bob-the-Kuhn, 2019-03-27:
Might be that the A4988 current is too low.  Strange things happen when the current is too low.  I've seen this on my printer.

Try swapping the A4988s and see if the problem goes away.  If not then maybe the E0 socket has a problem.  @7eggert solution would be the fix in that case.

pauhull, 2019-03-28:
> This sounds like a problem with the arduino or the connection to the DIR pin. Does it work if you use the E1 extruder socket?
>
> Change in pins_RAMPS.h :
>
> ```
> #define E0_STEP_PIN        36
> #define [...]

Thank you, that worked perfectly!

closed 2019-03-28 by pauhull