MarlinFirmware/Marlin #13509: [BUG] Serial errors can cause Marlin to restart on 32-bit Archim

marcio-ao, 2019-03-27:
### Description

Under certain conditions, Marlin can crash and restart the printer. I have not been able to find a way to consistently reproduce this, but it seems to involve occasions where the host fails to read the serial port and may be related to some buffer filling up. This could either be a problem in Marlin or possibly the code that drives the USB serial emulation on the SAM chip.

In the case we encountered, we changed Cura to read the serial port more regularly and the problem went away. Creating this issue so there is a record of this, even though I suspect it is a low priority since the problem can be avoided by having the host read the serial buffer consistently. We have not seen any problems with un-tethered prints, so it seems isolated to an open serial USB link.

LulzBot tickets associated with this:
- https://code.alephobjects.com/T5686

gloomyandy, 2019-03-27:
See this issue: https://github.com/MarlinFirmware/Marlin/issues/11715
On the LPC1768 we added a timeout that will drop output if the output queue is full for longer than a few milliseconds (Which is sort of what happens with a real hardware port).

marcio-ao, 2019-03-27:
@gloomyandy: A good to know. I guess I'll close this since it has already been discussed elsewhere and it is unclear whether this is a bug or a feature.

closed 2019-03-27 by marcio-ao