MarlinFirmware/Marlin #13510: [BUG] Re-arm + TMC2130 M80 problem

Ilia-SB, 2019-03-27:
### Description

Turning PSU off then on using M81/M80 results in unresponsive TMC2130 drivers

### Steps to Reproduce

1. Turn the printer on
2. Check that all axis work
3. Turn the PSU off with M81
4. Turn the PSU on with M80
5. Try to move any axis driven my TMC2130 - no movement

**Expected behavior:** Axis moving normally

**Actual behavior:** Axis  not moving

#### Additional Information

I am using a Re-arm board with RAMPS 1.6+ and TMC 2130 in SPI mode for X/Y/Z axis, E0 is driven by A4988. The TMC's are wired externally, (not using the RAMPS 1.6+ pins) and Marlin is configured to use the software SPI mode. Re-arm is powered by USB (jumper is in USB position, USB is connected to raspberry pi, barrel power connector not used), RAMPS has D1 diode removed and is powered by a 24v PSU. The system runs well, but whenever I power the PSU down and then back up, the system becomes unusable because the TMCs do not respond. The only way to fix that is to reboot the board. Extruder controlled by A4988 keeps running just fine after PSU off/on, heaters work too. M122 output shows no problem both before turnig the power off and after it.
I suspect this is related to the way I power my system. I know that when there is no VMOT voltage the TMCs won't even respond to SPI (M122 would report failure. All low or all high, sorry, can't remember). Now when I turn the PSU off, 24v is gone from VMOT pin. Can it be that the TMCs need to be reinitialized somehow when VMOT is back? Sorry, I have no idea how they work. Would gladly do more debugging if needed.
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3014859/Config.zip)

kAdonis, 2019-03-27:
Yes, the TMCs get reconfigured when power is restored. Are you using a relay?
try to increase delay in power.cpp line 114

```cpp
    #if HAS_TRINAMIC
      delay(100); // Wait for power to settle
      restore_stepper_drivers();
```

gloomyandy, 2019-03-27:
TMCs require that VMOT is present before VCC_IO, so you need to ensure that this is the case. The SilentStepStick Protector claims to do this by using a diode between VCC_IO and VMOT, so some of those may help you. You could also try adding the diode yourself (I think it need to be a schottky type). 
https://learn.watterott.com/silentstepstick/protector/

Ilia-SB, 2019-03-31:
> Yes, the TMCs get reconfigured when power is restored. Are you using a relay?
> try to increase delay in power.cpp line 114
> 
> ```c++
>     #if HAS_TRINAMIC
>       delay(100); // Wait for power to settle
>       restore_stepper_drivers();
> ```

Sorry for not replying earlier. Thanks for advices.
Yes, I'm using a solid state relay. I followed your advice and increasing the delay to 150ms seems to do the trick. Well, mostly, because sometimes the stepper still don't respond. Higher delay values lead to unresponsive TMCs too. So I'm staying with 150ms so far.
@gloomyandy, may be my chineese TMCs have a slightly different schematic, because they do work. Anyway, thanks for the hint, In fact I've ordered the protectors a month ago, still waiting for them to arrive. Will see how it goes once I have them installed.

boelle, 2019-07-17:
@Ilia-SB are you still having issues?

Ilia-SB, 2019-07-17:
Mostly yes. It's very inconsistent, works sometimes, but not often. I'm now
using Silent Step Stick Protectors from Big Tree Tech and Marlin bugfix
from about a week ago. I still have increased delay in power.cpp at 200ms
(if I remember correctly, increasing it further produced worse results).
I actually have up on the issue. I'll try the latest bugfix and will play
with delay once again to see if things have changed.

shitcreek, 2019-09-17:
Cleaning house. LMK if this issue needs to be reopened.

closed 2019-09-17 by shitcreek