MarlinFirmware/Marlin #13512: [question]

guestisp, 2019-03-27:
Hi to all.
I'm trying to use a genuine BLTouch sensor (I think v2.1) but without luck
The mesh is build properly, auto-home is working as expected, but when running a print, the nozzle is lifter too much even where there is no need to lift it and nothing stick to th bed.

If I disable the mesh and the autolevelling, with the same glass and the same manual leveling, i'm able to print properly. So, there is something strange with the UBL or with the auto bed levelling, because enabling it make more issues than without.

Any idea ? I've tried 2 different bltouch (both genuine) on 2 different printers, same behaviour, the nozzle is lifted and lowered with no apparent reason and for too much distance (even 1-2mm)

A glass can't have a 1mm "hole" in a couple of centimeters, it's physically impossible and that would be visible by eyes.

Bob-the-Kuhn, 2019-03-27:
If you have a system that works well then stick with it.

If it has problems then take a look at the following.  I prefer the UBL system myself.

[UBL](http://marlinfw.org/docs/features/unified_bed_leveling.html)

[Mesh leveling](http://marlinfw.org/docs/gcode/G029-mbl.html)

guestisp, 2019-03-28:
The working system is the manual bed levelling, that doesnt produce good first layers, obviously, as it doesnt level in the center but only in the corners

Already saw these guide, they are what i've followed to enable the UBL

reloxx13, 2019-03-28:
post ur startcode.
describe ur leveling procedure.

guestisp, 2019-03-28:
> upload configuration.h and configuration_adv.h
As soon as I go to office, give me 30 minutes or less

> which marlin version u use?
2.0.0-bugfix

> post ur startcode.

```
M190 S60 ; set bed temperature and wait for it to be reached
M104 S205 ; set temperature
M412 S0;
G28 ; Home extruder
G29 L1 ; Load the mesh stored in slot 1 (from G29 S1)
G29 J   ; No size specified on the J option tells G29 to probe the specified 3 points and tilt the mesh according to what it finds.
; Filament gcode
M109 S205 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
;LAYER:0
G1 Z0.300 F7800.000
```

> describe ur leveling procedure.

z-offset set as 0
G28
Z is set to 10
I move Z axis down until touching the paper sheet
Read the Z value from the info screen (ie: -2.3)
Set z-offset to -2.3
Store settings
G28
Z is set to 12.3
Move Z to 0
The nozzle touch the paper sheet as expected

> and fix the title of this issue.
Sorry for the title, i didn't see that the title was wrote properly, I was on mobile.

Bob-the-Kuhn, 2019-03-29:
When I enabled UBL my first matix was terrible.  I did the G29 P1 to get the initial matrix but after that I kept repeating the "G26 , G29 P4" loop until I was getting good first layer adhesion everywhere..  After that I used G29 P6 to shift everything up/down based on test prints.

guestisp, 2019-03-29:
I can try
What seems strange to me is that even by setting a z-offset much lower than needed, in example -3.50 when the proper one is -2.50, the nozzle is still too high

Is zoffset ignored when using the ubl mesh? I have to rebuild the mesh every time i change the Z offset?

guestisp, 2019-03-29:
Side question: with UBL enabled, can I temporary disable the mesh and just use a plain simple 3Point level or a LINEAR leveling with 4 or more points, without the mesh correction ?

Roxy-3D, 2019-03-29:
> I can try
> What seems strange to me is that even by setting a z-offset much lower than needed, in example -3.50 when the proper one is -2.50, the nozzle is still too high

> Is zoffset ignored when using the ubl mesh? I have to rebuild the mesh every time i change the Z offset?

Usually...   This behavior is because a person is changing the Z-Offset in the firmware (configuration.h) and not in the EEPROM.    Are you doing a M502 & M500 after changing the firmware?

> Side question: with UBL enabled, can I temporary disable the mesh and just use a plain simple 3Point level or a LINEAR leveling with 4 or more points, without the mesh correction ?

Yes.    You can do a G29 P0 to zero the mesh.    And then a G29 J to do a 3-point leveling.   You can do a G29 J 2   for a 2 x 2 grid of probe points.

guestisp, 2019-03-29:
> Usually... This behavior is because a person is changing the Z-Offset in the firmware (configuration.h) and not in the EEPROM. Are you doing a M502 & M500 after changing the firmware?

Yes. But when I'm talking about changing the z-offset, i'm referring to changing it from the LCD and the save (still from LCD)

> Yes. You can do a G29 P0 to zero the mesh. And then a G29 J to do a 3-point leveling. You can do a G29 J 2 for a 2 x 2 grid of probe points.

Thank you. This should disable the mesh and activate the 3Point or the Linear. What if I would like to try the BILINER? Just add a number of points greated than 2 to "G29 J" will enable BILINEAR or still LINEAR with multiple points ?

Vertabreak, 2019-03-29:
if you did not remove the original z end stop you will get false triggers from it as the probe and the end stop run on the same z min pin, if the original z end stop is removed it will allow the probe to function without being false triggered and blocked by the original z end stop. this is the only thing i can think of that would result in what you describe and has also been documented in another post here on GitHub.

guestisp, 2019-03-29:
I don't have the endstop.

Currently, I'm able to print a decent first layer, just trying to figure out the source of some strangeness. In example, does make sense, with chinese printers, to have z-offset with 2 decimal? The printer is able to move Z to 0.10 and 0.15 differently ? Asking this because from the motion menu I can move just up to 1 decimal place, but the z-offset has 2 decimal places.

Vertabreak, 2019-03-29:
the lcd menu motion moves are limited if you where to issue the gcode move command manually from a terminal you could type in a more exact number.

guestisp, 2019-03-29:
Sure, but does it make sense to adjust zoffset at 2 decimal place precision ? I don't think any FDM printer would be able to reach a 10 micron resolution on Z. Maybe i'm wrong

Vertabreak, 2019-03-29:
i have never used more then 1 decimal place setting my z offset personally but others claim they must.

guestisp, 2019-03-29:
So, if you can print at .06, some printers are able to print at that resolution.

My correct z-offset is -2.40, if I set -2.40 first layer seems to be too hight, if I set -2.50 seems to be too low. It seems that Marlin is correcting too much. The paper sheet at -2.50 offset is totally fixed, doesn't move at all, this usually means a too low nozzle.

This is what I'm trying to figure out. Why setting a -2.40 offset (that is correct) Marlin doesn't stay too hight and with -2.50 is too low? The mesh seems is correct

Vertabreak, 2019-03-29:
the offset moves the entire mesh up or down if your trying to correct 1 spot on the bed the offset will not do that.

guestisp, 2019-03-29:
Is not one spot. I have one spot a correct and one spot (a couple of centimeter on the right) too low.
Adjusting the zoffset a little bit should solve, the correct post would be a little bit higher with not apparent drawback and the too-low spot will be corrected. But doesn't seem to work. Currently I get a good, but not perfect, first layer in the center of the bed.

I'll try to print near the edge, if i'm able to print properly even there, everything would be ok. Not perfect but still good is something acceptable for my prints. When I have more spare time i'll do a mesh edit point by point

I'll keep you updated in about 30 minutes

Bob-the-Kuhn, 2019-03-29:
There's a better explanation of the G29 options/function in the file **ubl_G29.cpp**

I use G29 P4 to edit individual points in the mesh.  That should help with your "point X is OK but point Y isn't" problem.

I have a glass bed so I just circle the problem area with a dry erase marker, use G29 P4  and make an adjustment when the nozzle is over the problem area.

One possible cause of the "point X is OK but point Y isn't" problem is too few points in the matrix.  On my machine a 5x5 matrix is OK but I see others that use 7x7 or even 9x9.

guestisp, 2019-03-29:
I have a 7*7 Matrix. I can try a 8*8 or 9*9
Isn't an issue for the available memory?  I get some warnings during the compile

Vertabreak, 2019-03-29:
Can use all but the last 1kb ram it won't compile with less warnings don't mean much.
max ubl grid is 15x15 and takes hours to build a mesh.
Can disable other features to make space.

guestisp, 2019-03-29:
Build time is not an issue, i have to do this just one time . If the memory warning can be safely ignored, i could try with a 9 * 9 mesh. There will be 32 additional points

Vertabreak, 2019-03-29:
I tested all the way up to 15x15 then scaled back till it wasn't accurate enough I can get away with 3x3 on my bed if I wanted my builds usually have 7x7 due to community demand.

Bob-the-Kuhn, 2019-03-29:
7x7 is probably OK.

Have you tried G29 P4?

guestisp, 2019-03-29:
> 7x7 is probably OK.

Already using 7x7

> Have you tried G29 P4?

Not yet.

guestisp, 2019-03-29:
> Could even print on a 2x2 it wasn't as accurate but was useable.

Well, yes, I'm able to print with a simple 3 points leveling, but as i have a sensor probe, i would like to get the first layer as accurate as possible.

Bob-the-Kuhn, 2019-03-29:
G29 P1 with my BLTouch has never produced a useable matrix for me. Here's how I tune my matrix:
1. Move the head over the print bed & Issue G29 A and the G29 P1
2. Issue G29 S2 to save the matrix
3. Move the head over the print bed & Issue G26 command to print out a test pattern,
4. Using a dry erase marker, circle the problem areas 
5. Move the head near a problem area and the issue G29 P4 R10
6. The head moves to the nearest point in the matrix.  If that isn't a problem point then just press the knob (without turning it).  The head will move to the next point in the matrix (it goes in a circle around the original point).  Once you're at a problem point turn the knob.  I usually do 0.050 or 0.025 changes.
7. Repeat 5 and 6 until all problem areas have been adjusted
8. Issue G29 Sx where x is a new slot or one you don't mind writing over.
9. Clean off the print bed
10. Repeat 3-9 until happy
11. Try a real print
12. If there's problems with the first layer issue G29 P6 Hxxx where xxx is the amount you want to move the head up or down.  
13 Issue G29 Sx where x is a new slot or one you don't mind writing over.
14. Repeat 11-13 until happy
15. Issue M500 to make the last saved matrix the default.

The reason why I put the head over the bed before issuing a command is my head can go beyond the bed and some settings of the matrix result in the head being too low and hitting the edge.

Bob-the-Kuhn, 2019-03-29:
Rather than doing 12 & 13 you can use babystepping to adjust the head height during a real print.  

For me babystepping doesn't work about half the time.  Wonderful when it works.

Vertabreak, 2019-03-29:
Calling it auto bed leveling is rather deceptive honestly. makes it sound like it will do it all perfect and that isn't the whole truth.

Roxy-3D, 2019-03-29:
> What if I would like to try the BILINER? Just add a number of points greated than 2 to "G29 J" will enable BILINEAR or still LINEAR with multiple points ?

If you want to use Bi-Linear...   You need to change your Configuration.h file to turn it on and rebuild the firmware.    There are slight differences in how the Z-Height correction is calculated within a mesh cell.   But you would be hard pressed to actually see a difference in the print.

Vertabreak, 2019-03-29:
It's possible to setup bilinear to save the mesh also but I have far greater control and success with ubl. Can't edit the mesh with bilinear only build new.

guestisp, 2019-03-30:
Up to which level the UBL will correct the Z ?
I'm at 8.5 in a test print (benchy) and i can see the Z axis moving up and down in the same level. Is the UBL that makes (or try to) the Z distance from the bed flat and consistent on the whole surface?

Bob-the-Kuhn, 2019-03-30:
Yes, UBL is what moves the head up & down to keep the head the same height from the surface.

There's an **ENABLE_LEVELING_FADE_HEIGHT** option that will gradually eliminate the compensation being done by UBL.  That's the way most people run.  I don't know if it's on by default.

guestisp, 2019-03-30:
I've checked. Fade height is disabled here, so why z axis was moving up and down ?

Bob-the-Kuhn, 2019-03-30:
If you enable it then that'll gradually eliminate the compensation.  I think 10mm is the usual setting at which the compensation is totally gone.

guestisp, 2019-03-30:
It's set to off.

Roxy-3D, 2019-03-30:
> It's possible to setup bilinear to save the mesh also but I have far greater control and success with ubl. Can't edit the mesh with bilinear only build new.

It is possible to edit the Bi-Linear mesh's.    But you need to do it via the M421 command.   We really should make the interactive mesh editing available (as an option) for Bi-Linear also.   It would not be that difficult to extract the interactive mesh editing code and allow it to be compiled in for either bed leveling system.

Roxy-3D, 2019-04-01:
> Can't edit the mesh with bilinear only build new.

This is not true.    It is true you can't do the interactive mesh editing that UBL has.   But you can display your Bi-Linear mesh and then use M421 to move those mesh points up or down.

guestisp, 2019-04-04:
I've doubled checked and my fade height is disabled but I can still see the Z axis going up and down, even at layer 150 (about 30mm). How can I totally disable this compensation ? I would like to try the compensation just on the first layer.

AnHardt, 2019-04-04:
If you disable fade height - Marlin will not fade out the correction, but will always apply the full amount.
Better set fade_height to about 1 to 3 layers height.
When the first layer is fully compensated but the second if fully **not** compensated, you will likely see a adhesion problem between the first and second layer - about the same magnitude you'd have between bed and first layer when bed-leveling is off

guestisp, 2019-04-04:
Ok, so is normal that at layer 150-170 marlin still try to compensate the bed level.... It does to keep the height uniform across the whole print, right ?

spilz87, 2019-06-29:
Did you solve your original issue ?
I have the same (if I understand well) : 
If I do a homing G28 my nozzle is just on the bed, but after a G29 the nozzle is no more on the bed. I find a way to solved it by comment this line : https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/gcode/bedlevel/abl/G29.cpp#L959

Let us know

guestisp, 2019-06-29:
I've changed many config in these days (including moving from UBL to a simple 3points) but i don't think the original issue is solved

boelle, 2019-07-17:
@guestisp could you change the title from [question] to [BUG]?

guestisp, 2019-07-17:
Sure but i'm not really sure it's a bug or not

boelle, 2019-07-17:
put configuration.h and configuration_adv.h in a zip file and attach here
but make sure the hardware side of your printer is solid

to me it sounds strange that you get different results

guestisp, 2019-07-18:
Attached the Configuration.h
The _adv is almost untouched from the stock one and as far as I know, there isn't anything related to the bed levelling

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/3407252/Configuration.h.txt)

guestisp, 2019-07-18:
What I know for sure (but I don't know why) is that any ABL without mesh, in example the 3POINTS and LINEAR, does the G29 properly, but during the print, it doesn't correct.

If I enable a mesh-based levelling, without any additional changes to the configuration, the correction is done properly

guestisp, 2019-07-26:
Ok, so, let's see if this is a bug or not.

With a 3POINTS or LINEAR bed levelling, during a print, the firmware is supposed to correct the levelling, right ? Thus moving the Z axis up and down like with any mesh-based levelling.

Is this correct ? Because i've tried in almost every possible way, with 3POINTS or LINEAR levelling, there is no correction at all. If I change to any mesh-based levelling, a correction is made properly.

boelle, 2019-07-26:
i use bi-linear with 3 probes on each axis, this is the best i can do for the mk42 bed on my mendel90 printer

with that i have correction too

i think we can think of 3 point as a tilt correction only, ie the printer assumes that the bed is totally flat but tilted, so if the bed is bent you will have an issue

not sure how linear work

but with mesh you have more probe points and hence the printer can correct more

guestisp, 2019-07-26:
bi-linear is making a mesh, and is working.
3points doesn't make any mesh, but just fix the tilt. My glass is flat, but what I see is that it doesn't correct any tilt. If I put the right side lower than the left side, the firmware doesn't correct that. the Z axis doesn't move at all.

boelle, 2019-07-26:
maybe @Roxy-3D can shine some light on that?

lovethat777, 2019-08-03:
#14825 Got the same issue

Virtualight, 2019-08-25:
I have the same issue with Bugfix 2.0 (2 weeks ago). Bed Leveling done and saved (M500) but z motors do not move.

lovethat777, 2019-08-25:
Did you enable the bed leveling with  `M420 S1` ? I found this setting was reset for some reason.

guestisp, 2019-08-26:
> Did you enable the bed leveling with M420 S1 ? 

Yes

Virtualight, 2019-08-28:
@lovethat777 - I think I have fixed it - not sure how. I have started making a sketch of how all of the z-axis functions work as I was getting really confused. If anyone on this thread is interested in contributing please visit this thread; https://github.com/MarlinFirmware/Marlin/issues/5995 

[image]

guestisp, 2019-08-29:
Solved how ? Still doesn't work here.
Any LINEAR (3POINTS or LINEAR) levelling doesn't move Z during prints.
Any MESH-based levelling, would move Z too much (most of the time) during print.

boelle, 2019-09-24:
@guestisp i guess this still does not work for you?

guestisp, 2019-09-24:
No, still doesn't work

guestisp, 2019-09-30:
Any update on this ? 
What i'm trying to get is very simple: 3POINT levelling to find the bed tilt, than, during prints, I would expect that marlin will correct the Z position automatically.

In example, if the lower right corner is at -0.10 and the upper left corner is at +0.10, I have a 0.20mm tilted bed. When moving from the lower right to the upper left, the firmware should raise the nozzle automatically

[26 more replies]

closed 2019-11-03 by boelle