MarlinFirmware/Marlin #13515: Hotend temperature fluctuates and heating failed system stopped occurs

sriram135, 2019-03-28:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

J-charles-C, 2019-03-28:
I,

Have you set hotend PID ?

https://reprap.org/wiki/PID_Tuning

best

sriram135, 2019-03-28:
No... I haven't set

J-charles-C, 2019-03-28:
I my link you have the command to set it.

You can use octoprint, repetier, pronterface to launch the g-code (usb link) or activate the Pid Menu in Marlin to use it with your screen.

sriram135, 2019-03-28:
Is it required? Or without that is ok

sriram135, 2019-03-28:
I have previously tried uncommenting the pid open loop and everything of that stuff but agian it doesn't work

J-charles-C, 2019-03-28:
it is essential to have a stable temperature and to have no heating error or safety when the ventilation is active.

sriram135, 2019-03-28:
Okay it seems I have tried that earlier... Didn't work ..

J-charles-C, 2019-03-29:
I,

Don't forgot to save parameters in eeprom after PID set.

best

sriram135, 2019-03-29:
Should I use your kp ki kd ....
As I did pid autotune using m303 I got my own results... 
Kp 12.7
Ki.  0.65
Kd. 57.8
Is this ok or I should use urs?

J-charles-C, 2019-03-30:
I,

your values are comparable to mine.
Have you registered them in Marlin?

best

sriram135, 2019-03-30:
No...

J-charles-C, 2019-03-30:
In terminal 👍 

M304 P12.7 I0.65 D57.8
M500

sriram135, 2019-03-30:
It is showing unknown command

J-charles-C, 2019-03-30:
M30**1** P12.7 I0.65 D57.8
M500

sorry

sriram135, 2019-03-30:
https://youtu.be/OsBx6u4L83E 

See this please

Bob-the-Kuhn, 2019-03-30:
If you don't mind getting back to basics …

Please ZIP the following files up and then drop them on your reply:
- configuration.h
- configuration_adv.h
- pins_YOUR-BOARD.h

What controller are you using?

Looks like you have a PT100 thermal sensor on the E3D.  Is that correct?

Does the hot end heat up?

With the hot end off, does the temperature reported by Pronterface increase when you hold the heater block?

What pin is your extruder fan connected to?

sriram135, 2019-03-30:
Thank you for responding... Here are my configs...
[CONFIGS.zip](https://github.com/MarlinFirmware/Marlin/files/3025291/CONFIGS.zip)

I m using Arduino Mega2560 with Ramps 1.4...
I m quiet not sure about the temp. sensor i have.... 
yes the hotend do heat up...(only when i separately heat it i.e setting pla temp in pronterface)
Yes with the hotend off when i hold it the temp do increase....
extruder fan is normally connected to two pins on ramps 1.4 i dont know which pins they are but fan works correct.

Bob-the-Kuhn, 2019-03-30:
On my system my extruder fan overwhelmed the heater. I could only get to 125C with the fan running at full speed all the time.

BEFORE TURNING ON THE FAN, stick something in the fan to keep the blades from turning.  Then turn on the heater and see if the temperature rises.  Turn it off before it gets to 100C.

sriram135, 2019-03-30:
Yep I will try that....

Bob-the-Kuhn, 2019-03-30:
From your configuration files it looks like your fan is tied to the D9 pins/connector.

If that's correct then try the following files.  The only change I made was to the configuration_adv.h file.
[Configuration_1-1-9.zip](https://github.com/MarlinFirmware/Marlin/files/3025428/Configuration_1-1-9.zip)

This should result in the extruder fan turning on at 50C and running at half speed.  If it doesn't turn on then you'll need to change line 247 in configuration_adv.h from 128 to 255.  All it means is your fan isn't designed to run at anything other than full speed.

---

I suggest strongly that you switch from Marlin 1.1.9 to Marlin's bugfix_2.0.x.  1.1.9 is not being updated.  2.0.x is the future. 

To help with the conversion I'm attaching the 2.0.x config files with your customizations included.  I only found a few.
[Configuration_2-0-x.zip](https://github.com/MarlinFirmware/Marlin/files/3025430/Configuration_2-0-x.zip)

sriram135, 2019-03-30:
Hey here is where I have connected fan... Check this pic...
https://drive.google.com/file/d/1-GY2gDU0qnWxMr2DTL6-VuBaCNHE0KD8/view?usp=drivesdk

Bob-the-Kuhn, 2019-03-30:
I think that's just 12V.  Does the fan run as soon as power is applied?

If you are running 12V steppers then move the orange wire to the + terminal on D9 and the white wire to the other terminal on D9.  That way you'll have control over the fan via the software.  If you have +24V then don't do this.  You can damage a 12V fan by running it on 24V.

sriram135, 2019-03-30:
Yes... As soon as per is on fan starts...
I m running 5V stepper motors... Shall I still connect as you told?

Bob-the-Kuhn, 2019-03-30:
You weren't kidding about home built.  In the picture I saw the piece of wood with a bearng and a rod through the bearing.  

How did you acquire this machine?

---

I added a 3D print head to my CNC machine several years ago.  It was low performance back then and it was down more than it was up.  I finally realized that I'd spent more on fixes & improvements than it would have cost to buy a low end printer.  The Creality Ender 3 would have saved me $$ and I'd have a better printer.  Finally put down some $$ on a mid range printer a month ago.

Bob-the-Kuhn, 2019-03-30:
5V motors??  There's not a 12V or 24V power supply?

What stepper drivers are you using?

kAdonis, 2019-03-30:
@Bob-the-Kuhn 
look at the above linked video
@sriram135 
What is your power source,? It is probably overloaded, you need at least 12V and 4A

Bob-the-Kuhn, 2019-03-30:
There is just +5V tied to the big power inputs on the RAMPS card?

If that's the case then go ahead and move the fan wires.

That also might explain the heating problem.  Unless you have something special the heater expects to see 12V or 24V.  If you're running a 12V heater from 5V then it's only seeing (5*5)/(12812) = 17% of the power it was designed for.  It may not get hot enough to melt PLA even without the fan.

Bob-the-Kuhn, 2019-03-30:
I am amazed!!!

Do you actually have motors moving?  What motors do you have?

Those look like A4988 steppers.  They're only rated for 8V and above operation.  

___

If you're anything like me it won't be long before you'll be wanting faster printing and higher quality prints.  It's really easy to do 3-5 upgrades at $30 - $50 over the next 6 months.  That puts you close to the cost of an Ender 3.  Please seriously consider the Ender 3 rather than doing upgrades.  The Ender 3 has reasonable speed and produces prints that are as good as come from printers 3 times the cost.

Bob-the-Kuhn, 2019-03-30:
I agree on the power supply.  

I've only seen 30W and 40W heaters.  Even if you did find a 5V heater chances are excellent that you'll burn up the USB cable before you melted any plastic.

sriram135, 2019-03-30:
Hey nice suggestions @kAdonis and @Bob-the-kuhn.... I have built this printer at my house and the motors move just fine!! I m using A4988 drivers... The motors get a little hot after sometime.... But for smaller prints they are absolutely fine... The machine works perfectly as CNC the only problem is with hotend...😶
@Bob-the-kuhn I built this printer because I wanted to... I had no idea all this would happen...
Please I need your help in at least making it print.. Now I don't want to spend any money on buying a new one... Please help me with this... Its a kind request...

Bob-the-Kuhn, 2019-03-30:
I'll stick with you as long as you're working on it.

Wow - another person that started with a CNC machine and decided to add 3D print capabilities.  We must be both geniuses.  My wife has a rather different opinion of what I've built and how I've been spending my time.  **☹️**

Bob-the-Kuhn, 2019-03-30:
OK - 12V power supply.  That's good.  That explains a lot.

I would suggest going to a better power cord.  I'm surprised that it hasn't already burned up.

Yes, it's OK to move the fan to the D9 connector.

sriram135, 2019-03-30:
Oh sir u are older than me I m a student studying engineering....
I daily sit infront of the printer dealing with problem and get up without solving it... 
Sir what shall I do now? Shall I connect fan to D9 port? Or shall I update the marlin and flash marlin 2.0 bugfix?

kAdonis, 2019-03-30:
In the video, you could see the red led near the Hotend connector on the RAMPS flashing, indicating that the output is active, but **only for a split second**. It should be permanently on until the hotend temperature is nearly reached and then start pulsing.
I think your 12V power supply is switching off, because it is overloaded.
Your hotend might have a 24V heater cartridge or is shorted, could you measure the resistance?
And where is this beeping and clicking sound coming from?

Bob-the-Kuhn, 2019-03-30:
You'll need some cooling on the hot end or you'll melt plastic too far up the borden tube.  I ruined a heatsink trying to drill out a melted tube.

Let's stick with 1.1.9 and play with the fan to see if it can be throttled back far enough that you can melt plastic.

For a month or so I had a piece of cardboard partially covering my fan.   I eventually ordered one that could be throttled back.

sriram135, 2019-03-30:
@kAdonis the beeping and clicking sound is maybe from fan turning on and off ... Its a E3D V6 all metal hotend Wade extruder type... I think it maybe 12V heater cartridge....hmm I don't have multimeter right now to measure the resistance...😁

@Bob-the-kuhn throttle back means slowing it down or reversing the fan direction?

kAdonis, 2019-03-30:
> the beeping and clicking sound is maybe from fan turning on and off 

If you didn't change you connection yet, the fan is connected directly to 12V from your power supply, therefore it should run all the time the power supply is on.
For testing you could disconnect the heater from RAMPS and try to heat, the fan should stay on and the red led should light up (until heating failed error, of course)
If that works we know its problem with heater or power supply, if not we have to look further

sriram135, 2019-03-30:
Now I connected it to D9 and saw for first time the when I gave print the temp rose up to max 230  and kept stabilizaling it didn't resume the print at all... Then I disconnected and reconnected to pronterface then again I saw the beeping sound and clicking sound... And it gave thermal runaway error...

sriram135, 2019-03-30:
Now the red led is on only when temp is above 50 Celsius that too at half speed... As per Bob-the-kuhn

kAdonis, 2019-03-30:
What did you connect to D9, the fan or the heater?

sriram135, 2019-03-30:
The fan..

kAdonis, 2019-03-30:
are you using Bob-the-Kuhn's Configuration_1-1-9.zip?

sriram135, 2019-03-30:
Yes

kAdonis, 2019-03-30:
There are two leds on RAMPS for D9 and D10,
D9 should blink over 50 deg, thats fine 
Try now to disconnect the Heater from D10, let the printer cool down below 100deg and try to heat 
watch led 10, it should be on and no clicking and beeping

sriram135, 2019-03-30:
Should I heat it without the heater connected?

kAdonis, 2019-03-30:
yes
just a test

sriram135, 2019-03-30:
Yes it remained on i.e it started heating correctly...

kAdonis, 2019-03-30:
That means software is working
You have either a faulty heater, a 24v heater- you need a multimeter to verify
or your power supply is not able to deliver enough current

From your picture I think your hotend is probably a clone, not a genuine E3d 
Genuine E3d hotends need thermistor 5, but i have seen clones that need thermistor 11
Sorry, I have to leave now, good luck with further troubleshooting

sriram135, 2019-03-30:
Oh okay thank you very much for spending your valuable time in trying to solve my problem...

sriram135, 2019-03-30:
@Bob-the-kuhn how do I know what kind of E3d I have whether it's clone or genuine... I purchased it for 800 INR..

kAdonis, 2019-03-30:
I forgot...
The Polyswitch fuse on your RAMPS could be faulty

sriram135, 2019-03-30:
Oh is it.... Should I replace it with new one?

Bob-the-Kuhn, 2019-03-30:
At that price it's sure to be a clone.  Most of the time they work OK.

Take a look at the link where you bought it.  It should state there if it's 12V or 24V.  The fan may say if it's 12 or 24V.

---

>  temp rose up to max 230 and kept stabilizaling it didn't resume the print at all... Then I disconnected and reconnected to pronterface then again I saw the beeping sound and clicking sound... And it gave thermal runaway error.

Thermal runaway error is declared by the Marlin software and reported by pronterface..  Once marlin declares a thermal problem then it kills the print and will not resume it.  You need to cool things down and then restart the print.

230C is the max set in configuration.h.  Marlin will not allow the temperature to go beyond that.

There may be a problem with the commands in the print file.  Please ZIP it up and post it.

I'd like you to manually set the extruder temperature and observe what happens.  

I'd like you to go to the line that says HEAT and then type in 75 into the box and then click set.  You should see the temperature rise to 75C and then stabilize there  The fan should come on at 50C.

If that goes well then type in 125, hit set and see what happens.

Then 175.

Then 215.

sriram135, 2019-03-31:
Yep I will do that....

sriram135, 2019-03-31:
How to zip commands in print file? Please tell me that...

Bob-the-Kuhn, 2019-03-31:
Just take the file itself & ZIP it.    It's just a text file.

sriram135, 2019-03-31:
Ok sure

sriram135, 2019-03-31:
[Anet_A8_extruder_relese_knob.zip](https://github.com/MarlinFirmware/Marlin/files/3026539/Anet_A8_extruder_relese_knob.zip)
Here is the zip

Bob-the-Kuhn, 2019-03-31:
There's nothing strange in the gcode file.  It sets the temperature to 200C.

sriram135, 2019-04-01:
Oh okay... Then I checked the temp .... First I set it to 75.. the fan started normal at 50 but heater went on climbing above 75! It didn't stop at 75...

[39 more replies]

closed 2019-07-11 by thinkyhead