MarlinFirmware/Marlin #13517: Problem when stopping a print (through SD Card)

LordDarkneo, 2019-03-28:
Hi there!

### Description

When stopping a print (started via SD Card), the hotend is going down, towards the printed part. Damaging the part is not a problem (as I cancelled the print), but it forces on all the Z axis and the bed... That is not really good for the printer... 

### Steps to Reproduce

You caneasily reproduce the issue:
1- Get a temperature tower on thingi
2- Print the tower
3- Stop the print at like 50/60% of the print and prepare to remove the already printed part!
4- Stay close to remove the tower when the hotend is moving down where it was when you stopped the print

**Expected behavior:** When cancelling the print, I would like to have the hotend going up (for like 5 cm)

**Actual behavior:** The hotend goes up for few cm and then goes down all over the printed part...

7eggert, 2019-03-28:
It should not go down but stay in place. (Still the nozzle will burn a hole into the print.)

In order to do more than that, I created:
github.com/MarlinFirmware/Marlin/pull/12850; you can cherry-pick it.

robbycandra, 2019-04-01:
I create PR to Park Nozzle when stop printing. #13547

robbycandra, 2019-04-15:
This issue should be closed.

LordDarkneo, 2019-04-16:
I'll try on the 2.0.x version then, that's it? So far I am still on 1.1.9.... 

Thanks for fixing this ! :)

robbycandra, 2019-04-16:
@LordDarkneo , you can close this issue

closed 2019-04-16 by LordDarkneo