MarlinFirmware/Marlin #13518: Bltouch triggered but doesn't stop homing

TwoToneEddy, 2019-03-28:
Hi,
I've just installed a 3D touch sensor and I'm confused as to why it's not working. When I actuate the sensor and manually move it up m119 returns triggered, when it's down it returns open (for z min). This is right isn't it? Any reason why the home routine would continue to plough into the bed?

GMagician, 2019-03-28:
AFAIK bltouch should always return "open" since "triggered" last for only 10ms and it's almost impossible to get. You may try enable debugging info, M111 S255, with a firmware compiled for debug info (DEBUG_LEVELING_FEATURE)

TwoToneEddy, 2019-03-28:
Ok I'll try that. I would assume it returning tiggered would be the correct behaviour, it means it's doing something. It looks like the trigger latches. Mines a 3dtouch so perhaps that's different? I'll try what you said though. Thanks for the help

TwoToneEddy, 2019-03-28:
Sorry stupid question, how do I enable DEBUG_LEVELING_FEATURE? I've put a #define DEBUG_LEVELING_FEATURE in configuration.h but it doesn't compile. Also what do I do once I've got debugging enabled? Try homing?

Bob-the-Kuhn, 2019-03-29:
I don't remember if a 3D touch acts like a BLTouch or it should be setup as Z Servo Probe.  Try setting it up as a Z Servo Probe:
```
#define Z_PROBE_SERVO_NR 0   // Defaults to SERVO 0 connector.
#define Z_SERVO_ANGLES {90,10}  // Z Servo Deploy and Stow angles
#define NUM_SERVOS 1  // assuming the 3D touch is the only servo in the system
```

You can also use the M43 S1 command to test the unit.  It'll tell you if it's a BLTouch compatible device or not.

TwoToneEddy, 2019-03-30:
I've got a genuine bltouch now and it does the same thing. I've tried the m43 S1 command and it doesn't do anything, do I need to enable this in firmware and if so where? 

When the printer homes and I touch the probe manually it retracts but the printer doesn't stop. Could it be the power isn't enough? I'm taking it from the ISP pins

GMagician, 2019-03-30:
> do I need to enable this in firmware and if so where?

Yes but hard to help with no configuration posted...

Bob-the-Kuhn, 2019-03-30:
Enable PINS_DEBUGGING in configuration_adv.h

TwoToneEddy, 2019-03-31:
Ok thanks, i'll try that this morning. In the meantime here's my config.h file if there's anything glaringly obvious wrong:

`/**
* ************** How to use this firmware - READ THIS, yes actually read this. *********************************
*
* Uncomment means removing the 2 // in front of #define.
* 
* FLASHING NOTES:
* ALL slicers (Simplify3D, Cura, Slic3r, etc) or anything else that connects to the COM port must be CLOSED for the firmware to be flash.
* If anything is connected to the COM port when flashing it will fail, typically "access denied" is listed in the error section of the Arduino IDE.
* You MUST use the Included Arduino IDE to flash the firmware if on Windows and if you are on a Mac or Linux follow the guide in our knowledgebase to setup your IDE to work with the firmware.
* 
* EZABL SETUP NOTES: 
* If you have EZABL uncomment the mount you are using with the printer. 
* If you have a custom/unsupported mount uncomment #define CUSTOM_MOUNT and enter your offsets 
* below in the CUSTOM MOUNT section. Refer to the EZABL guide to get your offsets.
* 
* STEP 1:
* Select the correct board from the tools menu for the printer you are flashing.
* Read the printer title for the model you are flashing, it will show what board to select.
*
* STEP 2:
* Uncomment the printer you want to flash. The printers are sorted A-Z by brand name.
* If you are using the CR-10S DUAL board with the Ender 3 then read the specific section below on how to do this.
*
* STEP 3: 
* Select the COM port your printer is on from the Tools menu. If you do not see the COM port try
* downloading the latest drivers from the manufacturer or TH3D site on our knowledgebase.
* 
* STEP 4:
* Verify you have the correct board selected, printer model uncommented, and if you are using EZOUT and/or EZABL
* the lines you need to use them are also uncommented.
*
* STEP 5:
* Once you have your settings verified click the arrow in the upper left to upload to the board.
*
* STEP 6:
* Reset your eeprom. You can send M502 then M500 to reset the EEPROM OR on the printer LCD go to 
* Control > Reset EEPROM to clear out the EEPROM to defaults.
*
* BOOTLOADER FLASHING NOTES:
* For flashing your bootloader with an Uno make sure to select Arduino as ISP for the programmer
* 
* There are other features in the TH3D Extras section so look there for V6 Hotend,
* Bootscreen settings, Titan Extruder and more. You only need to edit this file.
* 
* ERROR NOTES:
* If you get errors flashing READ the message it gives you and double check that you selected
* the correct board from the Tools menu in Arduino. Turn off any AV systems and reboot the computer.
* 
* COMMUNITY REQUESTED FEATURES NOTE:
* All features in the community requested features section are provided as-is with no support from TH3D.
*/

#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010109

//===========================================================================
//============================ TH3D Configuration ===========================
//===========================================================================

// ONLY UNCOMMENT THINGS IN ONE PRINTER SECTION!!! IF YOU HAVE MULTIPLE MACHINES FLASH THEM ONE AT A TIME.

//===========================================================================
// *************************     TH3D PRINTERS      *************************
//===========================================================================

//===========================================================================
// TH3D Alpha EZ300 Options - Select Arduino Mega 2560 from Tools > Board
[... 620 lines cut ...]
//#define S_CURVE_ACCELERATION_DISABLE

//================================================================================================
// Language - This is provided for convenience and is unsupported with included product support.
// We only test compile with English language. If you run into space issues disable some features.
//================================================================================================

/**
 * LCD LANGUAGE
 *
 * Select the language to display on the LCD. These languages are available:
 *
 *    en, an, bg, ca, cn, cz, cz_utf8, de, el, el-gr, es, eu, fi, fr, fr_utf8, gl,
 *    hr, it, kana, kana_utf8, nl, pl, pt, pt_utf8, pt-br, pt-br_utf8, ru, sk_utf8,
 *    tr, uk, zh_CN, zh_TW, test
 */

#define LCD_LANGUAGE en

#include "Configuration_backend.h"

#define UNIFIED_VERSION "TH3D U1.R2.9b"

#endif // CONFIGURATION_H
`

TwoToneEddy, 2019-03-31:
Ok I don't think it's a firmware issue, I've tried a config file that's been proven to work and it doesn't. I also now realise there's a difference between vanilla marlin and th3d, sorry I posted this in the wrong place. Thanks for the help anyway, I'm going to try a dedicated 5v supply

TwoToneEddy, 2019-03-31:
Ok I'm now using this firmware and I've enabled the pins_debugging feature. When I run m43 S1 it makes the bltouch actuate 4 times then waits for a trigger but never gets a trigger no matter how many times I touch it. I've tried this as a bltouch and as a custom servo probe, both don't trigger. When I make the touch go into an error state it stops the homing so I think that proves my wiring is right, it sounds like the pulse is too small for the printer to see.

TwoToneEddy, 2019-03-31:
Ok I've got it working but it's not ideal. I had to wire the bltouch trigger signal to a separate Arduino, trigger an interrupt and output a 200ms pulse to the printer, looks like the printer isn't quick enough to pick up the pulse, I don't understand why though as it should be using interrupts shouldn't it?

Bob-the-Kuhn, 2019-03-31:
I just put a black 3DTouch on a DUE/RAMPS_FD_V1 system and it works just fine in response to the M43 S1 command.  I then tied a 1.2K pull up and then pull down with no adverse effects.

I don't see a problem with the unit itself.

But .. on your system if you unplug the intermediate Arduino and plug the yellow wire from the 3DTouch in it's place, the M43 S1 command says it didn't detect a pulse.

The only thing I can think of that might help is to tie a 1K (or there about) pull up to the yellow wire.

---

The title says BLTouch but the opening post says 3D touch.

If you're actually running a BLTouch then look at this [thread.](https://github.com/MarlinFirmware/Marlin/issues/13345)  The V3 BLTouch has problems with most of the Creality printers.  Most are fixed by switching to new firmware and enabling the correct options.  One also needs a 1K pullup.

StephenPArnold, 2019-04-13:
My $0.02 ...

I was having issues until I remembered:

#define SERVO0_PIN 27

on my Anet A8.

closed 2019-05-18 by thinkyhead