MarlinFirmware/Marlin #13521: [BUG] [bugfix-2.0.x] viki lcd compile error

an-tech, 2019-03-29:
### Description

compile error after uncommenting: #define LCD_I2C_VIKI

### Steps to Reproduce

1. uncommenting: //#define LCD_I2C_VIKI

**Expected behavior:** [What you expect to happen]
no compile error

**Actual behavior:** [What actually happens]
compile error:
```
Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\src\Marlin.cpp:33:0:

sketch\src\lcd/ultralcd.h: In static member function 'static void MarlinUI::buzz(long int, uint16_t)':

sketch\src\lcd/ultralcd.h:253:7: error: 'lcd' was not declared in this scope

       lcd.buzz(duration, freq);

       ^

sketch\src\lcd/ultralcd.h: In static member function 'static bool MarlinUI::button_pressed()':

sketch\src\lcd/ultralcd.h:156:19: error: 'BUTTON_SELECT' was not declared in this scope

     #define B_MI (BUTTON_SELECT << B_I2C_BTN_OFFSET)

                   ^

sketch\src\lcd/ultralcd.h:162:42: note: in expansion of macro 'B_MI'

       #define BUTTON_CLICK() (buttons & (B_MI|B_RI|B_ST)) // Pause/stop also acts as click until a proper pause/stop is implemented.

                                          ^

sketch\src\lcd/ultralcd.h:492:50: note: in expansion of macro 'BUTTON_CLICK'

     static inline bool button_pressed() { return BUTTON_CLICK(); }

                                                  ^

sketch\src\lcd/ultralcd.h:158:19: error: 'BUTTON_RIGHT' was not declared in this scope

     #define B_RI (BUTTON_RIGHT  << B_I2C_BTN_OFFSET)

                   ^

sketch\src\lcd/ultralcd.h:162:47: note: in expansion of macro 'B_RI'

       #define BUTTON_CLICK() (buttons & (B_MI|B_RI|B_ST)) // Pause/stop also acts as click until a proper pause/stop is implemented.

                                               ^

sketch\src\lcd/ultralcd.h:492:50: note: in expansion of macro 'BUTTON_CLICK'

     static inline bool button_pressed() { return BUTTON_CLICK(); }

                                                  ^

Multiple libraries were found for "LiquidTWI2.h"
 Used: C:\Users\A\Documents\Arduino\libraries\LiquidTWI2-master
 Not used: C:\Users\A\Documents\Arduino\libraries\arduino-display-lcdkeypad
exit status 1
In file included from sketch\src\gcode\control\M17_M18_M84.cpp:25:0:

c:\users\A\appdata\local\temp\arduino_build_742026\sketch\src\lcd\ultralcd.h: In static member function 'static void MarlinUI::buzz(long int, uint16_t)':

c:\users\A\appdata\local\temp\arduino_build_742026\sketch\src\lcd\ultralcd.h:253:7: error: 'lcd' was not declared in this scope

       lcd.buzz(duration, freq);

       ^

c:\users\A\appdata\local\temp\arduino_build_742026\sketch\src\lcd\ultralcd.h: In static member function 'static bool MarlinUI::button_pressed()':

c:\users\A\appdata\local\temp\arduino_build_742026\sketch\src\lcd\ultralcd.h:156:19: error: 'BUTTON_SELECT' was not declared in this scope

     #define B_MI (BUTTON_SELECT << B_I2C_BTN_OFFSET)

                   ^

c:\users\A\appdata\local\temp\arduino_build_742026\sketch\src\lcd\ultralcd.h:162:42: note: in expansion of macro 'B_MI'

       #define BUTTON_CLICK() (buttons & (B_MI|B_RI|B_ST)) // Pause/stop also acts as click until a proper pause/stop is implemented.

                                          ^

c:\users\A\appdata\local\temp\arduino_build_742026\sketch\src\lcd\ultralcd.h:492:50: note: in expansion of macro 'BUTTON_CLICK'

     static inline bool button_pressed() { return BUTTON_CLICK(); }

                                                  ^

c:\users\A\appdata\local\temp\arduino_build_742026\sketch\src\lcd\ultralcd.h:158:19: error: 'BUTTON_RIGHT' was not declared in this scope

     #define B_RI (BUTTON_RIGHT  << B_I2C_BTN_OFFSET)

                   ^

c:\users\A\appdata\local\temp\arduino_build_742026\sketch\src\lcd\ultralcd.h:162:47: note: in expansion of macro 'B_RI'

       #define BUTTON_CLICK() (buttons & (B_MI|B_RI|B_ST)) // Pause/stop also acts as click until a proper pause/stop is implemented.

                                               ^

c:\users\A\appdata\local\temp\arduino_build_742026\sketch\src\lcd\ultralcd.h:492:50: note: in expansion of macro 'BUTTON_CLICK'

     static inline bool button_pressed() { return BUTTON_CLICK(); }

                                                  ^

Error compiling for board Arduino/Genuino Mega or Mega 2560.
```

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3023734/Marlin.zip)

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

shitcreek, 2019-09-17:
This is a library issue and not with Marlin. You need to install this library: https://github.com/lincomatic/LiquidTWI2.

closed 2019-09-17 by shitcreek