MarlinFirmware/Marlin #13525: ESP32 board compile error

irshadmgs011, 2019-03-30:
[image]

Ludy87, 2019-03-30:
where are the config files and the pin _ #. h file?

Bob-the-Kuhn, 2019-03-30:
I know this is off topic but …   I have some WIFI newbie questions.

Yesterday it dawned on me that Marlin has WIFI support and I have a Cohesion3D Remix on the shelf.  Unfortunately the Remix has an ESP8266 and the Marlin WIFI is expecting an ESP32.  Some really basic questions:
- Configuration_adv.h says "over-the-air firmware updates" -  is this firmware for the ESP32 or Marlin?
- My dream is to cut the USB cable and, wirelessly, be able to access the virtual disk and do Marlin firmware updates.  Are either of these possible now (without spending $200)?

Link/pointers to where I can learn more would be appreciated.

irshadmgs011, 2019-03-30:
> where are the config files and the pin _ #. h file?
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020000

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
// config/examples/SCARA and customize for your machine.
//
[... 2013 lines cut ...]

/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// Delay (in milliseconds) before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Allow servo angle to be edited and saved to EEPROM
//#define EDITABLE_SERVO_ANGLES

Ludy87, 2019-03-30:
@Bob-the-Kuhn 

> I know this is off topic but … I have some WIFI newbie questions.
>
> Yesterday it dawned on me that Marlin has WIFI support and I have a Cohesion3D Remix on the shelf. Unfortunately the Remix has an [...]

See: https://www.youtube.com/watch?v=pJGBRriNc9I

irshadmgs011, 2019-03-30:
> where are the config files and the pin _ #. h file?

/**
 * Espressif ESP32 (Tensilica Xtensa LX6) pin assignments
 */

#ifndef ARDUINO_ARCH_ESP32
  "Oops! Select an ESP32 board in 'Tools > Board.'"
#endif

#define BOARD_NAME "Espressif ESP32"

//
// Limit Switches
//
#define X_MIN_PIN          34
#define Y_MIN_PIN          35
#define Z_MIN_PIN          15

//
// Steppers
//
#define X_STEP_PIN         128
#define X_DIR_PIN          129
#define X_ENABLE_PIN       130
//#define X_CS_PIN           0

#define Y_STEP_PIN         131
#define Y_DIR_PIN          132
#define Y_ENABLE_PIN       133
//#define Y_CS_PIN          13

#define Z_STEP_PIN         134
#define Z_DIR_PIN          135
#define Z_ENABLE_PIN       136
//#define Z_CS_PIN           5 // SS_PIN

#define E0_STEP_PIN        137
#define E0_DIR_PIN         138
#define E0_ENABLE_PIN      139
//#define E0_CS_PIN         21

//
// Temperature Sensors
//
#define TEMP_0_PIN          36   // Analog Input
#define TEMP_BED_PIN        39   // Analog Input

//
// Heaters / Fans
//
#define HEATER_0_PIN         2
#define FAN_PIN             13
#define HEATER_BED_PIN       4

irshadmgs011, 2019-03-30:
> where are the config files and the pin _ #. h file?

Now you can tell something?

Ludy87, 2019-03-30:
> > where are the config files and the pin _ #. h file?
> 
> Now you can tell something?

change value from `TX_BUFFER_SIZE 0` to `TX_BUFFER_SIZE 2` and try again

irshadmgs011, 2019-03-30:
> > > where are the config files and the pin _ #. h file?
> > 
> > 
> > Now you can tell something?
> 
> change value from `TX_BUFFER_SIZE 0` to `TX_BUFFER_SIZE 2` and try again

I have already done all this......  i thought you have the ability

Ludy87, 2019-03-30:
1. Du schreibst nicht was passiert ist und machst du machst nur ein Screenshot
2. Die Vorlage zum erstellen von Fehlerberichten sind dafür da um eine Struktur zu haben
3. Du wirst gebeten, die Dateien hochzuladen und du schickst die einzelnen Zeilen
4. Dir wird ein Ansatz geliefert, den ich extra getestet habe - für dich! Und wirst beleidigend

Dieser Report ist in meinem Augen zum Schließen verurteilt - siehe Community-Richtline!

P.s. ich schreibe auf deutsch, da es nicht wert voll ist, das auf Englisch für dich zu schreiben. Ich bin hier raus. Viel Spaß!

irshadmgs011, 2019-03-30:
> 1. Du schreibst nicht was passiert ist und was passiert ist - du machst nur ein Screenshot
> 2. Die Vorlage zum erstellen von Fehlerberichten sind dafür da um eine Struktur zu haben
> 3. Du wirst gebeten, die Dateien hochzuladen und du schickst die einzelnen Zeilen
> 4. Dir wird ein Ansatz geliefert, den ich extra getestet habe - für dich! Und wirst beleidigend
> 
> Dieser Report ist in meinem Augen zum Schließen verurteilt - siehe Community-Richtline!
> 
> P.s. ich schreibe auf deutsch, da es nicht wertvoll ist, das auf Englisch für dich zu schreiben. Ich bin hier raus. Viel Spaß!

The way you ask me I thought you are a developer, and I have not said any insult

Ludy87, 2019-03-30:
the error can not occur with the value 2

irshadmgs011, 2019-03-31:
> the error can not occur with the value 2

HAVE YOU DONE THIS?

Ludy87, 2019-03-31:
yes I did that compiled without errors

irshadmgs011, 2019-03-31:
> yes I did that compiled without errors

please make a video and send me, upload file and screenshot and bin file

ZS1FJK, 2019-04-01:
Hello
The only way i see this happening is if  `#define TX_BUFFER_SIZE 0` 
Please zip both Configuration_adv.h and configuration.h files and attach to this forum. 
This allows us to try build the files on our side and also try fix from our side.
Also make sure you have the latest bug fix or recent anyway.
I would try #define TX_BUFFER_SIZE 32 since this is a defined value in the websocketserial.h

`#ifndef RX_BUFFER_SIZE
  #define RX_BUFFER_SIZE 128
#endif
#ifndef TX_BUFFER_SIZE
  #define TX_BUFFER_SIZE 32
#endif
#if TX_BUFFER_SIZE <= 0
  #error "TX_BUFFER_SIZE is required for the WebSocket."
#endif`

Good luck

simon-jouet, 2019-04-12:
Late reply @Bob-the-Kuhn 

> Configuration_adv.h says "over-the-air firmware updates" - is this firmware for the ESP32 or Marlin?
My dream is to cut the USB cable and, wirelessly, be able to access the virtual disk and do Marlin firmware updates. Are either of these possible now (without spending $200)?

It's for Marlin, if you run Marlin on the ESP32 you can flash a new version through WiFi, there used to be some problems with the OTA and the interrupts though, but last time i tried it was working fine, and with the new I2S support it should free up a lot of resources

If you have an SDcard and an ESP32 you can basically do that, the web interface will need some work but it's fairly straightforward.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@irshadmgs011 is this still a problem?

irshadmgs011, 2019-09-25:
I left this project, I have to see it again

closed 2019-10-12 by boelle