MarlinFirmware/Marlin #13526: [BUG] not in sync

Sjoerrdd, 2019-03-30:
### Description
I'm trying to flash Marlin to my Creality Ender 3

### Steps to Reproduce
0. Unlock your bootloader
1. Connect your Ender 3 via USB
2. Open Marlin in Arduino IDE
3. Press Upload

**Expected behavior:**
Marlin instsalled on my 3D printer

**Actual behavior:** [What actually happens]
```
Sketch uses 122158 bytes (93%) of program storage space. Maximum is 130048 bytes.
Global variables use 4650 bytes (28%) of dynamic memory, leaving 11734 bytes for local variables. Maximum is 16384 bytes.
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x65
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x63
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x68
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x6f
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x3a
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x53
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x44
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x63
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x61
An error occurred while uploading the sketch
```

#### Additional Information

Configuration.h & Configuration_adv.h are the default files for the Creality Ender 3.
Video's I used: https://www.youtube.com/watch?v=hbX8Qu-QLnI

Bob-the-Kuhn, 2019-03-30:
Can't sleep so I'll bother you with my lack of knowledge on the Ender 3. **😀**

Were you able to install the bootloader?    What board did you use between your PC and the Ender 3?

FanDjango, 2019-03-30:
Right board selected in Arduino IDE?
USB Drivers up to date?
Do you need to press a reset or flash button on the board at exactly the right moment to allow avrdude to start?
And Bob said rightly, if the bootloader is not there or corrupted, you need to use a more basic tool to fix that first before avrdude can work. Have you done anything previously that might have overwritten/corrupted the bootloader?

ManuelMcLure, 2019-03-31:
That could be a baud rate mismatch between what Arduino is sending and the bootloader. I actually saw this today trying to upload to a clone Arduino Nano (for a non-3d printing related project) and I had to switch from the "Arduino Nano" to the "Arduino Nano (old bootloader)" boards so the baud rate switched from 115200 down to 57600.

Sjoerrdd, 2019-03-31:
This issue has been sold, pins aren't connected the correct way.
After I installed Marlin, my Ender 3 pro died.
No screen when BLTouch is connected, only backlight when BLTouch is disconnected. :\

Bob-the-Kuhn, 2019-03-31:
Dead Ender 3 means that something happened when you tried to flash the bootloader.  The synch failure means that either the bootloader didn't get flashed correctly or something is wrong with your Arduino setup.

I suggest trying to burn the bootloader again.  

Here's a different link on burning a bootloader with the Raspberry PI.  It's a quite detailed list of steps.
.https://www.fission3d.com/guides/flash-bootloader-and-install-firmware-with-raspberry-pi

Sjoerrdd, 2019-04-01:
Thanks @Bob-the-Kuhn,
I tried that also, but still no display :(

BlackienBad, 2019-04-05:
I can't even burn the bootloarder

Sjoerrdd, 2019-04-09:
My display was broken 😅 
Bought a new one and the issue is solved.

closed 2019-04-09 by Sjoerrdd

speed-pro, 2019-08-14:
I have a similar problem
I do everything as on video and get same error
Creality ender 3
Board :1.1.4

avrdude: stk500_getsync () attempt 1 of 10: not in sync: resp = 0x65
avrdude: stk500_getsync () attempt 2 of 10: not in sync: resp = 0x63
avrdude: stk500_getsync () attempt 3 of 10: not in sync: resp = 0x68
avrdude: stk500_getsync () attempt 4 of 10: not in sync: resp = 0x6f
avrdude: stk500_getsync () attempt 5 of 10: not in sync: resp = 0x3a
An error occurred while uploading the sketch
avrdude: stk500_getsync () attempt 6 of 10: not in sync: resp = 0x54
avrdude: stk500_getsync () attempt 7 of 10: not in sync: resp = 0x46
avrdude: stk500_getsync () attempt 8 of 10: not in sync: resp = 0x20
avrdude: stk500_getsync () attempt 9 of 10: not in sync: resp = 0x69
avrdude: stk500_getsync () attempt 10 of 10: not in sync: resp = 0x6e

Sjoerrdd, 2019-08-14:
@speed-pro try to install your bootloader again

speed-pro, 2019-08-14:
Thanks for your reply
I have only Arduino Nano here, 
I connect it as in the picture
https://imgur.com/a/pRQr8lG
and also get some error on bootloader install  :(

Sjoerrdd, 2019-08-14:
@speed-pro as far as I know (from TeachingTech's video), this is the wrong way. 
Rotate the uno 180 degrees, connect it again and it'll work

speed-pro, 2019-08-14:
it working thanks!!!
do it whit ardoino nano, only install the bootloader again and it work normal.
I'm Fix POWER_LOSS_RECOVERY
there Is anything else I need to know?