MarlinFirmware/Marlin #13527: limite de compensation de deviation

boubouexpress, 2019-03-30:
Bonjour

je voulais savoir si avec un plateau de se type la limite de compensation de déviation suffisait

et si non ou dans marlin on peux changer cette valeur

je sait que sur la prusa i3 mk3 la limite de compensation et de 0.05

mais dans marlin 1.1.9  je ne c'est de combien est la valeur

merci d'avance

FanDjango, 2019-03-30:
You are asking if there is a limit to the amount of deviation that can be compensated on such a warped bed? What leveling algorithm? Linear or what? We are looking at .5mm in toto? I know of no such limit offhand. Maybe someone else could chime on on this.

Bob-the-Kuhn, 2019-03-30:
> I  wanted to know if with a plateau of type the deviation compensation limit was enough
and if not or in marlin we can change this value
I know that on the Prusa i3 mk3 the limit of compensation and 0.05
but in marlin 1.1.9 I do not it's how much is the value

My system had 2.1mm of variation from high to low and UBL handled it very nicely.

Mon système présentait une variation de haut en bas de 2,1 mm et UBL l'a très bien gérée.

boubouexpress, 2019-04-03:
j'ai un bltouch avec du AUTO_BED_LEVELING_BILINEAR il ne corrige pas suffisamment

Bob-the-Kuhn, 2019-04-03:
> I have a bltouch with AUTO_BED_LEVELING_BILINEAR it does not correct enough

I do not know what options are available for bilinear.

I use UBL with a BLTouch.  The BLTouch is used to generate the initial rough mesh.  After that I use  G26 and G29 P4 to fine tune the loop.  When I slice I specify a large skirt so that I have enough time to use babystepping to dial that all important first layer. http://marlinfw.org/docs/features/unified_bed_leveling.html

Je ne sais pas quelles options sont disponibles pour la bilinéaire.

J'utilise UBL avec un BLTouch. Le BLTouch est utilisé pour générer le maillage brut initial. Après cela, j'utilise G26 et G29 P4 pour affiner la boucle. Lorsque je coupe, je spécifie une grande jupe pour avoir assez de temps pour utiliser babystepping pour composer le premier calque important.
http://marlinfw.org/docs/features/unified_bed_leveling.html

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

boubouexpress, 2019-07-21:
j'ai simplement utilisait bugfix-1.1.x et le problème a disparus

boelle, 2019-07-21:
i'm not french so dont understand, google translate says: 
I just used bugfix-1.1.x and the problem disappeared

so i will take that as fixed, unless its broken in 2.0.x