MarlinFirmware/Marlin #13528: [BUG] Marlin 2.0 with MAX31855 reads 0 degrees on STM32F1

Idorobots, 2019-03-30:
### Description

Hi. I'm trying to use a MAX31855-based temperature sensor with my DIY STM32F103CB-based controller, but Marlin always reads 0 degrees. The same exact setup with a simple Arduino schetch (included below) reads the temperature OK, so the MAX31855 breakout and the thermocouple I'm using are probably fine.

### Steps to Reproduce

Using a generic "blue pill" module and the following platformio.ini environment:

```
#
# STM32F103CB
#
[env:STM32F103CB]
platform      = ststm32
framework     = arduino
board         = genericSTM32F103CB
build_flags   = !python Marlin/src/HAL/HAL_STM32F1/STM32F1_flag_script.py
  ${common.build_flags}
  -DdigitalPinHasPWM(P)='(PIN_MAP[P].timer_device != NULL)'
lib_deps      = TMCStepper@<1.0.0
  U8glib-HAL=https://github.com/jmz52/U8glib-HAL/archive/stm32-maple-and-core-fix.zip
lib_ignore    = c1921b4
  libf3c
  lib066
  Adafruit NeoPixel_ID28
  Adafruit NeoPixel
  libf3e
  TMC26XStepper
lib_ldf_mode  = 1
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_STM32F1>
monitor_speed = 250000
```

1. Hook up the MAX31855 breakout board to the hardware SPI with PB1 as the SS pin.
2. Use the provided Configuration.h & pins_SPARTAN.h (in [configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3025872/configuration.zip) )
3. `platformio run -e STM32F103CB -t upload`

**Expected behavior:** Temperature value is read correctly.

**Actual behavior:** Temperature reading is reported as 0.

#### Additional Information

Disconnecting the thermocouple or the MAX31855 breakout results in a MAXTEMP error during boot. In case it was a timing issue, I tried changing the hardware SPI speed manually in `temperature.cpp`'s `read_max6675()` to no avail.

Working Arduino schetch:

```C
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <Adafruit_MAX31855.h>

Adafruit_SSD1306 display(128, 64, &Wire, -1);
Adafruit_MAX31855 thermocouple(PB1);

double temp = 0;
double temp0 = 0;
uint8_t error = 0;

void setup() {  
  display.begin(SSD1306_SWITCHCAPVCC, 0x3C);
  display.setTextColor(WHITE);
  thermocouple.begin();
}

void draw() {
  display.setCursor(0,0);
  display.clearDisplay();

  display.print("Temp: ");
  display.println(temp);
  
  display.print("Temp0: ");
  display.println(temp0);

  display.print("Error: ");
  display.println(error);
  display.display();
}

void readSensors() {
  temp = thermocouple.readCelsius();
  temp0 = thermocouple.readInternal();
  error = thermocouple.readError();
}

void loop() {
  draw();
  readSensors();
  delay(1000);
}
```

How would I go about debugging this further?

Idorobots, 2019-03-30:
It turns out the SS pin wasn't pulled up and the hardware SPI was initialized when software SPI was selected, resulting in MAX38155 never properly responding. I'm not sure if #13530 is a proper fix for this issue, but it certainly works on my setup.

Idorobots, 2019-03-31:
@thinkyhead Shall I close this as fixed?

shitcreek, 2019-05-15:
Please do if it is.

PR #13530 (2019-03-31): Fixed MAX6675 readout when in hardware SPI configuration.
### Requirements

* Filling out this template is required. Pull Requests without a clear description may be closed at the maintainers' discretion.

### Description

- Fixes hardware SPI initialization in `read_max6675()`.
- Adds pullups to the `MAX6675_SS#_PIN` pins before they are used.

### Benefits

MAX38155 sensors work in hardware SPI on STM32F1. Not tested in soft SPI (as it is not available on STM32F1) nor on other platforms.

### Related Issues

Fixes #13528
closed 2019-05-15 by Idorobots