MarlinFirmware/Marlin #13532: Suggestion for the overlay

cinealfa, 2019-03-31:
### Description

### Steps to Reproduce

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

cinealfa, 2019-03-31:
I propose moving the rotation arrow's orientation around so that they are pointing in the same direction that the direction arrow is pointing . That way if you turn the knob in the direction of the  arrow (either one) the probe will move in that direction. The way it is now the rotation arrows point on clockwise or anti clockwise and are not related to the direction of the probe. It would be a matter of rotating the up rotation arrow180 degrees and putting the point on the other end of the circle. That way the arrow is pointing upwards in line with the motion of the probe. The down arrow simple needs to be on the other side of the rotation arrow as it it is already on the correct end of the circle.this would provide a very intuitive way to translate the axial rotation of a know to the correct linear movement of the head/probe

marcio-ao, 2019-04-11:
@cinealfa : Good idea! I've put in a PR for this.

marcio-ao, 2019-04-12:
@cinealfa: This has been merged. Could you test it out and if it is what you requested, please close this issue ticket. Thanks!

cinealfa, 2019-04-13:
will do! As soon as this print finishes i will upload the update.

closed 2019-04-13 by thinkyhead

cinealfa, 2019-04-14:
Went to close this but it had already been done, I had to compile and test it. Anyway good thing it works.