MarlinFirmware/Marlin #13533: Problem with UBL and LCD-BED-Leveling Compile

tekknix, 2019-03-31:
Hi,

using the current BIQU SKRv1.1 Board incl. 4 TMC2208 with UART, BLTouch and the original Ender 3pro LCD-Display everything working fine until i try to use UBL instead of ABL. Compilation isn't working, due to the following error:

```
Marlin\src\lcd\menu\menu_bed_leveling.cpp: In function 'void menu_bed_leveling()':
Marlin\src\lcd\menu\menu_bed_leveling.cpp:263:71: error: '_lcd_toggle_bed_leveling' was not declared in this scope
MENU_ITEM_EDIT_CALLBACK(bool, MSG_BED_LEVELING, &new_level_state, _lcd_toggle_bed_leveling);
```
Indeed there isn't any void declaration of "_lcd_toogle_bed_leveling" included in the File "menu_bed_leveling.cpp".

Have tried to get rid of it, but UBL with LCD-BED-Leveling seems not to be working.

```
// Homed and leveling is valid? Then leveling can be toggled.
  if (is_homed && leveling_is_valid()) {
    bool new_level_state = planner.leveling_active;
    MENU_ITEM_EDIT_CALLBACK(bool, MSG_BED_LEVELING, &new_level_state, _lcd_toggle_bed_leveling);
  }
```

Someone seen this before?

Attached configuration-Files and Pins-Declarations.

Any help is apprectiated.

Regards
MoS-tekknix

[ender3pro+skrv1.1+lcd12864+bltouch.zip](https://github.com/MarlinFirmware/Marlin/files/3026409/ender3pro%2Bskrv1.1%2Blcd12864%2Bbltouch.zip)

Ludy87, 2019-03-31:
disable `SLIM_LCD_MENUS`

This is just a stopgap, a developer should deal with it.

tekknix, 2019-03-31:
Many thx for that. Will give it a try. Have read something about SLIM_LCD_MENUS but only if this was disabled ....

tekknix, 2019-04-01:
Tested and it works with "SLIM_LCD:MENUS" disabled. Thanks and close

closed 2019-04-01 by tekknix

Ludy87, 2019-04-01:
Nevertheless, the mistake should be looked at by someone like @thinkyhead