MarlinFirmware/Marlin #13534: [BUG] After Bed Leveling LCD stays in Print mode

reloxx13, 2019-03-31:
### Description

If i level the Bed with G29, the LCD stil says "Printing" and it shows only the printing menu items.
While in Octoprint i can start a print w/o problems.

### Steps to Reproduce

This is my level code in octoprint Bed VIsualizer which i use to level.

```
M190 S70 ;wait for heat
M155 S30 ;lower temp reports
G28 ;home
G29 T ;ABL
M500 ;save mesh
M501 ;load mesh
M155 S3 ;default temp report
M140 S0 ;stop heating
```

**Expected behavior:**
Printer LCD shows Non-Printing Mode menu

**Actual behavior:**
Printer LCD shows Printing Mode menu

#### Additional Information
Config:
https://github.com/reloxx13/Marlin/tree/my-marlin/Marlin

boelle, 2019-07-17:
is this still a problem?

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@reloxx13 is this still a problem?

reloxx13, 2019-09-24:
yes

boelle, 2019-10-21:
thinking of this, does it only happen when printing/leveling from octoprint?

AndKe, 2019-10-21:
@boelle yes, in my case, it is the only way I see it happen as the levelling is in the start of any job.

boelle, 2019-10-21:
i print from octoprint also but i dont do level before each print, only when print shows its needed

i have #define RESTORE_LEVELING_AFTER_G28 and the only thing i need to check on LCD is that bed level is enabled, if it says off i just click that to make it say on

normally i only do this after i power up the printer and after a disconnect, 

and even if i home its enabled, doing level before each print is kind of overkill, 

but this does not mean that there is not a bug, its just to figure out if its a problem with marlin or octoprint

reloxx13, 2019-10-22:
yes, i use octoprint.

i only level if needed with bed level visualation tool.

yes, still happens, had it again few days ago.

sometimes i level and want to change filament afterwards oder move z up to clean the bed and i cannot cause it stays in print mode (print mode menu and status text "printing"). 
had this again a few days ago.

boelle, 2019-11-03:
@reloxx13 dusting off this one, is the issue still there?

reloxx13, 2019-11-11:
+

boelle, 2019-11-11:
and that means?

reloxx13, 2019-11-11:
positive :D

boelle, 2019-11-11:
oki, a shame its still there :-(

boelle, 2019-11-21:
@thinkyhead @shitcreek time to have a look here before the issue gets lost and not fixed

boelle, 2019-12-07:
since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2019-12-24:
btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

reloxx13, 2020-01-07:
there were leveling changes last days and ill have to re-test.

boelle, 2020-01-07:
oki. looking forward to that :-D

boelle, 2020-01-14:
@reloxx13 did you retest?

reloxx13, 2020-01-17:
still happening.

load/unload text is also switched,
load shows text for unload, unload shows text for load, since months.
german fw latest bugfix 2.0.x from today.

i wont answer any further questions if its fixed or not. ask the devs or tell em to properly use git and #issue id in commits.

Vertabreak, 2020-01-25:
If you don't plan to be involved in the solution it's likely it won't be fixed unless someone else with the same issue gets involved to sort it out. I personally can't reduced this issue.

boelle, 2020-01-25:
@reloxx13  i agree here, you are the only one with this issue so far, i cant reproduce it either and tried many times but it works every single time

and you last 2 lines of comment are kind of uncalled for and belongs to where i call it "nasty"

thinkyhead, 2020-01-26:
Well to be fair, this issue has lingered a long time without a single human capable of investigation coming along to look into it. We could really use more initiative in that area. It can't all fall to me.

At a glance, the problem seems to be that you need `M190 S0` and not `M140 S0` at the end of your script. Please confirm if this helps!

thinkyhead, 2020-01-26:
Well to be fair, this issue has lingered a long time without a single human capable of investigation coming along to look into it. We could really use more initiative in that area. It can't all fall to me.

At a glance, the problem seems to be that you need `M104 S0` and not `M140 S0` at the end of your script. Please confirm if this helps!

thinkyhead, 2020-01-26:
Ach, never mind that last suggestion. But here's what I can tell you:
- It's unusual to use a G-code script from host to do `G29`.
- Marlin uses clues to detect that a print job has completed, but the host can help out.

So, see if it helps to add `M77` to the end of your script. If that doesn't do the trick, we can look further.

thinkyhead, 2020-01-26:
To remedy this (for future Marlins and beyond) we should take account of bed heating and not just hot-end heating in our evaluation of print job start and stop. Contribs should look at `PRINTJOB_TIMER_AUTOSTART` in `M104_M109.cpp` and contrast with `M140_M190.cpp`.

Vertabreak, 2020-01-26:
strange as i used a script for UBL on both octoprint and pronterface and it does not produce this effect.

closed 2020-01-28 by thinkyhead