MarlinFirmware/Marlin #13535: SKR 1.3 E0 E1 wont move with Lin Advance

devolf92, 2019-03-31:
Hello! Yesterday i have download latest bugfixed marlin and started to setting up skr 1.3
Everything works fine except E1 and E0 motor, they wont spin with hot nozzle or even disables minimum temperature protection.
With disabled Lin Adv everything works
With enabled and k=0.1 and MINIMUM_STEPPER_PULSE 0 - won't.
With enabled and k=0.1 MINIMUM_STEPPER_PULSE 2 - works

i don't have oscilloscope for a now to get a detailed information about step signals

ManuelMcLure, 2019-03-31:
What stepper drivers are you using? Did you properly set the driver in Configuration.h (``E0_DRIVER_TYPE ``/``E1_DRIVER_TYPE``)? Please attach a zip file containing your Configuration.h and Configuration_adv.h.

gloomyandy, 2019-03-31:
There are a number of other issues here reporting that you need a minimum stepper pulse of at least 1 with some drivers. Others have also reported that lin advance does not work well with TMC Stealthchop (and advise the use of Spreadcycle). I'm not sure any of this is really a bug though, more configuration options that may need to be set depending upon the machine.

devolf92, 2019-03-31:
Im using TMC2130 and [here is configuration](https://dropmefiles.com/IiL4S)
LinAdvance and minimum pulse are disabled for a now

sergeysi779, 2019-04-01:
I can confirm this issue:
https://github.com/MarlinFirmware/Marlin/issues/11614
https://github.com/MarlinFirmware/Marlin/issues/13463
In my case motor doesn't work even with spreadCycle.

ZS1FJK, 2019-04-01:
I can confirm this issue:
#13312 
also using TMC2130 on archim 2 board
Setting  MINIMUM_STEPPER_PULSE 1 works for me.

jamesarm97, 2019-04-20:
Just came to see if this issues existed. It has been driving me crazy trying to track down what made the extruder stop moving but it seems to be this feature. I even tried setting M900 K0 to disable and it still won't move unless the feature is disabled all together.

gloomyandy, 2019-04-20:
Just to add that for me Linear Advance works fine with TMC2208s using Spreadcycle and MINIMUM_STEPPER_PULSE 2

dralves, 2019-05-07:
Same problem on SKR 1.3 with TMC5160's (with BLTouch if that matters). Everything works great except linear advance (s_curve, junction_deviation, etc..). When it's on the E0 won't budge.

gloomyandy, 2019-05-07:
@dralves does setting MINIMUM_STEPPER_PULSE 2 fix the problem for you?

harleycarr, 2019-05-20:
Same issue as well. Was functioning great in 1.1.9 but switch to 2.0 and had to disable lin_adv. It was not as bad in spreadcycle but even so with lin_adv on it was severely under-extruding.

govido, 2019-05-28:
Since my old issue got closed als solved?! I will post this one here:
[image]
I've tried now 32! different firmware combinations and none of them will get the 2130 drivers to work correctly. They whine like hell in one mode, or are totally quiet when homeing, but are loud as hell when switching to printing mode.
I have no ideas left at all... I'm also using the SKR 1.3 board.
Even without linear advance the printer is a screaming loud unreliable mess.
How is it possible, that 32bit is so vastly different in Marlin 2.0?
All my drivers worked perfectly on my old ramps 8bit board... reliably and quiet.
Is there anything I can test or tweak?
The last ones in the list didn't work at all.
And a minimum stepper pulse of 1or higher is an absolute must! The documentation in the marlin file is just wrong for 32bit.

govido, 2019-05-28:
It got late last night... today i found that I made a mistake with the microsteps on Z when changing to 64/32 steps, that made some intersting prints.
I can get the printer to work pretty noisy for now.
But issue #14139 got me today. As of now only 2.0 builds before may 23rd are running on SKR 1.3 without modifications.

govido, 2019-05-29:
Hi, I want to give a quick update to the problem.
I got the printer running now with MINIMUM_STEPPER_PULSE 1.
Linear Advance is on but set to 0, I have specific filament settings in PursaSlicer that get set for each filament type. That seems to work.
Stealthchop is working beatifully, the real problem is spreadcycle.
It works but is horribly noisy. As soon as 1 stepper is changing to spreadcycle the motors make whining noises and even the PWM heaters frequency is audible.
@teemuatlut 
What is different driving the stepper drivers in spreadcycle mode? There must be specific difference...
Is it a hardware/software PWM/SPI problem?
But that would make stealthchop noisy too...
Otherwise the firmware seems to work OK now, if someone is interested in my settings check my github fork.

teemuatlut, 2019-05-29:
In spreadCycle the noise a motor makes is largely determined by the chopper settings and the electrical characteristics of your motor. A higher inductance motor results in slower changes in winding current which leads to lower and more audible chopper frequencies. I've tried to tune some predefined profiles for ease of use but ultimately this is an impossible task because as said, it depends on _your_ hardware.
When scoping the waveforms I tried to aim for around 20kHz. Too high leads to losses and too low means noise.

I see you're using the default 24V profile but perhaps this is too low f for your particular setup.
You can try something like this:
```cpp
#define CHOPPER_TIMING { 3, -1, 1 }
```
This will increase your chopper frequency when in spreadCycle mode.

I can't say much about why the TMCs don't like Linear Advance because I don't use it myself.
Tried it once quite a long time ago and saw much of the same problems as others have.
A puzzling aspect of it is that increasing minimum step pulse has an effect on it even when we're actually not coming close to the specified min pulse width.
A more likely cause (IMO) is that linear advance and the TMC microplyer don't play nice together. But seeing as many have tried disabling the feature...
All I can say for now is that we'll know more once I get to actually analyzing the E pulses and see what's really going on.

govido, 2019-05-29:
@teemuatlut 
I've used the 2130 drivers already with my old 8bit board with a much older version of Marlin 2.0 (over half a year old). In that version there was no option to tweak for the chopper. How was it handled 'back then'? My options where pretty much the same (LA, Stealth, Hybrid etc.) without the noise.
Or does the ATMega handle PWM differently?
But I will have look at the different chopper timings

teemuatlut, 2019-05-29:
Before implementing the profiles Marlin2.0 used the same chopper timings as 1.1.9.
Those values are still preserved in one of the profiles but they'll very likely just go the opposite way of what you'd want.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

cronos45, 2019-12-23:
@boelle loves closing