MarlinFirmware/Marlin #13536: [BUG] READ:error:SD read error

ZS1FJK, 2019-03-31:
Archim 2 board installed in Lulzbot Taz 6
Printing from SD card stop randomly in print . When i retry the print it stops in a different position every time.

1. Create stl using simplify 3D. 
2. Save file to printer SD card
3. Print from SD card using 3d printer interface. 

Expects printer to print complete file.

Printer stops printing. 
The SD card becomes inaccessible but is still present. Just the name disappears and cant open it.
If i connect to the machine and look at the log file i see this READ:error:SD read error.

I have tried other SD cards 4 different makes all 8 gig cards. I tried doing the SD card formatter by TUXERA. I have tried swopping the cables. From the board to the screen. 
[Uploading Marlin.zip…]()

Im not sure what to do next here.

ZS1FJK, 2019-03-31:
I should add. it prints fine over USB

ZS1FJK, 2019-03-31:
Issue fixed by changing
#define SPI_SIXTEENTH_SPEED  to
#define SPI_SPEED SPI_EIGHTH_SPEED

closed 2019-03-31 by ZS1FJK

jabirkhalid, 2020-04-29:
Please share a path for understand were to make this changes. and line number.