MarlinFirmware/Marlin #13541: [BUG] 2.0.bugfix - Power Loss Recovery: The second time does not work!

maxlinux2000, 2019-03-31:
### Description
I activated the Power_loss_recovery on my gt2560 card and it works very well!
I cut the electricity, I give it again, the resume menu is presented, and the printer reprints from where it was cut.

The problem comes out when I cut the electricity the second time. The printer stops, restarts and stays still without presenting the resume menu again.

I have tried to restart several times, but the resume menu does not appear.
I think that when a Print is resumed, is disabled the power_loss_recovery option.

Please fix it, tnx

### Steps to Reproduce

0. Make sure you do not have the USB cable connected to a computer.
1. start printing a piece
2. after a few levels, cut the electricity from the switch
3. After about 10 seconds turn on the printer
4. the "Resume" menu appears and select "resume"
5. The printer makes the "home" of XY and heats the "nozzle"
6. The printer continues to print from where it stopped as expected.
7. After a few layers, turn off the printer again.
8. After about 10 seconds turn on the printer

Expected behavior: The "Resume" menu must appear
Current behavior: The main info screen appears.

#### Additional Information

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3026870/Configuration.zip)

Mysel07, 2019-04-04:
Yes, I have the same problem.

maxlinux2000, 2019-04-04:
same problem in marlin 1.1.9 bugfix

robbycandra, 2019-04-11:
The problem fixed in bugfix 2.0.
Second time does not work because the Z axis status is unhomed after resume.

maxlinux2000, 2019-04-11:
> The problem fixed in bugfix 2.0.
> Second time does not work because the Z axis status is unhomed after resume.

but the power_loss_recovery actually is not useful at all.
Every time there is a blackout I have to cross my fingers so that it does not happen again?

If I have to make a long print several days and there is a blackout for example by a storm, it is very useful to turn off the printer.
Using a UPS is useless because after one hour it goes off equally, because of the batteries discharged.
Murphy's law says that when the light goes out, your wife decides it's time to make a washing machine, take a shower with the electric boiler, and wash the dishes in the dishwasher ... so the light will be missing again, and your You lose the pieces you were printing.

So please, change the power_loss_recovery so that you always keep in the SD the line number of the gcode file, where you wounded until the light is gone.
In this way if there is a type file on the SD card:
testprint.gcode.526
means that you have to continue printing the testprint.gcode file from line 526.
This is an example, I don't know how it working actually

regards
MaX

boelle, 2019-07-17:
@maxlinux2000 have you tried bugfix 2.0.x ?

boelle, 2019-07-17:
also remember for power loss recovery to be effective the powersupply has to provide enough juice for the printer to work long enough that it can save things

biggest power eater is the bed, so fit an rely that breaks power to bed if mains power is gone, you have to know what you are doing, next power eater is the hotend so you can do the same here

that will take less juice from the power supply if power is lost and give printer more time to save etc  etc

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@maxlinux2000 is this still a problem?

maxlinux2000, 2019-09-24:
sorry but in this period I don't have time to print nor update the firmware.

maxlinux2000, 2019-10-21:
sorry but in this period I have no time to print in 3D, so I can't confirm that.

boelle, 2019-10-21:
would it be reasonble to close this one as @robbycandra said it was fixed? if you get time to check then we can reopen this one or you can create a new one?

maxlinux2000, 2019-10-21:
yes, please close it.
When I can try again I will reopen if necessary
tnx

boelle, 2019-10-21:
oki doke :-D

closed 2019-10-21 by boelle