MarlinFirmware/Marlin #13542: [BUG] 2.x UBL mesh boundary z-hop

Andrei-Pozolotin, 2019-03-31:
### Description

When using UBL with large inset, leveling system introduces +0.5 ... +1.0 mm z-hop when crossing the inset boundary in the X direction.

### Steps to Reproduce

1. using leveling activation:
```
G28 ; home/reset
G29 P1 ; measure
G29 P5 C0.80 ; adjust
G29 T ; report
G29 A ; activate
```
2. using UBL 50 mm inset with 200x200 mm bed

3. using UBL 3x3=9 point mesh, 100x100 mm active area

4. using UBL with BLTouch probe, with (0,46) mm probe offset

5. when printing 90x90 mm square (attached as `pyra-test-090`), which fits inside 100x100 mm UBL active area, there is no problem

6. when printing 110x110 mm square (attached as `pyra-test-110`), which exceeds 100x100 mm UBL active area, firmware performs +0.5 ... +1.0 mm Z hop when crossing right UBL inset border in the positive X direction, and returns back to baseline in Z when crossing right UBL inset border in the negative X direction

7. UBL topography for `G29 T ; report`
```
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     ( 50,150)      (150,150)
Recv:         0       1       2
Recv:  2 | -0.800 [-0.800] -0.797
Recv:    |
Recv:  1 | -0.755  -0.789  -0.751
Recv:    |
Recv:  0 | -0.824  -0.863  -0.821
Recv:         0       1       2
Recv:     ( 50, 50)      (150, 50)
Recv: 
Recv: ok
```
#### Additional Information

1. config snapshot
https://github.com/random-builder/Marlin/tree/47d30fb20ca8b3587195bacce17e26836b166245

2. test model stl/gcode files:
[marlin-bug-ubl-hop.zip](https://github.com/MarlinFirmware/Marlin/files/3026892/marlin-bug-ubl-hop.zip)

gloomyandy, 2019-03-31:
This may well be a bug, but can I ask, why are you doing this? Seems a little strange to use a mesh that does not cover all of the bed and then print something that falls outside of it? Why not just use a mesh that covers all of the bed? You can always populate the non probed areas with some sort of values (either based on the probed values or otherwise). There is also the option to set what to do when outside the mesh (UBL_Z_RAISE_WHEN_OFF_MESH) if you really want to go this way. I'm sure you have reasons for doing this I'm just curious as to what they are!

Andrei-Pozolotin, 2019-03-31:
1. due to large probe/nozzle offset smaller inset results in loss of front row of points, resulting in 3x2 mesh
2. experiments show that [3x3 large inset mesh] is more accurate/uniform then the [3x2 small inset mesh]

gloomyandy, 2019-03-31:
Why not use a 3x4 (or larger) mesh (which should give you a 3x3 populated mesh) and then populate the missing row(s) using G29 P3?

The problem with simply interpolating is that it is a good way to crash the nozzle into the bed (and most people would not think that is a good thing, or expect it). Imagine if you have a dished bed and you keep the same Z height, or if a local bed dip means that the angle towards the edge of your mesh is going downwards, both cases will result in damage. It is for this reason that G29 P3 does not normally interpolate values that go downwards (probably not totally accurate, but it avoids generating a mesh that will crash the nozzle).

 Is there a reason why you are not using  UBL_Z_RAISE_WHEN_OFF_MESH?

Andrei-Pozolotin, 2019-03-31:
I am testing `G29 P3` now, thank you.

gloomyandy, 2019-03-31:
There are also a few (undocumented) options to G29 P3....
                  // P3.1  use least squares fit to fill missing mesh values
                  // P3.10 zero weighting for distance, all grid points equal, best fit tilted plane
                  // P3.11 10X weighting for nearest grid points versus farthest grid points
                  // P3.12 100X distance weighting
                  // P3.13 1000X distance weighting, approaches simple average of nearest points

Andrei-Pozolotin, 2019-03-31:
1. I switched to zero inset and 4x4 mesh
```
  #define MESH_INSET 0             
  #define GRID_MAX_POINTS_X 4
  #define GRID_MAX_POINTS_Y 4
```
2. I use following leveling activation with `P3.1`:
```
G28 ; home/reset
G29 P1 ; measure
G29 P3.1 ; populate
G29 P5 C0.75 ; adjust
G29 T ; report
G29 A ; activate
```
3. now UBL produces very good quality mesh, thank you so much! this also resolves my issue.
config snapshot:
https://github.com/random-builder/Marlin/tree/ae55f97b095951e3c292d195d87745074869c8ac

Roxy-3D, 2019-03-31:
I'm glad you are getting good results...   But personally....   I think it is safer to have more mesh points.    Maybe 8 x 8 ????

Andrei-Pozolotin, 2019-03-31:
@Roxy-3D:
1. thank you for looking into this
2. yes, 8x8 is good, but a bit slow, and glass bed is flat enough, and mesh is already good enough
3. I added video snapshot:
a) leveling https://www.youtube.com/watch?v=q1HR463qDrs
b) normal print https://www.youtube.com/watch?v=hlhMdj4al8I
c) print with z hop https://www.youtube.com/watch?v=sEUsLTSCoSE

ManuelMcLure, 2019-03-31:
You only need create the mesh once - then you can save it with ``G29 S`` and restore it with ``G29 L`` before printing.

Roxy-3D, 2019-03-31:
Yeah...    If it is working well for you....  Good!       I was just stating an opinion. 
I doubt I will ever have a piece of glass flat enough to do that.    But so be it!!!!

gloomyandy, 2019-03-31:
You know that you do not need to create the mesh each time? You can just create the mesh once and then store it in eeprom. Then load it at the start of your print. If needed to can "tweak" the level by running say a G29 J3 (which probes 9 points and adjusts the height and tilt of the loaded mesh) at the start of your print. This means you can have a bigger mesh and still only do a few (if any probes on each print).

Andrei-Pozolotin, 2019-03-31:
great, thank you guys for your ideas. reasons to re-level before each print via octoprint snippet:
1. there is non-repeatable (run-to-run) temperature creep when using 85C chamber
2. there is large mesh difference between 25C PLA and 85C POC printing mode
3. single-sided lead screw design is too flimsy, keeps bending over time
4. 50 seconds for 4x4 mesh leveling before print is fast enough

Roxy-3D, 2019-03-31:
OK...    That is all fine and good...     But if it were me...   I would generate (and tune) a mesh for PLA and a separate mesh for each type of material I was going to print.     And the other thing you might want to try is using the Z-BabyStepping when you start a print to fine tune the nozzle's extrusion when you start a print and it is doing the skirt.

Andrei-Pozolotin, 2019-04-01:
1. I still resist the idea of using static/stored leveling mesh 
2. mesh-on-demand has great "peace of mind" value: it should pick up changes when they occur
3. which brings in a new `G29 P7` "quality check" feature request, what do you think:
```
G29 P7 DXXX ; abort print when mesh fails a quality measure, say standard deviation
```

Roxy-3D, 2019-04-01:
> I still resist the idea of using static/stored leveling mesh

Well...   We are looking at the problem from different perspectives.   The whole reason 'Bed Leveling' needs to be done is because the 3D-Printers are not built at the same 'Quality' level as a CNC machine.    CNC machines don't have bed leveling because the tables and tools hold very tight tolerances in their movements.

Also...   The nozzles do not move perfectly 'square' with respect to the X-Axis (on a normal, Prusa type machine).   There is usually a little bit of 'twist' as the nozzle moves left to right.    What this does is change the probe distance from the nozzle a small amount depending upon where the measurement is being take on the bed.

So...  when it is all said and done, the Auto-Probing works and is helpful, but it is just a first pass at generating an accurate mesh to describe the bed's topology.    Once the first pass is generated, it is very normal to find areas where the mesh is close, but not perfectly describing the Z-Height correction needed.    And that is why mesh validation tools and the ability to edit the mesh are provided.

To each his own....   If you are getting good results doing what you are doing...   That is great!   But most people would be better off taking the described paths.

gloomyandy, 2019-04-01:
Personally I go the middle path. I create two meshes (at different temperatures) fine tune them and store them in slot 1 and 2. I then load the required mesh at the start of my print and use G29 J3 to fine tune any variation in home position/bed height or tilt. This combination works very well for me.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle

Andrei-Pozolotin, 2019-10-08:
@boelle : solved, thank you