MarlinFirmware/Marlin #13543: [BUG] G29 AUTO BILINEAR : First row shifted (once)

Bidulglloq, 2019-03-31:
### Settings

Cartesian printer, Dual Extruder, T0 (right) extruder active
HOTEND_OFFSET_X {0.0, -33.00}

BLTouch sensor, offsets:   X: -49 (left), Y -40 (front)  from T0 

AUTO_BED_LEVELING_BILINEAR: enabled
Z_SAFE_HOMING: Disabled
BED_CENTER_AT_0_0: enabled

### Description

After homing (G28), the **_first_** time I execute G29 after G28:
- the 1st probe point is shifted by d=33mm to the right from expected position
- the 2nd probe point is also shifted by d=33mm to the right from expected position
- while going to the 3rd probe point, the head bangs on the right side (no steps lost)
- the following probe points are correctly located

The **_second_** and following times I execute G29:
- all probe points are on their correct positions

(see picture)

[image]

### Steps to Reproduce

1. G28
2. T0
3. G29

**Expected behavior:** see picture

**Actual behavior:** see picture

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Config files.zip](https://github.com/MarlinFirmware/Marlin/files/3026890/Config.files.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Bidulglloq, 2019-03-31:
More information:
Printer: Bibo2 Touch (MKS GEN L board + MKS TFT 28)

Bidulglloq, 2019-03-31:
I tried to change the number of rows/cols to 4...
Same problem: the first G29 is completely wrong, the following ones are ok...

[image]

Bidulglloq, 2019-03-31:
I also tried to deactivate Bed centre at 0 0
`//#define BED_CENTER_AT_0_0 `

But now, all probe points are located on the back & right corner (coordinates > 0)

[image]

Bidulglloq, 2019-04-04:
I tried with PROBE_Y_FIRST defined.
Everything is shifted until the X max endstop is triggered...
[image]

Reminder: BLTouch sensor, offsets: X: -49 (left), Y -40 (front) (from T0, right?)

I've use M112 S38 and Verbose mode to get more information:

**First, G28:**
```
SENT: G28
Active Extruder: 0
READ: X:107.00 Y:93.00 Z:5.00 E:0.00 Count X:10700 Y:9300 Z:2000
READ: ok
```
you can see  after homing to the endstops, **X is reported at 107**, but the **left** extruder (T1) is at the edge of the bed (T0 is then at 107+33 = 140...)
G0 X0 Y0 with **T0** active is actually bringing **T1** at the bed centre (T0 is supposed to be there) => everything is shifted

**Now, G29 for the first time...**
```
SENT: G29 V4
READ: G29 Auto Bed Leveling
READ: Bed X: -97.000 Y: -83.000 Z: 0.080
READ: Bed X: -97.000 Y: -15.000 Z: 0.207
READ: Bed X: -97.000 Y: 53.000 Z: 0.175
READ: Bed X: -3.000 Y: 53.000 Z: 0.085
READ: Bed X: -3.000 Y: -15.000 Z: 0.162
READ: Bed X: -3.000 Y: -83.000 Z: 0.110
READ: Bed X: 91.000 Y: -83.000 Z: 0.172
READ: Bed X: 91.000 Y: -15.000 Z: 0.167
READ: Bed X: 91.000 Y: 53.000 Z: 0.008
READ: Bilinear Leveling Grid:
READ: 0 1 2
READ: 0 +0.080 +0.110 +0.172
READ: 1 +0.207 +0.162 +0.167
READ: 2 +0.175 +0.085 +0.008
READ: 
READ: X:140.00 Y:93.00 Z:4.99 E:0.00 Count X:14000 Y:9300 Z:2000
READ: ok
```
Now, you can see it reports **X being at 140** (thus, the head is exactly at the same spot as before)
From here, G0 X0 Y0 drive E0 really at the center..

**Just to make sure, another G29:**
```
SENT: G29 V4
READ: Bed X: -97.000 Y: -83.000 Z: 0.055
READ: Bed X: -97.000 Y: -15.000 Z: 0.167
READ: Bed X: -97.000 Y: 53.000 Z: 0.145
READ: Bed X: -3.000 Y: 53.000 Z: 0.123
READ: Bed X: -3.000 Y: -15.000 Z: 0.142
READ: Bed X: -3.000 Y: -83.000 Z: 0.105
READ: Bed X: 91.000 Y: -83.000 Z: 0.157
READ: Bed X: 91.000 Y: -15.000 Z: 0.152
READ: Bed X: 91.000 Y: 53.000 Z: 0.008
READ: Bilinear Leveling Grid:
READ: 0 1 2
READ: 0 +0.055 +0.105 +0.157
READ: 1 +0.167 +0.142 +0.152
READ: 2 +0.145 +0.123 +0.008
READ: 
READ: X:140.00 Y:93.00 Z:4.99 E:0.00 Count X:14000 Y:9300 Z:2000
READ: ok
```
Now, the points are probed at their correct location, **X at 140** is still ok

As a conclusion:
- After G28, everything is shifted to the right by 33mm
- X is reported incorrectly (verbose)
- If I force update X (M92 X140) to its correct value G0 moves are ok (but the first G92 will be still off)
- First G29 will probe everything shifted UNTIL the X max endstop is triggered (following probe points are ok
- From now on, everything is ok (including following G29)
- when I run again a G28, shifting is happening again

Bidulglloq, 2019-04-04:
More info as I investigate...

As said above, **with BED_CENTER_AT_0_0 defined**
G28 returns:
```
SENT: G28
READ: echo:Active Extruder: 0
Active Extruder: 0
READ: X:107.00 Y:93.00 Z:5.00 E:0.00 Count X:10700 Y:9300 Z:2000
```
 And **Without BED_CENTER_AT_0_0**
G28 returns:
```
SENT: G28
READ: echo:Active Extruder: 0
Active Extruder: 0
READ: X:141.00 Y:93.00 Z:5.00 E:0.00 Count X:14100 Y:9300 Z:2000
```

In both cases, the homing is physically the same, with the same endstops...

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-09-24:
@Bidulglloq is this still a problem?

closed 2019-10-12 by boelle