MarlinFirmware/Marlin #13545: Pause Menu Doesn't Work

cinealfa, 2019-03-31:
### Description

Marlin 2.0 Pause Menu Doesn't pause

### Steps to Reproduce

1.Start Print
2.Go to Pause in Menu
3.Select Pause

**Expected behavior:** Printer should pause
**Actual behavior:**Printer Doesn't Pause

Octoprint can pause the machine remotely but I am unable to do it through the menu
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/3027161/Archive.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

robbycandra, 2019-04-01:
Its working on my printer

psavva, 2019-04-01:
Try using the latest bugfix and let us know if it works?

Btw, it works for me too...

reloxx13, 2019-04-01:
do u print from octoprint and try to pause octoprint?

InsanityAutomation, 2019-04-04:
Turn on the rest of this : 

#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
  #define HOST_PROMPT_SUPPORT
#endif

And make sure your octoprint is up to date. Please note 1.3.11 has improvements here (Currently RC1) so it may be beneficial to you to update to that now.

cinealfa, 2019-04-06:
Octoprint is up to date. And it pauses just fine through octoprint as I said in my post. I will turn on what you suggest Insanity BTW the build was up to date as well. As far as it works on your printer robbycandra,  what time can I come over and use it? :-p
The three least helpful troubleshooting phrases of all time have to be: 
1 Well it worked yesterday, 
2 Well it works on mine (completely different board printer computer, etc. 
and my favorite: 
3 Well should work.
I will report back once I finish this print.

gloomyandy, 2019-04-06:
Whet exactly does "Start a print" mean in this context? Are you printing from the SD card or are you printing from a remote system like Octoprint?

cinealfa, 2019-04-07:
I am using octoprint and not upload loading to the SD card.I have made the adjustments to the settings and will report back on my next print

closed 2019-04-10 by thinkyhead

cinealfa, 2019-04-11:
That solved the problem. However it does raise the question should there be
a sanity check to make sure both are selected? Ot is there a situation
where only one would be used?