MarlinFirmware/Marlin #13548: No reprap discount or full graphic lcd display

curnowdj, 2019-04-01:
Hello I am trying to get a Arduino 2560 with ramps v1.4 working. I have Marlin 1.1.9 but I get no display on a reprap discount smart controller or a reprap full graphic smart controller. The marlin seems to compile and upload ok but no response on the lcd. I change lcd in the fw each time but no joy.
[image]
[image]

psavva, 2019-04-01:
Please attach your config files, and also attach a photo of your LCD

curnowdj, 2019-04-02:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3032360/Configuration.txt)
[image]

jmz52, 2019-04-02:
There are two version of display connectors, please attach a photo of displays' backs.
Is display's backlight lit when display is connected?

robbycandra, 2019-04-03:
Try turn the cable.  But first you have to grind the black connector so you can plug it upside down.

curnowdj, 2019-04-03:
[image]

cdedwards, 2019-04-03:
or you can pull those black plastic pieces off the connectors. Then no shaving required

curnowdj, 2019-04-03:
Is this a common issue? I have trouble believing these cables are wrong.

jmz52, 2019-04-03:
Makerbase (MKS) boards and displays have EXP1/EXP2 connectors turned by 180 degrees.
This results in two versions on EXP1/EXP2 connections RepRap and MKS.
I might be mistaken, but your controllers look like a RepRap version, so they should be working with RAMPS.

There is a simple test for boards and smart controller compatibility - if backlight in off when display is connected  to board and board is powered, then your display and board are of different flocks and you need to turn connectors.

ManuelMcLure, 2019-04-03:
Also, it’s easy to end up connecting EXP1 to EXP2 or vice versa by mistake - I know I’ve done it more than once.

curnowdj, 2019-04-03:
Yes they are both Reprap lcds. I have tried changing exp1 and exp2 cables around and when I do this the beeper sounds so I assume that is the wrong way.

curnowdj, 2019-04-03:
Both lcds have www.bigtree-tech.com web site as well.

Bob-the-Kuhn, 2019-04-03:
Here's pictures of 2560 + Ramps + the two LCDs.  Notice that the cables go straight across and that pin 1 of the cables is always to the top of the picture.

I dropped your config file onto Marlin 1.1.9 and then downloaded it to a 2560 + RAMPs system.  The RepRap graphics LCD worked correctly.

[image]

[image]

The shrouds on EXP1 & EXP2 in the pictures of the LCDs posted early in this thread are correct. Just in case .. here's a picture showing the correct orientation of the shrouds on EXP1 & 2 on the LCD adapter.

[image]

Another possibility is the cables being built inconsistently.  Here's a picture of the usual orientation of pin 1 of the cable to the raised keying plug on the connector.  Doesn't matter if the red stripe is above or below the connector.  

[image]

curnowdj, 2019-04-04:
OK so I am looking at some hardware failure. I will need to find out what has failed. I will have to think about where to start with that.

Bob-the-Kuhn, 2019-04-04:
What are your symptoms?

ManuelMcLure, 2019-04-04:
You said in the original post "The marlin seems to compile and upload ok" - have you confirmed this by connecting with a printer host such as Pronterface and making sure that you can communicate with the installed Marlin? This could be a more basic problem where Marlin isn't starting on the Mega. Make sure you can send something like an M119 (report endstops) and get back something.

One other thought - it seems like it might be possible to install the smart adapter (the little l-shaped board that plugs into the RAMPS and has the EXP1 and EXP2 connectors) shifted by one pin - that would definitely show a problem.

Does even the backlight on the LCD come on?

curnowdj, 2019-04-04:
Yes Back light comes on and There is a response to a M119 instruction
[image]

ManuelMcLure, 2019-04-04:
It does seem to be a hardware issue. You might want to take a magnifying glass and check the solder joints on the RAMPS, the Smart Adapter, and even the Arduino Mega. Given that neither display works, it's more likely that the problem is on that end.

jmz52, 2019-04-04:
One more thing - have you tried to adjust contrast on your displays?
There is an adjustable resistor at the back side of display.

ManuelMcLure, 2019-04-04:
On the Bigtree-Tech Full Graphic Controller clones the resistor is on the front - it's the blue and white gizmo on the bottom left.

curnowdj, 2019-04-05:
Yes I have turned the contrast pot and does not make any difference apart from darkening the display. I'll look more closely at the boards and see if I can spot something.

curnowdj, 2019-04-05:
Could these messages in the bottom panel be part of the trouble? 

Sketch uses 107958 bytes (42%) of program storage space. Maximum is 253952 bytes.
Global variables use 4468 bytes (54%) of dynamic memory, leaving 3724 bytes for local variables. Maximum is 8192 bytes.
Invalid library found in /home/dennis/Arduino/libraries/ADC_SEQR: no headers files (.h) found in /home/dennis/Arduino/libraries/ADC_SEQR
Invalid library found in /home/dennis/Arduino/libraries/arduino_108850: no headers files (.h) found in /home/dennis/Arduino/libraries/arduino_108850
Invalid library found in /home/dennis/Arduino/libraries/ADC_SEQR: no headers files (.h) found in /home/dennis/Arduino/libraries/ADC_SEQR
Invalid library found in /home/dennis/Arduino/libraries/arduino_108850: no headers files (.h) found in /home/dennis/Arduino/libraries/arduino_108850

ManuelMcLure, 2019-04-05:
I don't think so, although I don't know why you're getting them. It might make sense to reinstall Arduino and see if that helps.

Bob-the-Kuhn, 2019-04-05:
Personally, I like using PlatformIO to compile & upload.  It literally takes only 3-4 mouse clicks to compile & upload.  All the library selection & installation are done automatically in the background.

Instructions are [here.](http://marlinfw.org/docs/basics/install_platformio.html)

robbycandra, 2019-04-06:
Yes... PlatformIO is better
.

ManuelMcLure, 2019-04-06:
PlatformIO is kind of nice. I do wish I could figure out why the "Autobuild" stuff stopped working for me, but it's easy enough to just use the standard PlatformIO options.

Bob-the-Kuhn, 2019-04-06:
Please open another issue & we'll can fix Auto Build.

ManuelMcLure, 2019-04-06:
Done - see #13592

curnowdj, 2019-04-06:
Still trying to work out Platformio. Please be aware I am not a programmer. I think this program is more suited to someone who knows their way around writing code.

Bob-the-Kuhn, 2019-04-08:
@curnowdj - if' you've been following the link I gave you then the instructions need to be improved.

What problems are you having?  Screenshots of error messages and/or unexpected requests/questions are helpful.

curnowdj, 2019-04-09:
[image]
Is this EEPROM Mismatch part of the issue?

Bob-the-Kuhn, 2019-04-09:
In this case EEPROM Mismatch just means that you've never done the **M502** followed by **M500** command series.

I suggest turning off the EEPROM until after the display issue is solved.  It can cause a lot of confusion when debugging.

Bob-the-Kuhn, 2019-04-09:
I don't see what the problem is.  Time to start over.

Please ZIp and attach your current configuration.h file.

curnowdj, 2019-04-09:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3058737/Configuration.zip)
I have not disabled the eeprom on this yet.

Bob-the-Kuhn, 2019-04-10:
I changed the LCD from **REPRAP_DISCOUNT_SMART_CONTROLLER** to **REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER** and was able to download via Arduino 1.8.9 after changing U8Glib to the older version 1.18.0.

It's running just fine now.

closed 2019-04-10 by thinkyhead

Bob-the-Kuhn, 2019-04-11:
I think it's time to start replacing the hardware.  I believe something is defective.  There's no way to determine what may be bad any other way.

Bob-the-Kuhn, 2019-04-11:
@curnowdj  -  you mentioned that you were having problems installing PlatformIO.  I'm seriously thinking of putting out a YouTube video on this.  You'd be doing me a big favor if you'd watch it, try to use it and give me feedback.

[Marlin – easiest way to compile & upload (demo, installation & usage)](https://www.dropbox.com/s/qn1cvxf9xhdt3x8/Marlin%20Auto%20Build%20video%20%28rev%201%29.mp4?dl=0)

Here's the text that would accompany the video.

---

Marlin can be a pain to compile and upload due to fussing around with libraries or trying to figure out what compiler options to enable/disable.   Marlin’s Auto Build script/extension reduces the process to two to four mouse clicks.

Auto Build is a python 2.7 script that uses information available within Marlin to setup and run PlatformIO.  Access to this script is via menus that have been added to Atom, VScode and Sublime.

This video does a quick demo of how the Auto Build system works, goes through the process of installing all needed items on a new Windows 10 PC and then walks you through a very simple Marlin modification , compile and upload.

This video follows the Installing Marlin (PlatformIO) web page (link below).  It is highly recommended that this page be read in conjunction with this video.

URLS:
Installing Marlin (PlatformIO) - http://marlinfw.org/docs/basics/install_platformio.html
Marlin docs web page - http://marlinfw.org
Marlin graphic - http://clipart-library.com/clipart/774742.htm

thinkyhead, 2019-04-11:
Looks like a good complete tutorial in the works, and that's the kind of thing we should have embedded on the marlinfw site. I've added embedded videos for a couple of G-codes, so anytime we want to add videos to other types of pages I will have the hooks in place. For your 32-bit build tutorial, see if there are any pointers worth taking from Chris Riley's video https://www.youtube.com/watch?v=H-c8UTg-EMU